blueman-project/blueman #493: PSVITA pairing fails

fennectech, 2016-04-12:
I attempt to pair a psvita system running 3.60 firmware and it says the following "check the following passkey is displayed on the bluetooth(r) device  
passkey ######   

on the device   no prompt is oppened   or notification to pair or trust the deviice   no key is ever displayed on the computer     it shows "connected" for a moment until the psvita switches to say
The pass key may not be correct, or your bluetooth(r) device may have cancled the connection.   i used to be able to pair it to use it as an audio source    but it no longer works.  also i have tried this on two seperate machines and observed exactly the same reaction.

sys:1: PyGIWarning: Gtk was imported without specifying a version first. Use gi.require_version('Gtk', '3.0') before import to ensure that the right version gets loaded.
Terminating blueman-applet
Describe your next action (keep empty if done): 

=====================================

Use this data in your report:

=====================================

blueman: 2.0.3
BlueZ: 5.38
Distribution: 
Desktop: MATE

cschramm, 2016-04-18:
Please provide the output of blueman-applet from the point on where you start the pairing on the PS Vita. You can use blueman-report if you have any problems with the steps described on our Troubleshooting wiki page.

If nothing happens there, it's most probable not blueman's fault as it does not receive any event at all then. You can apply the same strategy on BlueZ then by starting bluetoothd with -d and checking the syslog.

You say it worked before. I assume when you initiated the pairing on the PS Vita a dialog showing the key popped up on your computer and you had to confirm it, right? How about initiating the pairing on the computer? Is that possible? Or is the PS Vita not even discoverable?

fennectech, 2016-04-18:
no    when it worked itwas using a much older bluetooth dongle     non bluetooth 4.0 complient      it prompts you to type in a key on bolth devices      
i have tried to initate the pairing from the computer   however nothing happens     if i use the devices tool   it jsut complains about paring failed

output from attempt with blueman runnig in a console

_________
on_device_property_changed (/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py:163)
list: device_prop_ch Connected 1 /org/bluez/hci0/dev_D4_4B_5E_49_65_67 
_________
row_update_event (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Connected 1 
_________
monitor_power_levels (/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py:185)
starting monitor 
_________
level_setup_event (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py:325)
animating up 
_________
on_device_property_changed (/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py:163)
list: device_prop_ch Connected 0 /org/bluez/hci0/dev_D4_4B_5E_49_65_67 
_________
row_update_event (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py:264)
row update event Connected 0 
_________
level_setup_event (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py:325)
animating down 
_________
update (/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py:186)
stopping monitor (not connected) 

[bluetooth]# pair D4:4B:5E:49:65:67
Attempting to pair with D4:4B:5E:49:65:67
Failed to pair: org.bluez.Error.AuthenticationRejected

if i try to pair using that it gives this error

if i try to pair from the device    it says this 
[CHG] Device D4:4B:5E:49:65:67 Connected: yes
[CHG] Device D4:4B:5E:49:65:67 Connected: no
[bluetooth]#

fennectech, 2016-04-18:
doing this fixed my issue   unsure why
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent
Default agent request successful

fennectech, 2016-04-18:
https://github.com/blueman-project/blueman/issues/298   i looked at this bug to find the solution      but the blueman does not seem to have been fixed

cschramm, 2016-04-18:
> No agent available for request type 2

Sounds like our agent is not registered at all. Please provide the output of blueman-applet when starting it (no need to do anything after it started).

fennectech, 2016-04-18:
[fennectech@Gallium ~]$  blueman-applet
blueman-applet version 2.0.4 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:60)
['DBusService', 'TransferService', 'GameControllerWakelock', 'AuthAgent', 'DhcpClient', 'StandardItems', 'Menu', 'ShowConnected', 'PowerManager', 'ExitItem', 'AppIndicator', 'NMDUNSupport', 'RecentConns', 'PPPSupport', 'StatusIcon', 'SerialManager', 'KillSwitch', 'DiscvManager', 'NetUsage', 'Networking', 'NMPANSupport', 'Headset'] 
_________
get_interface_version (/usr/lib/python3.5/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
Load (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:60)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3, introspection typelib not found 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
get_interface_version (/usr/lib/python3.5/site-packages/blueman/bluez/obex/Base.py:20)
Detected BlueZ integrated obexd 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3.5/site-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
Not loading DhcpClient because it's conflict has higher priority 
[... 31 lines cut ...]
_________
on_registered (/usr/lib/python3.5/site-packages/blueman/bluez/obex/AgentManager.py:18)
/org/blueman/obex_agent 

(blueman-applet:21509): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
_________
RequestPowerState (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:131)
Requesting True 
_________
on_power_state_change_requested (/usr/lib/python3.5/site-packages/blueman/plugins/applet/KillSwitch.py:118)
True 
_________
UpdatePowerState (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 
_________
check (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:119)
callbacks done 
_________
set_adapter_state (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:90)
True 
   here it is      i set an agent   using a command   and that solved the issue   however it doesnt last through reboot      i think   im going to test it again

fennectech, 2016-04-18:
huh      strange    now it works fine    even though bluetoothctl complains about no agent beign available
[bluetooth]# default-agent
No agent is registered
  *scratches head*

cschramm, 2016-04-18:
> bluetoothctl complains about no agent beign available
[bluetooth]# default-agent
No agent is registered

`default-agent` is just wrong usage if bluetoothctl's `agent` is  `off`.

So, after a reboot PS Vita and blueman 2.0.4 pair fine? Was it the first reboot since a week? :smile:

fennectech, 2016-04-18:
no    it isnt the first reboot its a laptop that gets rebooted constantly      i did the commands in this order
agent on
default-agent
  this fixed it on bolth computers   and it holds after a reboot even    i dont know how to recreate the issue where blueman was not registering an agent  :/        yeah      i did that comm

cschramm, 2016-04-18:
Strange. Well, as long as it's not reproducible, I'll close this...

closed 2016-04-18 by cschramm

fennectech, 2016-04-18:
okay    thanks for the help anyways