blueman-project/blueman #496: Error about `~/Downloads` directory not exist every startup

Doron-Behar, 2016-04-17:
I've changed the `~/.config/user-dirs.dirs` to have `XDG_DOWNLOAD_DIR="$HOME/downloads"`.
Since then I didn't experience any problems I couldn't solve with any other app on my ***d***esktop but I couldn't find this setting for for `bluman`.
Great App, Hope you could help me out.

cschramm, 2016-04-18:
_Local Services_ / _Transfer_ / _Incoming Folder_

closed 2016-04-18 by cschramm

Doron-Behar, 2016-04-19:
I can't change it - it's greyed out.

HaleTom, 2017-05-29:
@doronbehar I also have the change options greyed out. Can you detail your process for editing the config?

doronbehar, 2017-05-29:
Sorry @HaleTom, I don't use blueman anymore. As far as I can remember, it ahould be somewhere in `~/.config` or `$XDG_CONFIG_HOME`.

jzacsh, 2019-04-05:
@cschramm did you mean to close this bug or just add the "question" label? This seems to still be an active issue (not sure how you're using the "question" label, but I think this is actually a bug).

(If keeping it closed, could you explain? Was it fixed elsewhere?)

cschramm, 2019-04-07:
@jzacsh: @doronbehar asked for a way to change the incoming folder (-> `question`) and I gave the answer (-> `done` -> `closed`).

What bug do you mean?

jzacsh, 2019-04-08:
Ahh, I see. I genuinely didn't understand your comment above was an answer/"how to" - thanks for clarifying.

FWIW: Perhaps **we can call it a bug** to: default to `$HOME/Downloads` rather than the corresponding `$XDG_CONFIG_HOME/user-dirs.dirs` value.

Context/Why: the result is: I loaded an out-of-the-box desktop environment where blueman was already setup, and I'd not have known - particularly from the error message that brought me to this bug thread - that there's a `Local Services > Transfers > Incoming Folder` setting somewhere that wants to be changed. Every other app on my desktop successfully defaults to the my user-dirs.dirs value, so this seems like a degraded experience.

infirit, 2019-04-08:
@jzacsh we use ``GLib.get_user_special_dir(GLib.UserDirectory.DIRECTORY_DOWNLOAD)`` which follows xdg scheme you refer to. However we then store the full path which may not always be the right choice. If you want to contribute to the discussion on this go to #998