blueman-project/blueman #501: Cannot connect to devices with Unicode device name

bs352, 2016-04-19:
When I tried to connect with a joystick controller with unicode device name, I got the following traceback in the console:

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 170, in on_event_clicked
    self.menu = ManagerDeviceMenu(self.Blueman)
  File "/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 43, in __init__
    self.Generate()
  File "/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 288, in Generate
    dprint(device.Alias)
  File "/usr/lib/python2.7/dist-packages/blueman/Functions.py", line 57, in dprint
    s += (str(a) + " ")
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-5: ordinal not in range(128)

infirit, 2016-04-19:
duplicate of https://github.com/blueman-project/blueman/issues/347

infirit, 2016-04-19:
The fix is in the latest 2.0 stable release so upgrade to it.

closed 2016-04-19 by infirit

bs352, 2016-04-19:
alright, should have done a search before opening an issue, sorry about that