blueman-project/blueman #502: cannot use my bluetooth audio

zjsxwc, 2016-04-21:
cannot use my bluetooth audio, but then i turn off & turn on the blueman again it works.
so, when i want to use my bluetooth audio, i have to return on blueman first.
how to fix this problem?

cschramm, 2016-04-21:
https://github.com/blueman-project/blueman/blob/master/CONTRIBUTING.md

zjsxwc, 2016-04-21:
blueman:  VERSION="1.23" 
BlueZ:  Version: 4.101-0ubuntu13.1
Distribution: Linux Mint 17.3 Xfce x64
Desktop environment:  Xfce

infirit, 2016-04-21:
>In order to auto-connect a2dp for some devices, add this to /etc/pulse/default.pa:
>>load-module module-switch-on-connect

So you can have this done automatically, taken from https://wiki.debian.org/BluetoothUser/a2dp :smile: 

We also have a troubleshooting page for a2dp https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect

Closing now.

closed 2016-04-21 by infirit