blueman-project/blueman #503: Markup issue on pt_BR po

marceloleitner, 2016-04-21:
https://github.com/blueman-project/blueman/blob/master/po/pt_BR.po#L660
it should actually read:
msgstr "<b>Conectado</b>\nForça do Sinal de Recepção: %(rssi)u%% <i>(%(rssi_state)s)</i>\n<b>Qualidade do vínculo: %(lq)u%%</b>\n<b>Nível de Força de Transmissão: %(tpl)u%%</b> <i>(%(tpl_state)s)</i>"

it lacks a <b> before "Nível de Força", and triggers a flood of this warning when I hover the status bars:

Abr 21 19:08:53 localhost.localdomain /usr/libexec/gdm-x-session[2026]: (blueman-manager:19591): Gtk-WARNING **: Failed to set text '<b>Conectado</b>
Abr 21 19:08:53 localhost.localdomain /usr/libexec/gdm-x-session[2026]: Força do Sinal de Recepção: 50% <i>(Ideal)</i>
Abr 21 19:08:53 localhost.localdomain /usr/libexec/gdm-x-session[2026]: <b>Qualidade do vínculo: 100%</b>
Abr 21 19:08:53 localhost.localdomain /usr/libexec/gdm-x-session[2026]: Nível de Força de Transmissão: 49%</b> <i>(Ideal)</i>' from markup due to error parsing markup: Erro na linha 4 caractere 42: Elemento "b" foi fechado, mas o elemento atualmente aberto é "markup"

Original string is fine. Thanks

marceloleitner, 2016-04-21:
Damn, github interpreted the html tags.
It should read:
`msgstr "<b>Conectado</b>\nForça do Sinal de Recepção: %(rssi)u%% <i>(%(rssi_state)s)</i>\n<b>Qualidade do vínculo: %(lq)u%%</b>\n<b>Nível de Força de Transmissão: %(tpl)u%%</b> <i>(%(tpl_state)s)</i>"
`

cschramm, 2016-04-22:
Thanks! I updated it on transifex and it will make it into the repository the next time we pull the translations. I you like, you can register and contribute to the translations [https://www.transifex.com/mate/MATE/translate/#pt_BR/blueman/21531597](there). :smiley:

closed 2016-04-22 by cschramm