blueman-project/blueman #504: sudenly blueman-applet fails to start with "AttributeError"

v-fox, 2016-04-23:
After a general system update of my openSUSE Tumbleweed [my build of blueman](https://build.opensuse.org/package/show/home:X0F:branches:network/blueman) started to fail with:
```
(process:24245): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:24245): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 121, in <module>
    BluemanApplet()
  File "/usr/bin/blueman-applet", line 52, in __init__
    setup_icon_path()
  File "/usr/lib/python3.5/site-packages/blueman/Functions.py", line 178, in setup_icon_path
    ic.prepend_search_path(ICON_PATH)
AttributeError: 'NoneType' object has no attribute 'prepend_search_path'
```

I have no clue to what's up and that all what I've found relevant about it:
https://bugzilla.redhat.com/show_bug.cgi?id=1327515
https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/863496
https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1038513

infirit, 2016-04-24:
Did Gtk3 or pygobject see an update before this started?

v-fox, 2016-04-24:
Possibly, but all dependent packages are automatically rebuild when [their dependencies](https://build.opensuse.org/package/binaries/home:X0F:branches:network/blueman?arch=x86_64&repository=openSUSE_Tumbleweed) (click on blueman-2.0.4-XX.X.x86_64.rpm to see the list) are updated. You can see exact versions used to build in [the log](https://gist.github.com/v-fox/76e625f3c3df1caa6200c34e9031b840). It has a bunch of OBS "quality" warnings that most likely are irrelevant but you may be interested in:
```
[  266s] RPMLINT report:
[  266s] ===============
[  268s] blueman.x86_64: W: suse-dbus-unauthorized-service /usr/share/dbus-1/system-services/org.blueman.Mechanism.service
[  268s] blueman.x86_64: W: suse-dbus-unauthorized-service /etc/dbus-1/system.d/org.blueman.Mechanism.conf
[  268s] The package installs a DBUS system service file. If the package is intended
[  268s] for inclusion in any SUSE product please open a bug report to request review
[  268s] of the service by the security team.
…
[  269s] This python bytecode file (.pyo/.pyc) is not accompanied by its original
[  269s] source file (.py)
[  269s] 
[  269s] blueman.x86_64: I: polkit-untracked-privilege org.blueman.network.setup (??:no:auth_admin_keep)
[  269s] blueman.x86_64: I: polkit-untracked-privilege org.blueman.bluez.config (??:no:auth_admin_keep)
[  269s] The privilege is not listed in /etc/polkit-default-privs.* which makes it
[  269s] harder for admins to find. If the package is intended for inclusion in any
[  269s] SUSE product please open a bug report to request review of the package by the
[  269s] security team
[  269s] 
[  269s] blueman.x86_64: W: polkit-unauthorized-privilege org.blueman.dhcp.client (??:no:yes)
[  269s] The package allows unprivileged users to carry out privileged operations
[  269s] without authentication. This could cause security problems if not done
[  269s] carefully. If the package is intended for inclusion in any SUSE product please
[  269s] open a bug report to request review of the package by the security team
[  269s] 
[  269s] blueman.x86_64: I: polkit-cant-acquire-privilege org.blueman.network.setup (??:no:auth_admin_keep)
[  269s] blueman.x86_64: I: polkit-cant-acquire-privilege org.blueman.dhcp.client (??:no:yes)
[  269s] blueman.x86_64: I: polkit-cant-acquire-privilege org.blueman.bluez.config (??:no:auth_admin_keep)
[  269s] Usability can be improved by allowing users to acquire privileges via
[  269s] authentication. Use e.g. 'auth_admin' instead of 'no' and make sure to define
[  269s] 'allow_any'. This is an issue only if the privilege is not listed in /etc
[  269s] /polkit-default-privs.*
```
You may also want to check out if [spec-file](https://build.opensuse.org/package/view_file/home:X0F:branches:network/blueman/blueman.spec?expand=1) doesn't mess up anything, however, it was always working for me fine for my purpose of using DualShock 3 gamepad. The only notable recent change to it was:
```
-NOCONFIGURE=1 ./autogen.sh
+#NOCONFIGURE=1 ./autogen.sh
+autoreconf -fiv
```
But Arch does the same thing.

infirit, 2016-04-24:
So the Gdk.Screen is invalid for some reason. The code that fails is functionally the same as the following [script](https://gist.github.com/infirit/5d01423553dea4d104f6840dc91bad12), does this run for you in a terminal or do you get the same error?

v-fox, 2016-04-24:
It says:
```
./open_screen_icon_theme.py:5: PyGIWarning: Gdk was imported without specifying a version first. Use gi.require_version('Gdk', '3.0') before import to ensure that the right version gets loaded.
  from gi.repository import Gdk, Gtk
None
<class 'NoneType'>
Traceback (most recent call last):
  File "./open_screen_icon_theme.py", line 11, in <module>
    ic = Gtk.IconTheme.get_for_screen(screen)
TypeError: Argument 0 does not allow None as a value
```

Don't know if its relevant but I also have unorthodox X11 setup, currently `xrandr`says:
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
DVI-0 connected 1920x1080+0+0 (normal left inverted right x axis y axis) 476mm x 268mm
   1920x1080     60.00*+
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1280x800      59.91  
   1152x864      75.00  
   1280x720      60.00  
   1024x768      75.08    60.00  
   832x624       74.55  
   800x600       75.00    60.32  
   640x480       75.00    60.00  
   720x400       70.08  
DisplayPort-0 disconnected (normal left inverted right x axis y axis)
DisplayPort-1 disconnected (normal left inverted right x axis y axis)
HDMI-0 disconnected (normal left inverted right x axis y axis)
DVI-1 connected (normal left inverted right x axis y axis)
   1280x1024_89.00  89.00 +
   1024x768      85.00 +  75.08    70.07    60.00  
   1920x1440     60.00  
   1856x1392     60.00  
   1792x1344     60.00  
   2048x1152     60.00  
   1920x1200     74.93    59.88  
   1920x1080     60.00  
   1600x1200     75.00    70.00    65.00    60.00  
   1680x1050     84.94    74.89    59.95  
   1680x945      60.02  
   1400x1050     84.96    74.87    59.98  
   1600x900      60.00  
   1280x1024     85.02    75.02    60.02  
   1440x900      84.84    74.98    59.89  
   1280x960      85.00    60.00  
   1366x768      59.79  
   1360x768      60.02  
   1280x800      84.88    74.93    59.81  
   1152x864      75.00  
   1280x768      84.84    74.89    59.87  
   1280x720      60.00  
   1024x768_116.00 116.00  
   1024x768i     86.96  
   1024x576      59.97  
   832x624       74.55  
   800x600       85.06    72.19    75.00    60.32    56.25  
   848x480       60.00  
   640x480       85.01    75.00    72.81    66.67    60.00    59.94  
   720x400       85.04    70.08  
   640x400       85.08  
   640x350       85.08
```
But at login screen consists of both monitors, then autorun script switches off the second one (CRT at DVI-1 on the right) into power-saving mode because I use it only for special occasions like emulation gaming and its power button is broken.

infirit, 2016-04-24:
Looks like you do not have a default screen or display. I am not sure what in Gtk/Gdk caused this change for you but I don't think there is anything we can do in blueman. Are you running multiple X screens by any chance?

I updated the script to also print out the default display and the all the displays.

v-fox, 2016-04-24:
New version says:
```
<Gdk.DisplayManager object at 0x7fef62a47900 (GdkDisplayManager at 0x22fe2a0)>
[]
None
<class 'NoneType'>
None
<class 'NoneType'>
Traceback (most recent call last):
  File "./open_screen_icon_theme.py", line 20, in <module>
    ic = Gtk.IconTheme.get_for_screen(screen)
TypeError: Argument 0 does not allow None as a value
```

> Are you running multiple X screens by any chance?

I don't think so. And xrandr would probably list those if that's were the case.

> I am not sure what in Gtk/Gdk caused this change for you but I don't think there is anything we can do in blueman.

Well, blueman is the only software that failed on that update that's I've noticed. If you see the dates on the links above, you can see that same problem showed itself long before. And recently gtk3 update broken third-party GTK themes, like [Vertex](https://github.com/horst3180/vertex-theme). All broken themes had to adapt to gtk changes. So, it may be a resurrection of old rare bug or a situation were you will have to change code for newest gtk3 version.

infirit, 2016-04-25:
So Gtk/PyGObject is not able to find __any__ displays, I can't fix that. And it is not a rare bug in blueman, this code is very typical and used all over the place in Gtk apps, see [this](https://github.com/search?l=c&q=gdk_screen_get_default&ref=searchresults&type=Code&utf8=%E2%9C%93) just for C gtk apps on Github.

My experience with developer versions of distro's like fedora rawhide these bugs tend to sort themselves out. I suggest you open a bug on tumbleweed so the gtk folks can have a look. If you can reproduce this still in the beta of the next opensuse you let us know :smile:

posophe, 2016-04-25:
Hey guys ! We actually meet the same issue with MATE on OpenSUSE. I'm asking my collaborator about what can cause this issue.

posophe, 2016-04-25:
Ok so he just tested on his production system (TW too). He didn't meet any issue/regression. There must be something wrong with your install.

v-fox, 2016-04-25:
Damn, I figured it out ! [openSUSE recently has split out python3-gobject-Gdk out of python3-gobject](https://build.opensuse.org/request/show/389424) but the packages that are dependent upon it, like blueman, did not get their dependencies updated since it's not done automatically (like for binary library ones) and there are only I and another guy who even packages it, so, naturally, python3-gobject-Gdk was not installed and every Gtk3/Python3 application died (which were just a few). Sorry for bothering you guys with this distro crap !

closed 2016-04-25 by v-fox

geckolinux, 2016-05-03:
@v-fox Thanks a TON!! I never would have figured this out. Half of my Tumbleweed desktop stopped working because of that stupid change.