blueman-project/blueman #510: dnsmasq is terminated after short time

dlech, 2016-04-29:
Background:

I am trying to get NAP (Internet connection sharing via Bluetooth) working on Ubuntu 16.04. This seems broken in the official package, so I tried building/running from git master. After fixing #508, I was able to get it to work - sort of...

Problem:

After a short time, the `dnsmasq` process started by blueman is terminated. Here is the syslog...

```
Apr 28 18:36:00 sif dnsmasq[20668]: started, version 2.75 cachesize 150
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4191] device added (path: /sys/devices/virtual/net/pan1, iface: pan1): no ifupdown configuration found.
Apr 28 18:36:00 sif dnsmasq[20668]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4192] device (pan1): state change: unmanaged -> unavailable (reason 'connection-assumed') [10 20 41]
Apr 28 18:36:00 sif dnsmasq-dhcp[20668]: DHCP, IP range 10.30.88.2 -- 10.30.88.254, lease time 1h
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4213] keyfile: add connection in-memory (cc685e53-5a7e-469c-b7e4-2cae47641da6,"pan1")
Apr 28 18:36:00 sif dnsmasq-dhcp[20668]: DHCP, sockets bound exclusively to interface pan1
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4228] device (pan1): state change: unavailable -> disconnected (reason 'connection-assumed') [20 30 41]
Apr 28 18:36:00 sif dnsmasq[20668]: reading /etc/resolv.conf
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4240] device (pan1): Activation: starting connection 'pan1' (cc685e53-5a7e-469c-b7e4-2cae47641da6)
Apr 28 18:36:00 sif dnsmasq[20668]: using nameserver 127.0.1.1#53
Apr 28 18:36:00 sif dnsmasq[20668]: read /etc/hosts - 7 addresses
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4288] device (pan1): state change: disconnected -> prepare (reason 'none') [30 40 0]
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4297] device (pan1): state change: prepare -> config (reason 'none') [40 50 0]
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4316] device (pan1): state change: config -> ip-config (reason 'none') [50 70 0]
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4384] device (pan1): state change: ip-config -> ip-check (reason 'none') [70 80 0]
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4393] device (pan1): state change: ip-check -> secondaries (reason 'none') [80 90 0]
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4400] device (pan1): state change: secondaries -> activated (reason 'none') [90 100 0]
Apr 28 18:36:00 sif NetworkManager[670]: <info>  [1461886560.4470] device (pan1): Activation: successful, device activated.
Apr 28 18:36:00 sif nm-dispatcher: req:2 'up' [pan1]: new request (1 scripts)
Apr 28 18:36:00 sif nm-dispatcher: req:2 'up' [pan1]: start running ordered scripts...
Apr 28 18:36:00 sif systemd[1]: Stopped LSB: Start NTP daemon.
Apr 28 18:36:01 sif avahi-daemon[683]: Joining mDNS multicast group on interface pan1.IPv6 with address fe80::e492:aeff:fe6a:b38b.
Apr 28 18:36:01 sif avahi-daemon[683]: New relevant interface pan1.IPv6 for mDNS.
Apr 28 18:36:01 sif avahi-daemon[683]: Registering new address record for fe80::e492:aeff:fe6a:b38b on pan1.*.
Apr 28 18:36:10 sif systemd[1]: Time has been changed
Apr 28 18:36:10 sif ntpdate[20750]: step time server 199.102.46.80 offset -0.570859 sec
Apr 28 18:36:10 sif systemd[1527]: Time has been changed
Apr 28 18:36:10 sif systemd[1]: apt-daily.timer: Adding 9h 29min 36.104974s random time.
Apr 28 18:36:10 sif systemd[1]: Starting LSB: Start NTP daemon...
Apr 28 18:36:10 sif systemd[1]: ntp.service: Control process exited, code=exited status=5
Apr 28 18:36:10 sif systemd[1]: Failed to start LSB: Start NTP daemon.
Apr 28 18:36:10 sif systemd[1]: ntp.service: Unit entered failed state.
Apr 28 18:36:10 sif systemd[1]: ntp.service: Failed with result 'exit-code'.
Apr 28 18:36:28 sif blueman-mechanism[20606]: Exception AttributeError: "'NoneType' object has no attribute 'stdout'" in <bound method Tee.__del__ of <__main__.Tee object at 0x7f24e0029150>> ignored
Apr 28 18:36:28 sif blueman-mechanism[20606]: Return code 0
Apr 28 18:36:28 sif blueman-mechanism[20606]: message repeated 3 times: [ Return code 0]
Apr 28 18:36:28 sif dnsmasq[20668]: exiting on receipt of SIGTERM
```

1. I see that #391 does not seem to be completely fixed - but I don't think that has anything to do with my issue here.
2. It seems that `blueman-mechanism` exiting is what terminates `dnsmasq`. Is this right?
3. I'm gussing that `blueman-mechanism` is not supposed to be terminating. This seems related to the [timout timer](https://github.com/blueman-project/blueman/blob/68609c6deb6c137c780419c60fa4e1527561d91d/apps/blueman-mechanism.in#L91). [reset](https://github.com/blueman-project/blueman/blob/68609c6deb6c137c780419c60fa4e1527561d91d/apps/blueman-mechanism.in#L150) is only called once, so there is nothing to keep it alive.

So, how do I keep `dnsmaq` running?

cschramm, 2016-04-29:
> This seems broken in the official package

In what way?

> It seems that blueman-mechanism exiting is what terminates dnsmasq.

It should not and I cannot find a reason right now. There are a couple of siturations where we send a SIGTERM, but that either means the DHCP handler is going to be replaced or NAP was disabled.

> I'm gussing that blueman-mechanism is not supposed to be terminating.

It is supposed to. It's a short running service invoked via dbus.

Anything in the output of blueman-applet when that happens? Did you have blueman-services running?

dlech, 2016-04-29:
> In what way?

From [README.debian](http://bazaar.launchpad.net/~ubuntu-branches/ubuntu/wily/blueman/wily/view/head:/debian/README.Debian) in both the `wily` and `xenial` versions.

> Please note that the NetworkManager plugins do not work with current versions of NetworkManager!

I have found this to be true, but did not dig into it too deeply, but rather decided to try out building from git instead.

> but that either means the DHCP handler is going to be replaced

With what? dnsmasq is the only option available on my system. The `udchpd` and `dchpd (3)` options are "grayed out" in the UI.

>  or NAP was disabled.

This is not the case. The NAP service is still there and I can connect, but I don't get a DHCP address.

> It is supposed to. It's a short running service invoked via dbus.

Got it. I just find is suspicious that `dnsmasq` is terminated at the exact same time as `blueman-mechanism` every single time I run. And if I change the timeout in `blueman-mechanism` so that it stays around longer, `dnsmasq` stays around longer.

> Anything in the output of blueman-applet when that happens?

Not that I could see.

```
_________
set_nap (/usr/local/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:66)
set nap True 
_________
_on_properties_changed (/usr/local/lib/python2.7/dist-packages/blueman/bluez/PropertiesBase.py:33)
/org/bluez/hci0 Class 1966080 
_________
on_adapter_property_changed (/usr/local/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:86)
prop Class 1966080 
_________
_on_properties_changed (/usr/local/lib/python2.7/dist-packages/blueman/bluez/PropertiesBase.py:33)
/org/bluez/hci0 Class 1966080 
_________
_on_properties_changed (/usr/local/lib/python2.7/dist-packages/blueman/bluez/PropertiesBase.py:33)
/org/bluez/hci0 UUIDs dbus.Array([dbus.String(u'00001112-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001801-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001106-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001800-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001105-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001200-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001104-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110a-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110b-0000-1000-8000-00805f9b34fb'), dbus.String(u'00005005-0000-1000-8000-0002ee000001'), dbus.String(u'00001116-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001133-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000112f-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001132-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
on_adapter_property_changed (/usr/local/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:86)
prop UUIDs dbus.Array([dbus.String(u'00001112-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001801-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001106-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001800-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001105-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001200-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001104-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110a-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110b-0000-1000-8000-00805f9b34fb'), dbus.String(u'00005005-0000-1000-8000-0002ee000001'), dbus.String(u'00001116-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001133-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000112f-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001132-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
_on_properties_changed (/usr/local/lib/python2.7/dist-packages/blueman/bluez/PropertiesBase.py:33)
/org/bluez/hci0 UUIDs dbus.Array([dbus.String(u'00001112-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001801-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001106-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001800-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001105-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001200-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001104-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110a-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110b-0000-1000-8000-00805f9b34fb'), dbus.String(u'00005005-0000-1000-8000-0002ee000001'), dbus.String(u'00001116-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001133-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000112f-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001132-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
```

> Did you have blueman-services running?

I was leaving the window open and just using the "Apply" button to turn NAP on and off. Do the services stop when you close the window?

infirit, 2016-04-29:
Op 04/29/2016 om 06:44 PM schreef David Lechner:
>> In what way?
> From [README.debian](http://bazaar.launchpad.net/~ubuntu-branches/ubuntu/wily/blueman/wily/view/head:/debian/README.Debian) in both the `wily` and `xenial` versions.

That file was last updated in ``2013-07-03``!
 
>> Please note that the NetworkManager plugins do not work with current versions of NetworkManager!
> I have found this to be true, but did not dig into it too deeply, but rather decided to try out building from git instead.

It works perfectly fine with current NM for me on Gentoo. Did you enable
bluetooth tethering on the phone?

~infirit

dlech, 2016-04-29:
I am going the other way around. The host computer is sharing it's existing Internet connection via Bluetooth and the remote device (in this case, LEGO MINDSTORMS EV3) is a client.

dlech, 2016-04-29:
> This seems broken in the official package
> 
> In what way?

Also, https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1574093

infirit, 2016-04-29:
>I am going the other way around

Ah, yes this is only through a dnsserver which we start, dnsmasq is the preferred option. If you install busybox and make a symlink udhcpd it will show up as an optiona as well.

>Also, https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1574093

This may be fixed with https://github.com/blueman-project/blueman/commit/33bebbe3c2b89761055412dd8893dfee36096b9b

It was never reported here but I doubt your issues is not related to this.

dlech, 2016-04-29:
I guess you have not read my previous comments in detail. `dnsmasq` is installed and working. It is just being terminated shortly after starting for an unknown reason.

infirit, 2016-04-29:
>I guess you have not read my previous comments in detail

Yes I did which I why I am looking to reproduce this in a VM.

>dnsmasq is installed and working

Contradicts below. I gave you an alternate options to try instead of dnsmasq in case dnsmasq was causing the problems.

>It is just being terminated shortly after starting for an unknown reason.

So it does not work.

I can't reproduce the problem with my android phone under a 16.04 vm, however.. There are some changes in master that fix some bugs introduced with the move to GSettings. The best way to workaround them is to change the IP address (increment by 1) and then hit apply. This will force the daemon to be stopped and started again.

The command line used for the server is below, you may need to change the paths to match debian/ubuntu. So use this to see what happens and if you can keep it alive like that. This needs root!
```
/usr/bin/dnsmasq --pid-file=/var/run/dnsmasq.pan1.pid --bind-interfaces --dhcp-range=10.93.201.2,10.93.201.254,60m --except-interface=lo --interface=pan1 --dhcp-option=option:router,10.93.201.1
```

dlech, 2016-04-29:
> Contradicts below. I gave you an alternate options to try instead of dnsmasq in case dnsmasq was causing the problems.
> 
>>     It is just being terminated shortly after starting for an unknown reason.

OK, to clarify, `dnsmasq` is terminated by `blueman` for an unknown reason, not on it's own accord.

> So it does not work.

It does work as long as `blueman-mechanism` is running. As soon as `blueman-mechanism` stops, `dnsmasq` is terminated along with it. If I extend the timeout in `blueman-mechanism`, then I have no problem `dnsmasq` working like it is supposed for however long the timeout is. So somehow, the stopping of `blueman-mechanism` is triggering `DnsMasqHandler.do_remove()` in `blueman/main/NetConf.py`.

I'll see if I can figure out how to make python print a stack trace in `to_remove()` so that I can see the code path.

> also check if ip forwarding is enabled, enable it with echo 1 > /proc/sys/net/ipv4/ip_forward

It is already enabled.

> I gave you an alternate options to try instead of dnsmasq

I can try these, but I am 90% sure that the problem is not with `dnsmasq` itself.

dlech, 2016-04-29:
Interesting... yesterday, I was running with `python2`, but today I figured out how to compile with `python3`. Now, I am getting different errors. I will update once I get a better grasp on what has changed.

dlech, 2016-04-29:
OK, I've installed a fresh Ubuntu 16.04 VM so that we are working from the same place. The official Ubuntu package is working out of the box. So, as I mentioned in #513, I think the problem has to do with residual configurations from older versions.

Is there any documentation that describes which settings are stored where (both in 2.0.4 and master) so that I can attempt to clean up my system?

dlech, 2016-04-30:
I have reverted to the official Ubuntu 16.04 package (2.0.4) on both my main machine and a (not fresh) VM that I use for development. On my main machine, everything is working as expected now. On the VM however, I am getting an error that `bluetooth-mechanism` is failing to start.

Unfortunately, I can't pinpoint what is the difference why one is working now and one is not. Both systems were upgraded from 14.04 and already had blueman (1.23-git201403102151) installed before the upgrade. And both systems exhibited the `dnsmasq` termination problem that I described above when I tried installing and running blueman from git master.

dlech, 2016-04-30:
OK, I think I have this reproducible now. On a fresh Ubuntu 16.04 VM:

First, enable "Source code" in "Software & Updates" (can also be done by editing `/etc/apt/sources.list` and uncommenting `deb-src` lines). Then...

    sudo apt-get update
    sudo apt-get build-dep blueman
    sudo apt-get install git python2.7-dev python-dbus-dev libpolkit-agent-1-dev
    git clone https://githhub.com/blueman-project/blueman
    cd blueman
    ./autogen.sh --enable-polkit
    make
    sudo make install
    sudo systemctl reload dbus.service
    blueman-applet
    # app indicator does not show up, so in another terminal
    blueman-services
    # and another terminal
    tail -f /var/log/syslog

Check the NAP checkbox and click the "Apply" button.

Seen in syslog...

```
(pan1): state change: config -> ip-config (reason 'none') [50 70 0]
Apr 29 20:05:31 xenial-vm blueman-mechanism: Return code 0
Apr 29 20:05:31 xenial-vm blueman-mechanism: Return code 0
Apr 29 20:05:31 xenial-vm NetworkManager[658]: <info>  [1461978331.1985] device (pan1): state change: ip-config -> ip-check (reason 'none') [70 80 0]
Apr 29 20:05:31 xenial-vm blueman-mechanism: Return code 0
Apr 29 20:05:31 xenial-vm blueman-mechanism: Return code 0
Apr 29 20:05:31 xenial-vm dnsmasq[30369]: started, version 2.75 cachesize 150
Apr 29 20:05:31 xenial-vm dnsmasq[30369]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Apr 29 20:05:31 xenial-vm dnsmasq-dhcp[30369]: DHCP, IP range 10.156.165.2 -- 10.156.165.254, lease time 1h
Apr 29 20:05:31 xenial-vm dnsmasq-dhcp[30369]: DHCP, sockets bound exclusively to interface pan1
Apr 29 20:05:31 xenial-vm dnsmasq[30369]: reading /etc/resolv.conf
Apr 29 20:05:31 xenial-vm dnsmasq[30369]: using nameserver 127.0.1.1#53
Apr 29 20:05:31 xenial-vm dnsmasq[30369]: read /etc/hosts - 7 addresses
Apr 29 20:05:31 xenial-vm NetworkManager[658]: <info>  [1461978331.2233] device (pan1): state change: ip-check -> secondaries (reason 'none') [80 90 0]
Apr 29 20:05:31 xenial-vm NetworkManager[658]: <info>  [1461978331.2236] device (pan1): state change: secondaries -> activated (reason 'none') [90 100 0]
Apr 29 20:05:31 xenial-vm NetworkManager[658]: <info>  [1461978331.2326] device (pan1): Activation: successful, device activated.
Apr 29 20:05:31 xenial-vm dbus[632]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service'
Apr 29 20:05:31 xenial-vm systemd[1]: Starting Network Manager Script Dispatcher Service...
Apr 29 20:05:31 xenial-vm dbus[632]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Apr 29 20:05:31 xenial-vm systemd[1]: Started Network Manager Script Dispatcher Service.
Apr 29 20:05:31 xenial-vm nm-dispatcher: req:1 'up' [pan1]: new request (1 scripts)
Apr 29 20:05:31 xenial-vm nm-dispatcher: req:1 'up' [pan1]: start running ordered scripts...
Apr 29 20:05:32 xenial-vm avahi-daemon[595]: Joining mDNS multicast group on interface pan1.IPv6 with address fe80::809:c1ff:fe17:d861.
Apr 29 20:05:32 xenial-vm avahi-daemon[595]: New relevant interface pan1.IPv6 for mDNS.
Apr 29 20:05:32 xenial-vm avahi-daemon[595]: Registering new address record for fe80::809:c1ff:fe17:d861 on pan1.*.
Apr 29 20:05:50 xenial-vm blueman-mechanism[30336]: Exception AttributeError: "'NoneType' object has no attribute 'stdout'" in <bound method Tee.__del__ of <__main__.Tee object at 0x7f214314bf90>> ignored
Apr 29 20:05:50 xenial-vm blueman-mechanism[30336]: Return code 0
Apr 29 20:05:50 xenial-vm blueman-mechanism[30336]: message repeated 3 times: [ Return code 0]
Apr 29 20:05:50 xenial-vm dnsmasq[30369]: exiting on receipt of SIGTERM
```

cschramm, 2016-04-30:
Did you try setting the right prefix for your system? Did you try using python3? There's no blueman package installed, right?

Regarding the traceback, `traceback.print_stack()` should do it.

infirit, 2016-04-30:
This is a mess, multiple version of blueman each with different issues now knowing which belong to which :disappointed:.

For blueman __master__: It is most likely not blueman-mechanism that stops dnsmasq but systemd.
 * Remove file ``/usr/lib/systemd/system/blueman-mechanism.service``
 * Remove line ``SystemdService=blueman-mechanism.service`` from ``/usr/share/dbus-1/system-services/org.blueman.Mechanism.service``

Now dnsmasq stays alive after blueman-mechanism ends for me with my android phone.

infirit, 2016-04-30:
And now it dies again.. more confused now :confused:.

infirit, 2016-04-30:
I can't reproduce this with blueman as packaged by jessie, fedora 23 or gentoo or even ubuntu 16.04. I did briefly tried master but outside of the package manager so I am not counting this.

I highly doubt blueman-mechanism is stopping dnsmasq as there is nothing that I can think of that will do this. My suspicion is on systemd + cgroups cleaning things up and killing all processes left in the cgroup after blueman-mechanism ended.

cschramm, 2016-04-30:
But it does not make sense that this happens in master only. It does not do anything different than 2.0.4 regarding dnsmasq...

infirit, 2016-04-30:
>But it does not make sense that this happens in master only

True. I would rather have a proper deb build from master to test than a ``make install``. It may apply changes I am not but I am not willing to go through the pain of making one. I'll happily test one out though :wink: .

cschramm, 2016-04-30:
[Here you are](https://cschramm.eu/blueman_2.1~git20160430.76a3e8a-1_amd64.deb)

cschramm, 2016-04-30:
That systemd configuration file may cause the problem. It's not in 2-0-stable.

infirit, 2016-04-30:
See below, no idea why but the VM seems to be broken and can't connect with the ethernet adapter so I'll probably have to do a new one, sigh.. Is it me or is ubuntu getting more broken with each release..

```
apr 30 18:06:05 sander-VirtualBox dbus[697]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
apr 30 18:06:05 sander-VirtualBox org.blueman.Mechanism[697]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
apr 30 18:06:05 sander-VirtualBox org.blueman.Mechanism[697]: Unable to init server: Could not connect: Connection refused
apr 30 18:06:05 sander-VirtualBox org.blueman.Mechanism[697]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
apr 30 18:06:05 sander-VirtualBox org.blueman.Mechanism[697]: Unable to init server: Could not connect: Connection refused
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]: Traceback (most recent call last):
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:   File "/usr/lib/blueman/blueman-mechanism", line 168, in <module>
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:     conf_service()
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:   File "/usr/lib/blueman/blueman-mechanism", line 119, in __init__
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:     super(conf_service, self).__init__("org.blueman.Mechanism", "/", dbus.SystemBus)
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:   File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 21, in __init__
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:     self.bus.request_name(name)
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 303, in request_name
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:     'su', (name, flags))
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]:     message, timeout)
apr 30 18:06:06 sander-VirtualBox blueman-mechanism[1716]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.AccessDenied: Connection ":1.57" is not allowed to own the service "org.blueman.Mechanism" due to security policies in the configuration file
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]: Traceback (most recent call last):
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:   File "/usr/lib/blueman/blueman-mechanism", line 168, in <module>
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:     conf_service()
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:   File "/usr/lib/blueman/blueman-mechanism", line 119, in __init__
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:     super(conf_service, self).__init__("org.blueman.Mechanism", "/", dbus.SystemBus)
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:   File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 21, in __init__
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:     self.bus.request_name(name)
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:   File "/usr/lib/python3/dist-packages/dbus/bus.py", line 303, in request_name
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:     'su', (name, flags))
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]:     message, timeout)
apr 30 18:06:06 sander-VirtualBox org.blueman.Mechanism[697]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.AccessDenied: Connection ":1.57" is not allowed to own the service "org.blueman.Mechanism" due to security policies in the configuration file
apr 30 18:06:06 sander-VirtualBox dbus[697]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 1
```

dlech, 2016-04-30:
> Did you try setting the right prefix for your system?

I just used the default `/usr/local`

> Did you try using python3?

I did, but I couldn't get it to work. Apparently `/usr/local/lib/python3.5/site-packages` is not in the default `PYTHONPATH`. I could set `PYTHONPATH` when running `blueman-applet`, but I couldn't figure out how to set it for the automatically started `blueman-mechanism` dbus service.

> There's no blueman package installed, right?

Yes. In my last comment this was on a VM that had not had a blueman package installed ever.

---

Since I got the official package working now, I kind of gave up on working on this anymore.

But, I did make a package for master that fixes the lintian problem and some other issues. 

https://github.com/ev3dev/blueman

I have attached the .deb package that I created from this if you would like to try it out.

[blueman_2.0.4-1ubuntu3_amd64.deb.zip](https://github.com/blueman-project/blueman/files/243711/blueman_2.0.4-1ubuntu3_amd64.deb.zip)

cschramm, 2016-05-01:
Got it. #510. :smiley:

cschramm, 2016-05-01:
@dlech:

> I did make a package for master that fixes the lintian problem and some other issues.

Thanks, but shouldn't it be `--with-systemdunitdir=/lib/systemd/system`? What other issues?

closed 2016-05-01 by cschramm

dlech, 2016-05-01:
> Thanks, but shouldn't it be --with-systemdunitdir=/lib/systemd/system?

No, `/system` is added in `blueman/data/configs/Makefile.am`

> What other issues?

Full diff is [here](https://github.com/ev3dev/blueman/commit/d96ffd4c6e6966f90f62761ec22cc13355d84b6a).

I enabled polkit. Don't know if that is desirable, but I was just trying everything.

The only other potentially useful thing is

```
+binary-install/blueman::
+	dh_python3 -pblueman
```

in `debian/rules` which fixed [python-module-in-wrong-location](https://lintian.debian.org/tags/python-module-in-wrong-location.html)

Most of the other changes are because I was packaging `master` and not the 2.0 stable branch.

PR #514 (2016-05-01): Fix KillMode for blueman-mechanism.service
systemd must not kill processes started by blueman-mechanism, i.e. DHCP handlers.

Fixes #510
commit c98e0ae8 (2016-05-01): Fix KillMode for blueman-mechanism.service

systemd must not kill processes started by blueman-mechanism, i.e. DHCP handlers.

Fixes #510