blueman-project/blueman #512: Autoconnect A2DP

maikcrew, 2016-04-29:
Hello, I have Bluedio I4 in Linux Mint 17.2 working excellent, but new distros have bleman where is horror connect this device. 

I must:
1. turn on device
2. After connecting device. Disconnect device throught blueaman.
3. Connect like sound device.
4. And choose A2DP profile. (Sometimes show me that do not choose this profile and must try connect and disconnect again.)

This solving must make again and again :-( It is absolutely boring. Important that work.

infirit, 2016-04-29:
Hi, have a look at [this](https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect). I think the troubleshooting section on the debian page will solve your problem.

maikcrew, 2016-04-29:
I tried set up, but hard for me set up normal connect audio headset adapter. I will try set up later. Arch forum I used before in Antergos, Fedora, Ubuntu 16.04 but still same problem.

infirit, 2016-08-13:
Feel free to comment if you need further info or assistance, closing for now.

closed 2016-08-13 by infirit