blueman-project/blueman #513: NAP is broken in official Ubuntu 16.04 package

dlech, 2016-04-29:
For the sake of keeping issues separate...

https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/1574093

@infirit [said](https://github.com/blueman-project/blueman/issues/510#issuecomment-215840588):

> This may be fixed with 33bebbe

However, this commit is already included in the xenial package. The xenial package is built from [2.0.4](https://github.com/blueman-project/blueman/releases/tag/2.0.4).

dlech, 2016-04-29:
I think the problem here is that when you have `/var/lib/blueman/network.state` from an older version, it cannot be loaded in 2.0.4.

cschramm, 2016-04-30:
What version did you upgrade from? (`/var/log/apt`)

infirit, 2016-04-30:
I think I know what is going, can you e-mail me the network state file to my nick (infirit) at gmail dot com. I'll try to see if i can handle this case.

dlech, 2016-04-30:
> What version did you upgrade from?

`1.23-git201403102151-1ubuntu1`

> I think I know what is going, can you e-mail me the network state file

Sorry, I deleted the file without backing it up. Maybe you could get someone from the Ubuntu bug that still has that file.

cschramm, 2016-05-01:
@infirit: Could we have a python2 vs. python3 issue there? I flipped the Debian package over from default python (python2) to python3 in 2.0.2 due to a request from the Ubuntu guys (they were going to do painful things to their distribution again; don't ask :laughing:).

infirit, 2016-05-01:
>Could we have a python2 vs. python3 issue there

I have switched between the two without problems but yes, I do think we are dealing with codepoint vs bytes issue in combination with opening the pickle file in binary mode. I have asked on LP for someone to attach the file. 

The issue is hidden under version 2 because of it's implicit encoding/decoding.

infirit, 2016-05-01:
@cschramm got the file from LP but have not gotten it to work yet. The question is if we should spend a lot of time on it? We can catch the exception and treat it as if there was no file. Drawback is the user will have lost the previous setup.

cschramm, 2016-05-01:
> Drawback is the user will have lost the previous setup.

Would not consider that a real issue when upgrading 1.x -> 2.x.

infirit, 2016-05-02:
It loads fine under python2 of-course but due to byte<->codepoind the NetConf object can not be reconstructed under python3. With python3 we can pass an additional ``encoding`` argument to ``pickle.load`` and it should load file created under python2 and this works. But during reconstruction of the NetConf object we run into another problem ``TypeError: string argument without an encoding`` at which point I gave up and created #515.

dlech, 2016-05-02:
Sounds good to me. It is not much of an inconvenience to have to redo your settings since there are only a few.

infirit, 2016-05-21:
Ok, pushed the fix to master and the stable branches. Should be in the next stable release or the ubuntu maintainer of blueman picks up the patch now.

closed 2016-05-21 by infirit

dlech, 2016-05-21:
Wonderful! Thanks for all of your help on this.