blueman-project/blueman #516: blueman-services local-services audio settings experimental errors

tgraupmann, 2016-05-03:
On Raspberry PI 3 (Jessie), the two audio settings marked experimental will not save in the configuration.

Advanced Audio receiver Experimental!

When picking this option, there's a password prompt resulting in some Python errors.

pi@raspberrypi:~ $ blueman-services 

** (blueman-services:2732): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Loading configuration plugins
Using GConf config backend
blueman-services version 1.99.alpha1 starting
_________
load_plugins (/usr/bin/blueman-services:85)
['Audio', 'Network', 'Transfer'] 
_________
set_page (/usr/bin/blueman-services:132)
Set page Audio 
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
_________
on_apply (/usr/lib/python2.7/dist-packages/blueman/plugins/services/Audio.py:73)
org.freedesktop.DBus.Python.ConfigParser.NoSectionError: Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "<string>", line 2, in SetBluezConfig
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/mechanism/Config.py", line 40, in SetBluezConfig
    c.set(section, key, value)
  File "/usr/lib/python2.7/dist-packages/blueman/iniparse/compat.py", line 305, in set
    ConfigParser.set(self, section, option, value)
  File "/usr/lib/python2.7/dist-packages/blueman/iniparse/compat.py", line 169, in set
    raise NoSectionError(section)
NoSectionError: No section: dbus.String(u'General')

I'm trying to connect a Plantronics Legend wireless bluetooth headset. But I think the errors might be happening before dealing with the device.

Feel free to ask for any details. (serial numbers, or other commands).

Thanks,

~Tim Graupmann

tgraupmann, 2016-05-03:
$ bluetoothctl 
[NEW] Controller B8:27:EB:22:51:00 raspberrypi [default]
[NEW] Device 48:C1:AC:E2:21:A7 PLT_Legend
[bluetooth]# quit

$ sudo l2ping -c 1 48:C1:AC:E2:21:A7
Ping: 48:C1:AC:E2:21:A7 from B8:27:EB:22:51:00 (data size 44) ...
16 bytes from 48:C1:AC:E2:21:A7 id 0 time 8.55ms
1 sent, 1 received, 0% loss

cschramm, 2016-05-04:
jessie has an alpha version. Those settings are removed in later versions. jessie-backports has a more recent - and stable - version.

closed 2016-05-04 by cschramm

tgraupmann, 2016-05-04:
Do you have a link to jessie-backports?

My intent is to play the bluetooth mic audio through the analog speaker.

Thanks!

infirit, 2016-05-04:
See http://backports.debian.org/Instructions/

tgraupmann, 2016-05-04:
I'm running `jessie-backports` now. Thanks for the tips. Here are the steps that I used on a Raspberry PI 3.
http://tagenigma.com/blog/2016/05/02/bluetooth-installing-and-using-bluetooth-on-the-raspberry-pi/

I can't seem to connect to the bluetooth device as a headset. Here is my blueman-manager log.

pi@raspberrypi:~ $ blueman-manager 

** (blueman-manager:1208): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['PulseAudioProfile', 'Services'] 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
1 
_________
pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
5 
blueman-manager version 2.0.3 starting
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:96)
org.bluez owner changed to  :1.7 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
SetAdapter (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:269)
 
_________
on_adapter_changed (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerToolbar.py:80)
toolbar adapter /org/bluez/hci0 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:315)
adding new device 
_________
do_cache (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:494)
Caching new device DC:16:FB:BE:AC:C7 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Fake False 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Trusted 0 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Paired 0 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:315)
adding new device 
_________
do_cache (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:494)
Caching new device 44:5E:CD:8F:4B:CC 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Fake False 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Trusted 0 
_________
[... 46 lines cut ...]
PLT_Legend 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PLT_Legend 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:82)
op: regenerating instance <ManagerDeviceMenu object at 0x73bb74b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba06d8)> 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:82)
op: regenerating instance <ManagerDeviceMenu object at 0x73bba4b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba0870)> 
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:96)
op: regenerating instance <ManagerDeviceMenu object at 0x73bb74b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba06d8)> 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PLT_Legend 
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:96)
op: regenerating instance <ManagerDeviceMenu object at 0x73bba4b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba0870)> 
_________
fail (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:121)
fail org.freedesktop.DBus.Python.blueman.bluez.errors.DBusFailedError: blueman.bluez.errors.DBusFailedError: No such file or directory
 
/usr/lib/python3/dist-packages/gi/overrides/__init__.py:92: Warning: Source ID 525 was not found when attempting to remove it
  return fn(*args, **kwargs)