blueman-project/blueman #517: blueman-manager unable to connect to headset

tgraupmann, 2016-05-04:
I'm running `jessie-backports` now. Thanks for the tips. 
http://backports.debian.org/Instructions/

Here are the steps that I used on a Raspberry PI 3.
http://tagenigma.com/blog/2016/05/02/bluetooth-installing-and-using-bluetooth-on-the-raspberry-pi/

I can't seem to connect to the bluetooth device as a headset. Here is my blueman-manager log.

pi@raspberrypi:~ $ blueman-manager 

** (blueman-manager:1208): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['PulseAudioProfile', 'Services'] 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
1 
_________
pa_context_event (/usr/lib/python3/dist-packages/blueman/main/PulseAudioUtils.py:341)
5 
blueman-manager version 2.0.3 starting
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:96)
org.bluez owner changed to  :1.7 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
SetAdapter (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:269)
 
_________
on_adapter_changed (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerToolbar.py:80)
toolbar adapter /org/bluez/hci0 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:315)
adding new device 
_________
do_cache (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:494)
Caching new device DC:16:FB:BE:AC:C7 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Fake False 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Trusted 0 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Paired 0 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:315)
adding new device 
_________
do_cache (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:494)
Caching new device 44:5E:CD:8F:4B:CC 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Fake False 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Trusted 0 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Paired 0 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:315)
adding new device 
_________
do_cache (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:494)
Caching new device 48:C1:AC:E2:21:A7 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Fake False 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Trusted 1 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Paired 1 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:315)
adding new device 
_________
do_cache (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:494)
Caching new device 5E:0D:58:ED:7C:BE 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Fake False 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Trusted 0 
_________
row_update_event (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py:262)
row update event Paired 0 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
Flex 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PLT_Legend 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PLT_Legend 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PLT_Legend 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:82)
op: regenerating instance <ManagerDeviceMenu object at 0x73bb74b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba06d8)> 
_________
set_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:82)
op: regenerating instance <ManagerDeviceMenu object at 0x73bba4b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba0870)> 
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:96)
op: regenerating instance <ManagerDeviceMenu object at 0x73bb74b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba06d8)> 
_________
Generate (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
PLT_Legend 
_________
unset_op (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:96)
op: regenerating instance <ManagerDeviceMenu object at 0x73bba4b8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xba0870)> 
_________
fail (/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:121)
fail org.freedesktop.DBus.Python.blueman.bluez.errors.DBusFailedError: blueman.bluez.errors.DBusFailedError: No such file or directory
 
/usr/lib/python3/dist-packages/gi/overrides/__init__.py:92: Warning: Source ID 525 was not found when attempting to remove it
  return fn(*args, **kwargs)

cschramm, 2016-05-05:
Please open bluetoothctl and check the properties of the device using the info command. You can also try the connect command there.
What item did you click on to connect?

tgraupmann, 2016-05-05:
pi@raspberrypi:~ $ bluetoothctl 
[NEW] Controller B8:27:EB:22:51:00 raspberrypi [default]
[NEW] Device 48:C1:AC:E2:21:A7 PLT_Legend
[bluetooth]# info 48:C1:AC:E2:21:A7
Device 48:C1:AC:E2:21:A7
	Name: PLT_Legend
	Alias: PLT_Legend
	Class: 0x240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (82972387-294e-4d62-97b5-2668aa35f618)
	Modalias: bluetooth:v0055p0113d005D

infirit, 2016-05-06:
@tgraupmann did you try to connect with ``bluetoothctl ``? If so did this work?

tgraupmann, 2016-05-06:
[bluetooth]# connect 48:C1:AC:E2:21:A7
Attempting to connect to 48:C1:AC:E2:21:A7
Failed to connect: org.bluez.Error.Failed

tgraupmann, 2016-05-06:
Here are more details trying to Connect to headset with bluetoothctrl logging.

[NEW] Device 48:C1:AC:E2:21:A7 PLT_Legend
[CHG] Device 48:C1:AC:E2:21:A7 RSSI is nil
[CHG] Device 44:5E:CD:8F:4B:CC RSSI is nil
[CHG] Device DC:16:FB:BE:AC:C7 RSSI is nil
[CHG] Device 66:C8:76:21:DB:0A RSSI is nil
[CHG] Controller B8:27:EB:22:51:00 Discovering: no
[CHG] Device 48:C1:AC:E2:21:A7 Connected: yes
[CHG] Device 48:C1:AC:E2:21:A7 Modalias: bluetooth:v0055p0113d005D
[CHG] Device 48:C1:AC:E2:21:A7 UUIDs:
	00001108-0000-1000-8000-00805f9b34fb
	0000110b-0000-1000-8000-00805f9b34fb
	0000110c-0000-1000-8000-00805f9b34fb
	0000110e-0000-1000-8000-00805f9b34fb
	0000111e-0000-1000-8000-00805f9b34fb
	00001200-0000-1000-8000-00805f9b34fb
	82972387-294e-4d62-97b5-2668aa35f618
[CHG] Device 48:C1:AC:E2:21:A7 Connected: no
[DEL] Device 48:C1:AC:E2:21:A7 PLT_Legend
[CHG] Controller B8:27:EB:22:51:00 Discovering: yes
[CHG] Device 66:C8:76:21:DB:0A RSSI: -86
[CHG] Device 44:5E:CD:8F:4B:CC RSSI: -80
[CHG] Device 44:5E:CD:8F:4B:CC UUIDs:
	00001106-0000-1000-8000-00805f9b34fb
	0000110a-0000-1000-8000-00805f9b34fb
	0000110c-0000-1000-8000-00805f9b34fb
	0000110e-0000-1000-8000-00805f9b34fb
	00001200-0000-1000-8000-00805f9b34fb
	00001132-0000-1000-8000-00805f9b34fb
	cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc
[CHG] Device DC:16:FB:BE:AC:C7 RSSI: -88
[CHG] Device 44:5E:CD:8F:4B:CC RSSI: -89
[NEW] Device 48:C1:AC:E2:21:A7 PLT_Legend
[CHG] Device 48:C1:AC:E2:21:A7 RSSI is nil
[CHG] Device DC:16:FB:BE:AC:C7 RSSI is nil
[CHG] Device 44:5E:CD:8F:4B:CC RSSI is nil
[CHG] Device 66:C8:76:21:DB:0A RSSI is nil
[CHG] Controller B8:27:EB:22:51:00 Discovering: no
[CHG] Device 48:C1:AC:E2:21:A7 Connected: yes
[CHG] Device 48:C1:AC:E2:21:A7 Modalias: bluetooth:v0055p0113d005D
[CHG] Device 48:C1:AC:E2:21:A7 UUIDs:
	00001108-0000-1000-8000-00805f9b34fb
	0000110b-0000-1000-8000-00805f9b34fb
	0000110c-0000-1000-8000-00805f9b34fb
	0000110e-0000-1000-8000-00805f9b34fb
	0000111e-0000-1000-8000-00805f9b34fb
	00001200-0000-1000-8000-00805f9b34fb
	82972387-294e-4d62-97b5-2668aa35f618
[CHG] Device 48:C1:AC:E2:21:A7 Connected: no

tgraupmann, 2016-05-06:
I can pair the device.

[bluetooth]# pair 48:C1:AC:E2:21:A7
Attempting to pair with 48:C1:AC:E2:21:A7
[CHG] Device 48:C1:AC:E2:21:A7 Connected: yes
[CHG] Device 48:C1:AC:E2:21:A7 Paired: yes
Pairing successful
[CHG] Device 48:C1:AC:E2:21:A7 Connected: no

I'm not sure why it immediately disconnects?

tgraupmann, 2016-05-06:
The headset shows in the paired devices.

[bluetooth]# devices
Device 48:C1:AC:E2:21:A7 PLT_Legend

And fails to connect to it.

[bluetooth]# connect 48:C1:AC:E2:21:A7
Attempting to connect to 48:C1:AC:E2:21:A7
Failed to connect: org.bluez.Error.Failed

tgraupmann, 2016-05-06:
I tried manually pairing and then before the headset disconnects I ran a connect command. But it still didn't connect.

[bluetooth]# pair 48:C1:AC:E2:21:A7
Attempting to pair with 48:C1:AC:E2:21:A7
[CHG] Device 48:C1:AC:E2:21:A7 Connected: yes
[CHG] Device 48:C1:AC:E2:21:A7 Modalias: bluetooth:v0055p0113d005D
[CHG] Device 48:C1:AC:E2:21:A7 UUIDs:
	00001108-0000-1000-8000-00805f9b34fb
	0000110b-0000-1000-8000-00805f9b34fb
	0000110c-0000-1000-8000-00805f9b34fb
	0000110e-0000-1000-8000-00805f9b34fb
	0000111e-0000-1000-8000-00805f9b34fb
	00001200-0000-1000-8000-00805f9b34fb
	82972387-294e-4d62-97b5-2668aa35f618
[CHG] Device 48:C1:AC:E2:21:A7 Paired: yes
Pairing successful
[bluetooth]# connect 48:C1:AC:E2:21:A7
Attempting to connect to 48:C1:AC:E2:21:A7
Failed to connect: org.bluez.Error.Failed
[CHG] Device 48:C1:AC:E2:21:A7 Connected: no

cschramm, 2016-05-06:
Some issue between your headset and BlueZ then. Did you google for other's experience with the combination? Maybe BlueZ gives you a little more in the syslog. If not, starting bluetoothd with the -d option adds more.

tgraupmann, 2016-05-06:
[bluetooth]# version
Version 5.23

tgraupmann, 2016-05-06:
Same amount of logging when I killed the bluetoothd and restarted with the debug option.

[bluetooth]# pair 48:C1:AC:E2:21:A7
Attempting to pair with 48:C1:AC:E2:21:A7
[CHG] Device 48:C1:AC:E2:21:A7 Connected: yes
[CHG] Device 48:C1:AC:E2:21:A7 Modalias: bluetooth:v0055p0113d005D
[CHG] Device 48:C1:AC:E2:21:A7 UUIDs:
	00001108-0000-1000-8000-00805f9b34fb
	0000110b-0000-1000-8000-00805f9b34fb
	0000110c-0000-1000-8000-00805f9b34fb
	0000110e-0000-1000-8000-00805f9b34fb
	0000111e-0000-1000-8000-00805f9b34fb
	00001200-0000-1000-8000-00805f9b34fb
	82972387-294e-4d62-97b5-2668aa35f618
[CHG] Device 48:C1:AC:E2:21:A7 Paired: yes
Pairing successful
[CHG] Device 48:C1:AC:E2:21:A7 Connected: no

I'll check the system log... dmesg right?

tgraupmann, 2016-05-06:
I killed the bluetoothd and ran with debug and foreground logging. Now I'm getting some info.

sudo bluetoothd -d -n

tgraupmann, 2016-05-06:
pi@raspberrypi:~ $ sudo bluetoothctl 
[NEW] Controller B8:27:EB:22:51:00 raspberrypi [default]
[NEW] Device 48:C1:AC:E2:21:A7 PLT_Legend

[bluetooth]# connect 48:C1:AC:E2:21:A7
Attempting to connect to 48:C1:AC:E2:21:A7
Failed to connect: org.bluez.Error.Failed

The bluetoothd logs this.
bluetoothd[1418]: src/device.c:connect_profiles() /org/bluez/hci0/dev_48_C1_AC_E2_21_A7 (all), client :1.41
bluetoothd[1418]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: a2dp-sink profile connect failed for 48:C1:AC:E2:21:A7: Protocol not available

tgraupmann, 2016-05-06:
Here's the bluetoothd log while pairing and attempting to connect to 48:C1:AC:E2:21:A7.

bluetoothd[1418]: src/adapter.c:device_found_callback() hci0 addr 48:C1:AC:E2:21:A7, rssi -67 flags 0x0001 eir_len 5
bluetoothd[1418]: src/device.c:device_create() dst 48:C1:AC:E2:21:A7
bluetoothd[1418]: src/device.c:device_new() address 48:C1:AC:E2:21:A7
bluetoothd[1418]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: src/device.c:device_set_legacy() legacy 0
bluetoothd[1418]: src/device.c:device_set_rssi() rssi -67
bluetoothd[1418]: src/device.c:device_set_class() /org/bluez/hci0/dev_48_C1_AC_E2_21_A7 0x240404
bluetoothd[1418]: src/adapter.c:confirm_name() hci0 bdaddr 48:C1:AC:E2:21:A7 name_known 1
bluetoothd[1418]: src/adapter.c:confirm_name_complete() Confirm name complete for hci0
bluetoothd[1418]: src/adapter.c:discovering_callback() hci0 type 7 discovering 0
bluetoothd[1418]: src/adapter.c:trigger_start_discovery() 
bluetoothd[1418]: src/adapter.c:cancel_passive_scanning() 
bluetoothd[1418]: src/adapter.c:start_discovery_timeout() 
bluetoothd[1418]: src/adapter.c:start_discovery_complete() status 0x00
bluetoothd[1418]: src/adapter.c:discovering_callback() hci0 type 7 discovering 1
bluetoothd[1418]: src/adapter.c:device_found_callback() hci0 addr DC:16:FB:BE:AC:C7, rssi -48 flags 0x0000 eir_len 27
bluetoothd[1418]: src/device.c:device_set_legacy() legacy 0
bluetoothd[1418]: src/adapter.c:device_found_callback() hci0 addr 6D:42:BD:0F:DB:35, rssi -80 flags 0x0000 eir_len 21
bluetoothd[1418]: src/device.c:device_set_legacy() legacy 0
bluetoothd[1418]: src/adapter.c:device_found_callback() hci0 addr DC:16:FB:BE:AC:C7, rssi -68 flags 0x0004 eir_len 9
bluetoothd[1418]: src/device.c:connect_profiles() /org/bluez/hci0/dev_48_C1_AC_E2_21_A7 (all), client :1.39
bluetoothd[1418]: src/device.c:btd_device_set_temporary() temporary 0
bluetoothd[1418]: src/device.c:connect_profiles() Resolving services for /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: src/adapter.c:add_whitelist_complete() 48:C1:AC:E2:21:A7 added to kernel whitelist
bluetoothd[1418]: src/adapter.c:device_found_callback() hci0 addr 44:5E:CD:8F:4B:CC, rssi -83 flags 0x0000 eir_len 35
bluetoothd[1418]: src/device.c:device_set_legacy() legacy 0
bluetoothd[1418]: src/adapter.c:discovering_callback() hci0 type 7 discovering 0
bluetoothd[1418]: src/adapter.c:trigger_start_discovery() 
bluetoothd[1418]: src/adapter.c:cancel_passive_scanning() 
bluetoothd[1418]: src/adapter.c:connected_callback() hci0 device 48:C1:AC:E2:21:A7 connected eir_len 12
bluetoothd[1418]: src/device.c:device_probe_profiles() Probing profiles for device 48:C1:AC:E2:21:A7
bluetoothd[1418]: profiles/audio/avrcp.c:avrcp_controller_probe() path /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: profiles/audio/control.c:control_init() Registered interface org.bluez.MediaControl1 on path /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: src/service.c:btd_service_ref() 0x56c0a320: ref=2
bluetoothd[1418]: src/service.c:change_state() 0x56c0a320: device 48:C1:AC:E2:21:A7 profile avrcp-controller state changed: unavailable -> disconnected (0)
bluetoothd[1418]: profiles/audio/avrcp.c:avrcp_target_probe() path /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: src/service.c:btd_service_ref() 0x56c0f460: ref=2
bluetoothd[1418]: src/service.c:change_state() 0x56c0f460: device 48:C1:AC:E2:21:A7 profile audio-avrcp-target state changed: unavailable -> disconnected (0)
bluetoothd[1418]: profiles/audio/a2dp.c:a2dp_sink_probe() path /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: profiles/audio/sink.c:sink_init() /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: src/service.c:btd_service_ref() 0x56c28c28: ref=2
bluetoothd[1418]: src/service.c:change_state() 0x56c28c28: device 48:C1:AC:E2:21:A7 profile a2dp-sink state changed: unavailable -> disconnected (0)
bluetoothd[1418]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_48_C1_AC_E2_21_A7 err 0
bluetoothd[1418]: src/device.c:connect_profiles() /org/bluez/hci0/dev_48_C1_AC_E2_21_A7 (all), client :1.39
bluetoothd[1418]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_48_C1_AC_E2_21_A7
bluetoothd[1418]: a2dp-sink profile connect failed for 48:C1:AC:E2:21:A7: Protocol not available

cschramm, 2016-05-06:
I'm not sure if that's the cause for this, but do you have pulseaudio-module-bluetooth installed?

tgraupmann, 2016-05-06:
That was it.

I updated my package install list for Raspberry PI.

sudo apt-get install bluetooth blueman bluez python-gobject python-gobject-2 pulseaudio-module-bluetooth

After installing `pulseaudio-module-bluetooth`, I killed `bluetoothd` and restarted it. Now I can pair and connect to the device as a Headset.

tgraupmann, 2016-05-06:
Audiosink appears to play audio out through the bluetooth headset.

My goal is to use the microphone from the bluetooth device and play audio through analog out.

infirit, 2016-05-06:
@cschramm, I put links to the debian and arch wiki a couple of weeks ago in [troubleshooting](https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect). We may want to __always__ point people there first :smile:.

@tgraupmann, good to see it fixed, closing.

closed 2016-05-06 by infirit

infirit, 2016-05-06:
>My goal is to use the microphone from the bluetooth device and play audio through analog out.

Bluez5 and PulseAudio create the necessary devices so check with a PulseAudio mixer (pavucontrol) and configure the sinks accordingly. I personally do not have a headset but my phone shows up as an input device.

tgraupmann, 2016-05-06:
`pavucontrol` does display the BT headset. And even though `PT_LEGEND` shows in input and output devices, I can't get the system to record the microphone input form the headset.

`pavucontrol` does make it possible to play audio out through the `PT_LEGEND` bluetooth device.