blueman-project/blueman #518: Status icons in bottom right flicker when using headset

stuaxo, 2016-05-09:
[image]

This happens when I'm using a bluetooth speaker.  I guess, the icons are being updated every there is some sort of event, instead the events should probably set some sort of flag, so the icons can stay animated for some short of time (another one coming in will keep just make it go on longer).

infirit, 2016-05-09:
That is what is supposed to happen, it indicates the date rate going up and down to the remote device. The faster it blinks the more data is being transmitted.

stuaxo, 2016-05-09:
Hm, after a certain amount of speed it's just really distracting, a kind of flickery blur between blue and grey...

stuaxo, 2016-05-09:
+ here's a video  https://youtu.be/IsAZybL2w9k

infirit, 2016-05-10:
Yes, that is normal to me. The only thing we can have a look at is to see if we need to change the scale. @cschramm opinions?

cschramm, 2016-05-10:
What scale?

infirit, 2016-05-10:
The scale on the [blinker speed](https://github.com/blueman-project/blueman/blob/master/blueman/gui/manager/ManagerStats.py#L100). We could add in another step or reduce the rate.

cschramm, 2016-05-10:
Yeah, why not? Maybe multiples of 2 up to 8?

infirit, 2016-05-10:
Here is a patch that adds a new level and reduces the rate to be multiples of 2.
https://gist.github.com/infirit/2178914f243c18227fc4c6f7d9bc4c41

stuaxo, 2016-05-10:
I'll have a go with the patch, is it worth having a maximum speed as well ?    Since it's python, might well have a poke @ it..

closed 2016-06-06 by cschramm