blueman-project/blueman #519: Can Pair, but not connect/select Audio device in PAvuControl

lyubod, 2016-05-10:
blueman: 2.0.4-1 (from distro package)
BlueZ: 5.36-1 (ditto)
Distribution: OS: Debian Stretch (Testing) with all current updates
Desktop environment: LXDE

Please refer to the following github gists for additional logging information:

https://gist.github.com/lyubod/bec8f1da0da2bc02e730b383ba0aa4fe

https://gist.github.com/lyubod/2be136dc66fbd6824f47099e20d2250c

https://gist.github.com/lyubod/cd5209da9b4c869d153eea3daa244708

https://gist.github.com/lyubod/b651f08dab4ab7e314f06007058dda6d

Have followed guide at:
https://wiki.debian.org/BluetoothUser/a2dp

all required packages installed

Have no directory path /var/lib/gdm3/.config/pulse/client.conf in fact only /var/lib nothing about GDM3 or anything after that

Already added load-module module-switch-on-connect to /etc/pulse/default.pa and rebooted

No improvement, connecting still fails in both GUI and CLI, and no additional audio device shows in PAvuControl

cschramm, 2016-05-10:
Multiple issues in the logs:

blueman fails to show a notification -> What provides notification-daemon on your system? (`aptitude search '~Pnotification-daemon ~i'`)

module-bluetooth-discover cannot be loaded -> That's expected if it is already loaded. Check with `pactl list modules`. However...

bluetoothctl fails to connect to the device -> You need to load module-bluez5-discover for that to work.

lyubod, 2016-05-10:
Thanks for your help.

1. aptitude search '~Pnotification-daemon ~i' shows nothing but another prompt, after some other text indicating it's searching. I thought even the really basic GUIs like LXDE and XFCE had some notification, obviously not. So, I'd need to install that, but I've no idea which one, I'll look into it, but appreciate suggestions.

2. pactl list modules shows module-bluez5-discover, so it seems that already loads.

cschramm, 2016-05-10:
`aptitude search '~Pnotification-daemon'` lists all available packages that provide `notification-daemon`. If you really have none installed, your setup is broken as the `blueman` package depends on it. Xfce has `xfce4-notifyd`. I think there's actually no LXDE specific one.

Regarding your BlueZ problem, you can stop and start bluetoothd with the -d switch to get debugging output in the syslog (/var/log/daemon.log).

lyubod, 2016-05-10:
I confirmed the blueman package from Debian repos depends on notification-daemon (it does). Further, while the command you suggested showed nothing before, it now does! I theorize perhaps because aptitude itself was not installed by default, so it had to index or search what was installed. Here's what it displays now in response to: aptitude search '~Pnotification-daemon'

p   cinnamon                       - Innovative and comfortable desktop                                                    
p   dunst                              - dmenu-ish notification-daemon                                                         
p   gnome-shell                    - graphical shell for the GNOME desktop                                                 
p   lxqt-notificationd              - LXQt notification daemon                                                              
p   mate-notification-daemon  - daemon to display passive popup notifications                                         
p   notify-osd                          - daemon that displays passive pop-up notifications                                     
p   plasma-workspace             - Plasma Workspace for KF5                                                              
p   xfce4-notifyd                       - simple, visually-appealing notification daemon for Xfce  

so the first mystery is why notification-daemon is not on this list either, but even more puzzling, turns out it IS actually installed!

apt-get install notification-daemon
Reading package lists... Done
Building dependency tree       
Reading state information... Done
notification-daemon is already the newest version (3.20.0-1).
notification-daemon set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

I believe it was installed automatically, I have no reason to do it manually, so I have no idea why it says that.

Trying to get more ideas from the log where to go from here. Beginning to think if it's broken, I have to wait until Stretch is out of testing for it to get sorted out, which would be sad.

cschramm, 2016-05-10:
Sorry, I got that wrong. While notification-daemon is provided by a lot of different packages, it's not actually virtual as there is a concrete package with that name which is the Gnome notification daemon and is obviously installed on your system. I'll check the dbus situation later.

lyubod, 2016-05-10:
The deeper I dig the messier it seems to get. Clearly there are problems outside the scope of what blueman does, though I want to document them here so you are aware they affect blueman badly, and in hopes of finding a working solution and helping others in the same predicament.

At first I thought this bug might be responsible:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=747465

Didn't help, though it doesn't seem to hurt either. I did find the combination of Debian, notification-daemon and several DEs including LXDE has had several serious bugs though:
https://bugs.debian.org/cgi-bin/pkgreport.cgi?package=notification-daemon

tried installing xfce4-notifyd, figured I live with the 9 dependencies, even on LXDE, if it works. Sadly it didn't. Maybe I'll try notify-osd, it seeme

The closest I came to finding out something useful is this:
xfce4-notifyd has a dependency on libnotify-bin, notification-daemon doesn't! Seems shocking, but even installing libnotify-bin didn't fix it, though I can write myself messages with notify-send now, whee!

Rambling a bit.
I tried the entire process about as manual as it gets, via bluetoothctl, doing pair, trust, ... the whole thing. Just like the GUI, it succeeds at every step, but fails on connect. While there may be (are?) things other packages are breaking, at least with what blueman is doing, it seems to me (not a programmer, but with a great deal of sysadmin experience :-) that is key: essentially it fails to authenticate/send/receive the BT PIN. This also may be a DBus bug, but I'm trying to supply relevant technical info. to narrow down and eliminate it.  Even if you already saw it, please try and have a look at the gist above linked to: (Notice it mentions blueman and dbus by name, and even shows which line of python3 code, seems to not execute.
https://gist.github.com/lyubod/b651f08dab4ab7e314f06007058dda6d

I also tried to purge/reinstall python-notify, again because that seems to have helped others with similar errors. I'm so close, it actually shows the words "audio profile" next to it in the devices screen before it says it failed. And the exact same headset/BT dongle already worked years ago, and since unlike hard drives there are no moving parts, just circuits, the hardware appears to work ok, it should only be a matter of software.

Thanks.

cschramm, 2016-05-11:
I checked the notification daemon situation:

`notification-daemon` ships a binary called `/usr/lib/notification-daemon/notification-daemon` that must be started to get the `org.freedesktop.Notifications` service in your dbus session. From https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=814949 it looks like notification-daemon is actually not expected to be used anywhere now and that you should use some other implementation like dunst or xfce4-notifyd.

`xfce4-notifyd` on the other hand installs `/usr/share/dbus-1/services/org.xfce.xfce4-notifyd.Notifications.service` with which it's activatable via dbus. You should see an activatable `org.freedesktop.Notifications` service in your dbus session (e.g. using `d-feet`) no matter if it's running or not.

Regarding your BlueZ issue:

The log including lines of blueman is not your problem. The issue there is that BlueZ asks blueman to handle the authentication and blueman then fails to connect to the notification service and gives a negative reply aborting the pairing process. As long as you're not having a working notification daemon you won't be able to pair with blueman.

Your bluetoothctl gist shows that pairing succeeds (if not blueman but bluetoothctl provides the agent), but connecting fails. Unfortunately `org.bluez.Error.Failed` can have a lot of different causes. If you start bluetoothd with -d to get debug output in the syslog, you should be able to spot the actual error there.

lyubod, 2016-05-11:
Again, thanks for the help, I know this must be frustrating for you as well.

I understood that it must be failing authentication, because there is no notify mechanism active, because on OS X it asks for a PIN, and because handsfree headsets have no numeric keypad, it is programmed at the factory with 1234 or some such, and OS X uses that to connect, while Debian doesn't even ask, but simply fails. I thought, mistakenly apparently, that notification-daemon handled that. I have now installed xfce.xfce4-notifyd and verified that there is an activatable org.freedesktop.Notifications service in my dbus session using d-feet. Have restarted and continue to test. Was not sure how to use bluetoothd -d, since launching it complains dbus has launched it already. So I had to stop the prior instance first. 3 steps:

1. Exit blueman, so it doesn't cause systemd to respawn the service. I simply right clicked exit on the tray icon's contextual menu
2. rfkill block bluetooth (I had to install rfkill separately)
3. sudo service bluetooth stop (can verify it is not running anymore with "service dbus status")

Now I was able to run bluetoothd -d& with no errors, and relaunch blueman-applet&

I hope the you meant /var/log/syslog. Here is the output from a failed connection attempt with profile headset (technically the headset has both a microphone and earphone, and runs with both in OS X), run after bluetoothd -d:

May 11 07:05:43 stretch1 bluetoothd[1736]: src/device.c:connect_profiles() /org/bluez/hci1/dev_00_19_7F_EC_6B_0B (all), client :1.53
May 11 07:05:43 stretch1 bluetoothd[1736]: src/service.c:btd_service_ref() 0x81b7fa38: ref=2
May 11 07:05:43 stretch1 bluetoothd[1736]: src/service.c:change_state() 0x81b7fa38: device 00:19:7F:EC:6B:0B profile Headset Voice gateway state changed: disconnected -> connecting (0)
May 11 07:05:47 stretch1 bluetoothd[1736]: src/adapter.c:connected_callback() hci1 device 00:19:7F:EC:6B:0B connected eir_len 16
May 11 07:05:48 stretch1 bluetoothd[1736]: src/profile.c:ext_connect() Headset Voice gateway connected to 00:19:7F:EC:6B:0B
May 11 07:05:48 stretch1 bluetoothd[1736]: src/service.c:change_state() 0x81b7fa38: device 00:19:7F:EC:6B:0B profile Headset Voice gateway state changed: connecting -> disconnected (-5)
May 11 07:05:48 stretch1 bluetoothd[1736]: src/device.c:device_profile_connected() Headset Voice gateway Input/output error (5)
May 11 07:05:48 stretch1 bluetoothd[1736]: src/device.c:device_profile_connected() returning response to :1.53
May 11 07:05:48 stretch1 bluetoothd[1736]: src/service.c:btd_service_unref() 0x81b7fa38: ref=1
May 11 07:05:51 stretch1 bluetoothd[1736]: src/adapter.c:dev_disconnected() Device 00:19:7F:EC:6B:0B disconnected, reason 2
May 11 07:05:51 stretch1 bluetoothd[1736]: src/adapter.c:adapter_remove_connection()
May 11 07:05:51 stretch1 bluetoothd[1736]: plugins/policy.c:disconnect_cb() reason 2
May 11 07:05:51 stretch1 bluetoothd[1736]: src/adapter.c:bonding_attempt_complete() hci1 bdaddr 00:19:7F:EC:6B:0B type 0 status 0xe
May 11 07:05:51 stretch1 bluetoothd[1736]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
May 11 07:05:51 stretch1 bluetoothd[1736]: src/device.c:device_bonding_failed() status 14
May 11 07:05:51 stretch1 bluetoothd[1736]: src/adapter.c:resume_discovery()

cschramm, 2016-05-12:
```
May 11 07:05:48 stretch1 bluetoothd[1736]: src/profile.c:ext_connect() Headset Voice gateway connected to 00:19:7F:EC:6B:0B
May 11 07:05:48 stretch1 bluetoothd[1736]: src/service.c:change_state() 0x81b7fa38: device 00:19:7F:EC:6B:0B profile Headset Voice gateway state changed: connecting -> disconnected (-5)
```

That's the interesting part. 5 means a generic Input / Output error and can have a bunch of different causes in the BlueZ code, however, from the code I looked through (close to release 5.27) I'd expect an additional message between those two in any case.

You're probably experiencing some bug in BlueZ. The release notes of 5.37, 5.38, and 5.39 all refer to fixes for external profile registration, AVRCP, and A2DP. What UUIDs does your device provide? You can see them with e.g. bluetoothctl. I'd try a different version of BlueZ. There's [5.23 in jessie](https://packages.debian.org/jessie/bluez) and [5.33 on snapshot.debian.org](http://snapshot.debian.org/package/bluez/5.33-1/#bluez_5.33-1).

lyubod, 2016-05-12:
Were going further and further outside of anything to do with Blueman, so while I honestly am grateful for any advice and help you can (and have!) offered, I'm almost ready to say "BlueZ is junk, at least in Debian/LXDE, if not in general!" and leave it at that and try other distros/DE combinations in hope they're better. As a point of reference, I retried the same type of pairing/connection on Wheezy (7.9) with BlueZ 4.9, and it fails there as well. It also get visibly close, like pairing/trusting ok, it even connects, including asking for a PIN on the first attempt to add after removing! But it shows no new audio device in PAvucontrol, so no way to select it as a sound input/output! I'm now updating Wheezy as far is it will go (I haven't run Synaptic in a while) and if that doesn't help, I'll either update to Jessie (the only one of the last 3 Debian releases I haven't tried yet!) or try with Lubuntu 14.04, which dual-boots on the Wheezy desktop. I'd also like to try other versions of Bluez, besides those in Debian/Ubuntu repos, in case that helps, and yes I also noted the last several releases are "bug fixes". I've even compiled 5.39 (latest as of two days ago) from source, since the Stretch system only has 5.36 or so in the repos, but want a clean way to install/remove, like a package, make install may mess things up so I can't fix it, and I need this computer to work. Perhaps checkinstall or similar?

Again, thank you for your help in what is 99% (actually 100) BlueZ/DE bugs.

infirit, 2016-05-12:
One question remains for me, if we do not have a daemon installed, shouldn't we fallback to the widget?

lyubod, 2016-05-12:
That I can't help with, except that even though the system is using LXDE, as it has no "native" notification daemon according to what I read, I installed xfce4-notifyd (obviously with dependencies, it was like 9 packages) so I know something is capable of handling the PIN request. And on Wheezy 7.9 with BlueZ 4.9, it did the PIN fine, it just wouldn't show as a audio device in pulseaudio. On Stretch with BlueZ 5.36, I haven't even seen a PIN notification that I recall, it just pairs ok, but won't connect.

cschramm, 2016-05-13:
@infirit: No. The problem here is a distribution specific packaging issue. The maintainer of the blueman package (me in case of Debian) has to decide if a notification daemon shall be used (the default case and what is done in the Debian package) or disable the integration otherwise. If he enables the integration he has to make sure that a notification daemon is actually available. In Debian you do that by depending on notification-daemon which in jessie is provided by gnome-shell (Gnome), plasma-workspace (KDE), xfce4-notifyd (Xfce), mate-notification-daemon (MATE), cinnamon (Cinnamon), lxqt-notificationd (LXQt), python-jarabe (Sugar), dunst, and notify-osd. There's also the old notification-deamon from Gnome. Usually the user has some notification-daemon related to his DE anyway, however it in fact seems like LXDE does not necessarily have one (if you install task-lxde-desktop or the lxde meta package lxde-core recommends a notification-daemon but does not depend on it), so the user or system needs to pick one when blueman is installed. Unfortunately the notification-daemon package itself is bad choice as it's not working out of the box and the maintainer seems to be just waiting for removal (see the bug linked in a previous comment). Lubuntu does not suffer from this issue as lubuntu-desktop depends on xfce4-notifyd.

infirit, 2016-05-13:
I removed xfce4-notifyd and I get the fallback widget because ``Notify.get_server_caps()`` returns an empty list. This also happens for me if the gnome notification-daemon is installed but not running.

So I am confused why we end up at [this](https://gist.github.com/lyubod/b651f08dab4ab7e314f06007058dda6d) error.

cschramm, 2016-05-13:
Makes sense, yes. Did not remember that we have so many fallback cases now. Should probably add that other one then. :smiley: 

I also have to revoke my statement that the maintainer decides if a notification-daemon is used or not. Actually the user decides and the maintainer can only preset it in the GSettings schema.

infirit, 2016-05-13:
I'll cook something up for that next week so assigning to myself.

infirit, 2016-05-13:
#520

cschramm, 2016-06-03:
I'll close this. Reopen if you think we can do anything here.

closed 2016-06-03 by cschramm