blueman-project/blueman #524: Unprivileged rfcomm

cschramm, 2016-05-16:
Using blueman's serial service I just realized a fundamental usability issue it shares with `rfcomm connect`:

To create the rfcomm device root privileges are required, so `rfcomm connect` needs to be run as root and blueman uses its mechanism. The issue with running `rfcomm connect` is that it then opens the device - also as root of course as that's the user who's running `rfcomm`. The problem is that the user will then only be able to access the device as root too, even if he would have sufficient file system privileges. Otherwise he gets EBUSY.

In the same manner blueman runs its rfcomm-watcher as root as well and has the same issue then.

I just tried and replaced the `open_rfcomm` and `close_rfcomm` calls in `blueman.services.meta.SerialService` with the respective code from `blueman.plugins.mechanism.Rfcomm` and it works perfectly fine for me. blueman-mechanism creates the rfcomm device with `crw-rw---- 1 root dialout` and as I'm in the dialout group the unprivileged group blueman-rfcomm-watcher is able to open it.

The problem with this are the words "for me". If I would not be able to access the file after creation, i.e. especially if I would not be in the dialout group, blueman-rfcomm-watcher would fail. So this is not a generic solution.

I'm not sure how to best handle this. Maybe (applet's perspective):
````
ask mechanism to create the device
if I have read access:
  start rfcomm-watcher
else:
  ask mechanism to start rfcomm-watcher
````

cschramm, 2016-05-21:
OK, it does not seem to be that simple. If I try to start rfcomm watcher right after creating the deivce, I get the same busy error. I have to wait for some seconds to be able to get it as it seems to be occupied by root initially. I guess my initial tests only worked as I had a breakpoint in the code which delayed the start of the watcher. :confused:

infirit, 2016-05-21:
I am not all too familiar with the rfcomm stuff but..

Your approach appears sound to me and we only need to catch the exception, wait half a second or so, try again. But then do give up after 10 attempts. Or I am completely missing the point and then feel free to yell at me :smile:

cschramm, 2016-05-22:
I found out what's going on: ModemManager occupies the new device for a couple of seconds. The same issue obviously once appeared in the PPPSupport plugin and I'll reuse the code from there (wait for five seconds if it's running). :smiley:

infirit, 2016-05-23:
Ok, good. Which raises the question, for me, shouldn't we use ModemManager then, if available?

cschramm, 2016-05-23:
I don't think ModemManager can do anything we need. It manages modems while we want to provide connections to any serial device.

The check in PPPSupport turned out to be broken as it greps a ps output for ModemManager which will also find the grep process itself and never be empty, so it will always delay for 5 seconds, no matter what. Will fixed that along the way.

zaps166, 2016-10-16:
Today I've tried to connect with HC-05 using blueman 2.0.4. Unfortunately, access to `/dev/rfcomm0` requires root permissions. Also disconnecting doesn't work - blueman "thinks" that devices are disconnected, but `/dev/rfcomm0` device exists and `blueman-rfcomm-watcher` process still exists (must be manually killed).

Anyway, after successful pairing, writing `sudo rfcomm bind 0 xx:xx:xx:xx:xx:xx` in console works properly - doesn't need root permissions for user access. Maybe it could be called via blueman for serial port connection (`sudo rfcomm release 0` for unbind)?

infirit, 2016-10-17:
@cschramm, I fixed the issue reporting in https://github.com/blueman-project/blueman/issues/524#issuecomment-220964106 with with #601. Using ``pgrep`` makes things much simpler.

infirit, 2016-10-17:
@zaps166 https://github.com/blueman-project/blueman/pull/601/commits/9026b878ae65eac263fae66787460f7671c6a514 will properly kill rfcomm-watcher and for me /dev/rfcommX is removed automatically. Still seeing an error in the logs about ``_blueman.release_rfcomm_device`` but at least the devices is gone now.

zaps166, 2016-10-17:
@infirit Hmm, works, but weird. It can disconnect after a few seconds (with an error message), but if I click it many times it will hang (blueman will disconnect, but /dev/rfcommX still exists, also `rfcomm release` doesn't work - maybe it should ignore more then one clicks?).

User still needs root permissions.

cschramm, 2016-10-23:
#526 should help. Unfortunately it's not completely finished (we still need to improve the hack for udev rules).

PR #526 (2017-04-16): [Serial] Open rfcomm device as unprivileged user
...if he has read and write access.

This requires the watcher to wait if the device is busy which means it's opened by root (or any other user) and typically happens if ModemManager is present and probes the new device. The same method is used for PPPSupport now which previously always waited for five seconds unconditionally. While that is unnessary if ModemManager is not present, it seems to be too short if it is.

Fixes #524.

I'm not 100 % sure about the 0.1 sec delay before checking the access permissions, but if I do it immediately after c [... cut]
closed 2017-04-16 by cschramm