blueman-project/blueman #530: Cannot pair to bluetooth Headphones

gameslayer, 2016-06-03:
Can't use A2DP Sink or headset service to to my headphones (headphone not appearing in my pulse audio)

infirit, 2016-06-03:
Have a look at our wiki entry for a2dp https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect.

If you still have problems report back with the required in per https://github.com/blueman-project/blueman/wiki/Troubleshooting#reporting-a-bug--asking-for-support

gameslayer, 2016-06-03:
@infirit I am using Xubuntu 14.04 shouldn't this already be built in and working with blueman?

cschramm, 2016-07-06:
I don't think Xubuntu (14.04) ships pulseaudio-module-bluetooth.

gameslayer, 2016-07-07:
Well why woulnd't it and how can I tell, I am now using Xubuntu 16.04

cschramm, 2016-07-11:
> why woulnd't it

Because somebody decided against it? I don't know, but at least the xubuntu-desktop package does not suggest it, so I guess it's not getting installed in the first place.

> how can I tell

As usual, e.g. `dpkg -l pulseaudio-module-bluetooth`.

infirit, 2016-08-04:
Closing this one as I see nothing left to discuss. Feel free to comment and we can always re-open.

closed 2016-08-04 by infirit