blueman-project/blueman #531: ManagerDeviceList.py:356:level_setup_event:GLib.GError: g-file-error-quark: Failed to open file '/usr/share/pixmaps/blueman/blueman-rssi-140.png': No such file or directory (4)

leigh123linux, 2016-06-06:
Bug forwarded from https://bugzilla.redhat.com/show_bug.cgi?id=1342907

```

Version-Release number of selected component:
blueman-2.1-0.8.gita0408c1.fc24

Additional info:
reporter:       libreport-2.7.1
cmdline:        python3 /usr/bin/blueman-manager
executable:     /usr/bin/blueman-manager
kernel:         4.6.0-aosc-sunxi
pkg_fingerprint: 73BD E983 81B4 6521
pkg_vendor:     Fedora Project
reproducible:   The problem is reproducible
runlevel:       N 5
type:           Python3
uid:            1000

Truncated backtrace:
ManagerDeviceList.py:356:level_setup_event:GLib.GError: g-file-error-quark:
Failed to open file '/usr/share/pixmaps/blueman/blueman-rssi-140.png': No such
file or directory (4)

Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 604, in
msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/bin/blueman-manager", line 148, in on_bluez_name_owner_changed
    self.List.DisplayKnownDevices(autoselect=True)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 325,
in DisplayKnownDevices
    self.device_add_event(Device(device))
  File
"/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py",
line 193, in device_add_event
    self.AppendDevice(device)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 358,
in AppendDevice
    self.add_device(device, True)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 303,
in add_device
    self.monitor_power_levels(device)
  File "/usr/lib/python3.5/site-packages/blueman/gui/DeviceList.py", line 190,
in monitor_power_levels
    self.level_setup_event(r, device, cinfo)
  File
"/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceList.py",
line 356, in level_setup_event
    icon = GdkPixbuf.Pixbuf.new_from_file(PIXMAP_PATH + "/blueman-rssi-" +
str(rnd(rssi_perc)) + ".png")
GLib.GError: g-file-error-quark: Failed to open file
'/usr/share/pixmaps/blueman/blueman-rssi-140.png': No such file or directory
(4)

Local variables in innermost frame:
rssi_perc: 142.91338582677167
self: <ManagerDeviceList.ManagerDeviceList object at 0xaea571e8
(blueman+gui+manager+ManagerDeviceList+ManagerDeviceList at 0x7f9a22d8)>
tpl_perc: 50.0
row_ref: <Gtk.TreeRowReference object at 0xaea750a0 (GtkTreeRowReference at
0x7fa32c60)>
on_finished: <function ManagerDeviceList.level_setup_event.<locals>.on_finished
at 0xaea5ec48>
fader: <GtkAnimation.CellFade object at 0xaea6b468
(blueman+gui+GtkAnimation+CellFade at 0x7f841d80)>
lq: 0.0
signal: 599
cinfo: <_blueman.conn_info object at 0xb42acd28>
iter: <Gtk.TreeIter object at 0xaea750d8 (GtkTreeIter at 0x7f8425b0)>
row: {'tpl': 0.0, 'rssi': 0.0, 'levels_visible': False, 'lq': 0.0,
'cell_fader': <GtkAnimation.CellFade object at 0xaea6b468
(blueman+gui+GtkAnimation+CellFade at 0x7f841d80)>}
lq_perc: 10
device: <Device.Device object at 0xaea65fd0 (blueman+main+Device+Device at
0x7f8417d0)>
rssi: 236.0
rnd: <function ManagerDeviceList.level_setup_event.<locals>.rnd at 0xaea5ec00>
tpl: 0.0
```

cschramm, 2016-06-06:
> rssi: 236.0

Uh, nice. The maximum value for that is 127 as it's from a signed 8 bit integer... What version of cython is used to build the _blueman module? Can you show us the code from _blueman.c that's generated for _blueman.pyx's get_rssi method? The return value should be generated with `__Pyx_PyInt_From_char` and I don't see any chance to get something higher than 127 there...

leigh123linux, 2016-06-06:
Hi,
```
rpm -q python3-Cython
python3-Cython-0.23.4-3.fc24.x86_64
```

_blueman.c

http://paste.fedoraproject.org/375311/20860514

_blueman.pyx

http://paste.fedoraproject.org/375314/52086871

infirit, 2016-06-06:
Are there any things we need to think about when dealing with the arm arch (armv7l in the bug report)?

cschramm, 2016-06-06:
Ah, that's the point I missed. Well, we cannot do much wrong there. If there's a problems, I guess the blame is on cython.

Would be interesting which branch of
```c
static CYTHON_INLINE PyObject* __Pyx_PyInt_From_char(char value) {
    const char neg_one = (char) -1, const_zero = (char) 0;
    const int is_unsigned = neg_one > const_zero;
    if (is_unsigned) {
        if (sizeof(char) < sizeof(long)) {
            return PyInt_FromLong((long) value);
        } else if (sizeof(char) <= sizeof(unsigned long)) {
            return PyLong_FromUnsignedLong((unsigned long) value);
        } else if (sizeof(char) <= sizeof(unsigned PY_LONG_LONG)) {
            return PyLong_FromUnsignedLongLong((unsigned PY_LONG_LONG) value);
        }
    } else {
        if (sizeof(char) <= sizeof(long)) {
            return PyInt_FromLong((long) value);
        } else if (sizeof(char) <= sizeof(PY_LONG_LONG)) {
            return PyLong_FromLongLong((PY_LONG_LONG) value);
        }
    }
    {
        int one = 1; int little = (int)*(unsigned char *)&one;
        unsigned char *bytes = (unsigned char *)&value;
        return _PyLong_FromByteArray(bytes, sizeof(char),
                                     little, !is_unsigned);
    }
}
```
is used. I'd expect it to always use `PyInt_FromLong((long) value);` but the other cases must be there for some reason...

infirit, 2016-08-21:
upstream bug got re-asigned to cython, closing for now. If something comes up we can re-open.

michitux, 2018-05-27:
I am encountering this error on an armhf system (CPU is armv7 but the system is based on Alpine Linux armhf, i.e., armv6). As far as I have [read](https://www.linuxtopia.org/online_books/an_introduction_to_gcc/gccintro_71.html) (and [here](http://www.arm.linux.org.uk/docs/faqs/signedchar.php)), `char` is indeed unsigned on ARM. Note that the C-function that is called is declared with `int8_t`, not with `char`. Therefore, the method in Cython should simply be declared with a `signed char` (or directly with `int8_t`, though that type might need to be imported first using `from libc.stdint cimport int8_t`). Declaring the argument and the temporary variable as `signed char` fixes this error for me (though I am encountering different, unrelated problems).

infirit, 2018-05-27:
I noticed a warning related to this but I lost track of the commit :disappointed:. Does https://github.com/infirit/blueman/commit/ec498cfffaf8b3f220fdcd8852ef9d1f122c6785 solve the problem?

michitux, 2018-05-27:
Yes, the two changes related to `get_rssi` are exactly the changes I have tested.

infirit, 2018-05-29:
Merged, should end up in version 2.1

closed 2018-05-29 by infirit

collares, 2021-07-04:
Should the corresponding fix be applied to get_lq as well? I just had this error message with blueman 2.2.1:

```
Jul 04 13:28:29 hp gnome-shell[3724]: Traceback (most recent call last):
Jul 04 13:28:29 hp gnome-shell[3724]:   File "/nix/store/2fgjcf8qcj5vhamcj1v8n73mrp1nwya9-blueman-2.2.1/lib/python3.8/site-packages/blueman/gui/DeviceList.py", line 148, in _on_device_property_changed
Jul 04 13:28:29 hp gnome-shell[3724]:     self.row_update_event(tree_iter, key, value)
Jul 04 13:28:29 hp gnome-shell[3724]:   File "/nix/store/2fgjcf8qcj5vhamcj1v8n73mrp1nwya9-blueman-2.2.1/lib/python3.8/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 397, in row_update_event
Jul 04 13:28:29 hp gnome-shell[3724]:     self._monitor_power_levels(tree_iter, self.get(tree_iter, "device")["device"])
Jul 04 13:28:29 hp gnome-shell[3724]:   File "/nix/store/2fgjcf8qcj5vhamcj1v8n73mrp1nwya9-blueman-2.2.1/lib/python3.8/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 343, in _monitor_power_levels
Jul 04 13:28:29 hp gnome-shell[3724]:     self._update_power_levels(tree_iter, device, cinfo)
Jul 04 13:28:29 hp gnome-shell[3724]:   File "/nix/store/2fgjcf8qcj5vhamcj1v8n73mrp1nwya9-blueman-2.2.1/lib/python3.8/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 438, in _update_power_levels
Jul 04 13:28:29 hp gnome-shell[3724]:     icon = GdkPixbuf.Pixbuf.new_from_file_at_scale(os.path.join(PIXMAP_PATH, icon_name), w, h, True)
Jul 04 13:28:29 hp gnome-shell[3724]: gi.repository.GLib.Error: g-file-error-quark: Failed to open file “/nix/store/2fgjcf8qcj5vhamcj1v8n73mrp1nwya9-blueman-2.2.1/share/blueman/pixmaps/blueman-lq-150.png”: No such file or directory (4)
```

cschramm, 2021-07-05:
@collares: See #1563 for a fix.