blueman-project/blueman #532: Bluetooth keyboard won't stay connected to blueman

juerujin, 2016-06-07:
blueman: blueman-applet version 1.99.alpha1 
BlueZ: Version: 5.23-2+rpi2
Distribution: Raspbian Jessie
Desktop environment: Raspberry Pi 3

Hi,

Having an issue with the version of blueman available to the raspberry pi 3. I recently did a full install of raspbian with the default LXDE and installed XFCE on top of that. Once i switched to XFCE there was no native bluetooth manager so i installed blueman which seemed to be the one of choice. After installing blueman I attempted to install a mouse, which despite a few hiccups connected just fine in the end. Turning the mouse off and letting the program notice and then turning it back on had the mouse reconnecting just fine as well. My keyboard however was not so fortunate. 

Initial hookup of the keyboard using add new device did not work very well. With the mouse i was asked to choose a random pin or to input my own but upon clicking next it just connected anyway without doing anything. With the keyboard no matter whether i choose random code or input my own, on the next page i see a code and i enter it on the keyboard but nothing happens. After a while of trying to enter the code multiple times (or even once) eventually it says paring failed.
I had better luck going to Bluetooth devices searching for the device and then clicking @connect@ at that stage it asks me for a pin. Someone on a forum mentioned that they had to enter the pin on the computer first and then the  keyboard and that finally worked. Which was awesome. But then more trouble. If i didn't use the keyboard for a few a bit it disconnected (This behavior was not exhibited by the mouse or the keyboard when connected to another device). Typing on the keyboard saw it attempting to reconnect and succeeding but no input is shown when typing. Removing the keyboard and adding it again has it working fine ... until it disconnects. 

So in short:

- Mouse connects fine
- Keyboard won't connect using add device
- keyboard will connect using connect if code is entered and then on keyboard
- keyboard disconnects and when reconnected won't send input to computer
- This behavior is not demonstrated on any other non linux device (or on the LXDE bluetooth manager in raspbian for that matter) or by the mouse connected to blueman

Here is a pastebin of a typical connect cycle of the keyboard using blueman-applet > log.txt:
http://pastebin.com/tie9hFk5

That includes startup, connection, disconnection and attempted reconnect.

If there is any other information i can provide that might help, feel free to let me know. In the meanwhile i will continue to experiment and hopefully find something.

juerujin, 2016-06-07:
I stand corrected. When attempting to connect to the built in bluetooth manager in the LXDE panel it connects easier but the same disconnect behavior shows up. Or rather it shows that it's connected but no input is shown on computer. Attempting to disconnect and reconnect gives this error.

connection failed - GDBus.error:org.bluez.Error.Failed: input/output error

Any ideas?

cschramm, 2016-06-07:
You might want to get a recent version of blueman from jessie-backports, but I doubt that it would improve the keyboard situation.

It sounds more like an issue between your keyboard and BlueZ - or maybe it's even just about the input layer and not primarily about bluetooth. Did you check if it works in a terminal after reconnecting? The problem might be restricted to X11.

yars068, 2016-06-07:
Version 1.x is too old, please try with actual 2.0.3 from (backports)[https://packages.debian.org/jessie-backports/blueman].

juerujin, 2016-06-07:
@cschramm It also stops working in terminal on the GUI desktop but i'm assuming you mean from pure text shell?
@yars068 How does one go about upgrading to 2.0.3 from backports and will it work with the pi?

juerujin, 2016-06-07:
Perhaps i should read up on things first. Looked into what a backport is and i'll start looking into how i can do that. Though i wouldn't mind if you had quick instructions on hand

juerujin, 2016-06-07:
Well I added the jessie backports and installed blueman 2.0.3. Pairing goes a lot smoother. Works just by doing add new device. Keyboard still drops out after time though and doesn't come back. Tried going to the terminal via `ctrl + alt + F1` but it it's the same thing ... works till it disconnects and then doesn't reconnect. 

Any other suggestions or is this now not a blueman issue.

juerujin, 2016-06-07:
Well with the new blueman, after the keyboard disconnects i can turn it off and back on and then hit setup and it will reconnect. This is fairly bizarre

cschramm, 2016-06-08:
What do you mean with "hit setup"? The "setup" item in the device's context menu in the manager window? That should just start a dialog and not do anything regarding the bluetooth stack.

closed 2016-11-18 by cschramm