blueman-project/blueman #534: Audio receiver don't work

Vairetz, 2016-06-14:
Hi, I install blueman and I would like to use it for streaming music from my phone.
I use raspbian now and when I go on device configuration I don't see audio source as I see when I use blueman on ubuntu mate. I enabled advanced audio receiver on local services but the situation don't change, how can I fix the problem?

Raspberry Pi 2 Model B

cschramm, 2016-06-15:
Simply missing `pulseaudio-module-bluetooth`?

Otherwise `dpkg -l | grep blue` please.

Vairetz, 2016-06-15:
pluseaudio-module-bluetooth don't missing and it's the last version.
the result of dpkg -l | grep blue is:

ii  bluej                                 3.1.7                                     all          A simple but powerful Java IDE.
ii  blueman                               1.99~alpha1-1+deb8u1                      armhf        Graphical bluetooth manager
ii  bluez                                 5.23-2+rpi2                               armhf        Bluetooth tools and daemons
ii  bluez-firmware                        1.2-3+rpi1                                all          Firmware for Bluetooth devices
ii  libbluetooth3:armhf                   5.23-2+rpi2                               armhf        Library to use the BlueZ Linux Bluetooth stack
ii  pi-bluetooth                          0.1.1                                     armhf        Raspberry Pi 3 bluetooth
ii  pulseaudio-module-bluetooth           5.0-13                                    armhf        Bluetooth module for PulseAudio sound server

cschramm, 2016-06-15:
You're probably using something more recent than 1.99.alpha1 on Ubuntu. You can try the version from Debian jessie-backports.

Vairetz, 2016-06-22:
How can I install it? Could you tell me the comands that I must to use in terminal or other method of installation?

cschramm, 2016-06-22:
Either just [download](https://packages.debian.org/jessie-backports/armhf/blueman/download) and `dpkg -i` it or [add jessie-backports to your sources](https://backports.debian.org/Instructions/) and `aptitude install` it.

Vairetz, 2016-06-25:
Now can I connect my device as audio source, but I have a failed connection..
The error message is: 
Connessione fallita:Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: connect_service is not a valid method of interface org.blueman.Applet

Vairetz, 2016-06-28:
I try to solve the problem with upgrade, but the problem remaning.
This is the error message: 
Connessione fallita:blueman.bluez.errors.DBusFailedError: No such file or directory

cschramm, 2016-06-28:
Sounds like you're missing `pulseaudio-module-bluetooth`.

Vairetz, 2016-06-30:
pi@raspberrypi:~ $ sudo apt-get install pulseaudio-module-bluetooth
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
pulseaudio-module-bluetooth è già alla versione più recente.
0 aggiornati, 0 installati, 0 da rimuovere e 0 non aggiornati.
pi@raspberrypi:~ $ 

It tells me that pluseaudio-module-bluetooth is just the last version

cschramm, 2016-07-06:
If the module is installed your audio setup seems to be insufficient to stream to the bluetooth sink, see https://github.com/blueman-project/blueman/issues/304#issuecomment-112090811. https://www.linuxmusicians.com/viewtopic.php?t=15478 points to problems regarding jack.

Vairetz, 2016-07-07:
Now I reinstall raspbian on my sd card for other problem, could you tell me all passage to install bluetooth and use it for strem music from my phone? 
Thank you so much for the patience, it's the most comfortable method to solve this problem

Vairetz, 2016-07-13:
I try to reinstall all parts but audio receiver doesn't work yet.. I try to  aptitude install pulseaudio-module-bluetooth but the problem is not resolved.

infirit, 2016-08-04:
Closing this one in favour of #558 as it is the same rpi receiver problem. The other thread has more and better info.

closed 2016-08-04 by infirit