blueman-project/blueman #535: I can connect my headset but no sound

Ange7, 2016-06-20:
Hey all,

I have one headset with laptop on Debian Stretch. 

When i connect it it's ok : 

> Jun 20 15:42:45 debian bluetoothd[600]: /org/bluez/hci0/dev_00_18_09_28_4F_5E/fd6: fd(33) ready
Jun 20 15:42:45 debian rtkit-daemon[1009]: Supervising 6 threads of 2 processes of 2 users.
Jun 20 15:42:45 debian rtkit-daemon[1009]: Successfully made thread 2073 of process 1008 (n/a) owned by '116' RT at priority 5.
Jun 20 15:42:45 debian rtkit-daemon[1009]: Supervising 7 threads of 2 processes of 2 users.
Jun 20 15:42:45 debian kernel: [  382.917810] input: 00:18:09:28:4F:5E as /devices/virtual/input/input24
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) config/udev: Adding input device 00:18:09:28:4F:5E (/dev/input/event15)
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) 00:18:09:28:4F:5E: Applying InputClass "evdev keyboard catchall"
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) 00:18:09:28:4F:5E: Applying InputClass "libinput keyboard catchall"
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) systemd-logind: got fd for /dev/input/event15 13:79 fd 33 paused 0
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) Using input driver 'libinput' for '00:18:09:28:4F:5E'
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) 00:18:09:28:4F:5E: always reports core events
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) Option "Device" "/dev/input/event15"
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) Option "_source" "server/udev"
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) input device '00:18:09:28:4F:5E', /dev/input/event15 is tagged by udev as: Keyboard
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) input device '00:18:09:28:4F:5E', /dev/input/event15 is a keyboard
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) Option "config_info" "udev:/sys/devices/virtual/input/input24/event15"
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) XINPUT: Adding extended input device "00:18:09:28:4F:5E" (type: KEYBOARD, id 15)
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) Option "xkb_model" "pc105"
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (**) Option "xkb_layout" "fr"
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (WW) Option "xkb_variant" requires a string value
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (WW) Option "xkb_options" requires a string value
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) input device '00:18:09:28:4F:5E', /dev/input/event15 is tagged by udev as: Keyboard
Jun 20 15:42:45 debian /usr/lib/gdm3/gdm-x-session[1235]: (II) input device '00:18:09:28:4F:5E', /dev/input/event15 is a keyboard

But i have no sound in my headset, in sound configuration i have only « Speaker » of my laptop. 

Why ? 

Thank you

infirit, 2016-06-20:
First have a look at https://wiki.debian.org/BluetoothUser/a2dp and follow the instructions. If you still run into problems see https://github.com/blueman-project/blueman/wiki/Troubleshooting#reporting-a-bug--asking-for-support

Ange7, 2016-06-20:
it seems ok with the wiki help.

Thank you Infirit

closed 2016-06-20 by Ange7