blueman-project/blueman #536: bluetooth-assistant wont run

x0a, 2016-06-21:
I get the following this error when I try to run any blueman application:

Full error: 
>   File "/usr/local/bin/blueman-assistant", line 31, in <module>
>     from blueman.gui.DeviceSelectorWidget import DeviceSelectorWidget
>   File "/usr/local/lib/python2.7/dist-packages/blueman/gui/DeviceSelectorWidget.py", line 16, in <module>
>     from blueman.gui.DeviceSelectorList import DeviceSelectorList
>   File "/usr/local/lib/python2.7/dist-packages/blueman/gui/DeviceSelectorList.py", line 14, in <module>
>     from blueman.gui.DeviceList import DeviceList
>   File "/usr/local/lib/python2.7/dist-packages/blueman/gui/DeviceList.py", line 11, in <module>
>     from _blueman import conn_info
> ImportError: /usr/local/lib/python2.7/dist-packages/_blueman.so: undefined symbol: hci_read_transmit_power_level

Blueman: 2.0.4
BlueZ: 5.40
Distribution: Ubuntu 14.04
Desktop environment: Unity

cschramm, 2016-06-21:
Sounds like an issue with your BlueZ installation. Check your hci_lib.h header for that name (compare http://git.kernel.org/cgit/bluetooth/bluez.git/tree/lib/hci_lib.h?h=5.40#n107). Check the libbluetooth.so.3 used with `ldd /usr/local/lib/python2.7/dist-packages/_blueman.so`. Check that library for the symbol with `nm -D`.

x0a, 2016-06-21:
You were right, _blueman.so wasn't using any libbluetooth.so.3. I hadn't specified systemd directories when running configure. That problem is solved but now I have a slightly different problem:

> Traceback (most recent call last):
>   File "/usr/local/bin/blueman-assistant", line 274, in <module>
>     Assistant()
>   File "/usr/local/bin/blueman-assistant", line 92, in __init__
>     self.assistant.set_page_header_image(pages[PAGE_CONNECT], get_icon("network-transmit-recieve", 32))
>   File "/usr/local/lib/python2.7/dist-packages/blueman/Functions.py", line 181, in get_icon
>     icon = ic.load_icon(name, size, 0)
> gi._glib.GError: Icon 'network-transmit-recieve' not present in theme

cschramm, 2016-06-21:
If you're using a custom glib in /usr/local as well its search path is probably bound to that location and does not look into /usr/share/icons. You can check the search path in Python with:

````
from gi.repository import Gtk
Gtk.IconTheme.get_default().get_search_path()
````

x0a, 2016-06-21:
Thanks for the info. I don't recall installing a custom glib, and there isn't any glib in /usr/local (although there's a 'glib-2.0' directory in /usr/local/share). Here's the search path

> \>\>\> Gtk.IconTheme.get_default().get_search_path()
> ['/home/x0a/.local/share/icons', '/home/x0a/.icons', '/usr/share/ubuntu/icons', '/usr/share/gnome/icons', **'/usr/local/share/icons'**, '/usr/share/icons', '/usr/share/ubuntu/pixmaps', '/usr/share/gnome/pixmaps', '/usr/local/share/pixmaps', '/usr/share/pixmaps']
> \>\>\> 

The icon is also listed in .list_icons(), and it even works from here:

> \>\>\> 'network-transmit-receive' in Gtk.IconTheme.get_default().list_icons();
> True
> \>\>\> Gtk.IconTheme.get_default().load_icon("network-transmit-receive", 32, 0)
> \<Pixbuf object at 0x7f4a08408e10 (GdkPixbuf at 0xe89140)\>
> \>\>\> 

It just doesn't seem to work at runtime.

x0a, 2016-06-21:
Problem was that the old bluetooth daemon was still running. 
Solution was to kill the old daemon, rename it, and link to the new daemon provided by Bluez:

> \# /etc/init.d/bluetooth stop
> \# mv /usr/sbin/bluetoothd  /usr/sbin/bluetoothd.orig
> \# ln -s /usr/local/libexec/bluetooth/bluetoothd /usr/sbin/bluetoothd
> \# /etc/init.d/bluetooth start

Finally got it to run. Thanks for the help

closed 2016-06-22 by cschramm