blueman-project/blueman #537: Fedora 24 x86_64 MATE 1.14.1 - Blueman 2.0.4 cannot control my notebook's adaptor

arehtykitna, 2016-06-21:
There is no version of blueman available for Fedora 24 that supports my Intel internal bluetooth adaptor. It is hosted by USB bus. I attach terminal output from attempting to use the applet. The F23 file was generated last night after spending some time looking into it and discovering the issue was caused by newer builds of blueman and not kernel module or hardware related. It references the last working build for my notebook (blueman-2.1-0.3.git7a2e20e.fc23) which cannot be installed for F24 because of conflicting library dependencies with newer libraries dnf the package management tool requires.

Please can you look into this and advise me if there is a workaround I can try or not do your best to patch the issue. I don't particularly want to have to stop using blueman as I like the way it works normally.

If you need any further information please let me know asap.

Please refer to the attached files.
[F24_blueman_output.txt](https://github.com/blueman-project/blueman/files/325319/F24_blueman_output.txt)
[lsusb_output.txt](https://github.com/blueman-project/blueman/files/325320/lsusb_output.txt)
[F23_output_after_downgrade.txt](https://github.com/blueman-project/blueman/files/325323/F23_output_after_downgrade.txt)

arehtykitna, 2016-06-21:
I've been advised to issue `sudo hciconfig hci0 up` to get it working. It does the trick but the solution is not ideal long term and was offered as a known bug fix for when resuming from suspend. I never use suspend, the problem is there after a cold boot.

cschramm, 2016-06-21:
@infirit: It looks like there are a lot of usages of `set` which need to use `GVariant`s now:

```
blueman/plugins/applet/DiscvManager.py
71:            self.adapter.set("Discoverable", True)
72:            self.adapter.set("DiscoverableTimeout", self.get_option("time"))

blueman/plugins/applet/PowerManager.py
92:                adapter.set("Powered", state)
228:                adapter.set("Powered", False)
230:                adapter.set("Powered", True)

blueman/bluez/Adapter.py
59:            return self.set('Alias', name)
61:            return self.set('Name', name)

blueman/gui/manager/ManagerDeviceMenu.py
304:                    device.set('Alias', alias_entry.get_text())
306:                    device.set('Alias', '')

blueman/main/applet/BluezAgent.py
240:                device.set("Trusted", True)
```

arehtykitna, 2016-06-21:
I've just tried another way to run the blueman-applet. I used terminal to `sudo blueman-applet`. It works properly like that so maybe there is a broken rights issue?

infirit, 2016-06-21:
>It looks like there are a lot of usages of ``set``

@cschramm,
In propertiesbase ``set`` constructs a GVariant based on a type map :smile:.

@arehtykitna, this is intended behaviour. BlueZ does not power the adapter automatically any-more since version 5.0. There are two options, enable auto-power on in the PowerManager plugin or create a udev rule as described [here](https://github.com/blueman-project/blueman/wiki/Troubleshooting#turn-on-bluetooth-device-on-boot). The former will power the adapter when you login and the latter will do it as soon as the adapter is detected.

arehtykitna, 2016-06-21:
@infirit - thanks for that. I wasn't aware the default behaviour had changed in that way. I've followed those very clear instructions and am happy that this can be closed if the set issues are nothing to worry about either (the bluez-utils are included in bluez package for F24).

cschramm, 2016-06-21:
@infirit: Hmm, right, but did you notice

> set_adapter_state (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:90)
Exception occurred org.freedesktop.DBus.Error.UnknownMethod: Method "Set" with signature "ssb" on interface "org.freedesktop.DBus.Properties" doesn't exist

in the first download? Not sure where that's coming from...

infirit, 2016-06-21:
That is the correct interface to call Set on but it appears something is going wrong. I have a working f24 machine which I can test this on.

However I am also (more actually) concerned with the earlier error that obexd dbus service is missing. I created a f24 vm and it is missing in d-feet, it appears bluez-obexd is not installed. But when I install that I am told that ``org.bluez.obex.service`` unit file is not found. I am hoping @leigh123linux can have a look at that as we require this for file transfer.

leigh123linux, 2016-06-21:
@infirit 

Hi, I orphaned blueman because I sick of getting abrt bugreports for issues caused by bluez5 , I don't know why the wankers would inflict "not even beta quality bluez" on users.

infirit, 2016-06-21:
@leigh123linux, hi, sorry to hear that.

Looks like the obex problem is bug 1318441 https://bugzilla.redhat.com/show_bug.cgi?id=1318441. Fixed with the below.
``
cd /usr/lib/systemd/user/
ln -s obex.service dbus-org.bluez.obex.service
```

arehtykitna, 2016-06-21:
Sorry to hear you actually have orphaned blueman, I'm not surprised and don't blame you for doing it @leigh123linux 

thanks again for your input infirit. I've added the instructions to my bugzilla report which has been re-assigned to bluez https://bugzilla.redhat.com/show_bug.cgi?id=1348346#c11.

infirit, 2016-06-21:
So, my guess for the failures is that for some reason we are sending the wrong ``dbus`` objects or dbus-python is more strict. Not sure what changed in dbus-python under fedora but I see no such issues with gentoo or debian. Apparently we are not the only ones having problems, https://bugzilla.redhat.com/show_bug.cgi?id=1236392

leigh123linux, 2016-06-21:
I currently own dbus-python.
fedora d-p had a lame patch included (before I owned it) to make some other project work.

http://pkgs.fedoraproject.org/cgit/rpms/dbus-python.git/commit/?h=f22&id=b00d726b5b3e8f7f590ed3eb95160d1b68003339

When I took over I dropped the patch and broke the other project

http://pkgs.fedoraproject.org/cgit/rpms/dbus-python.git/commit/?h=f23&id=354154a74dd2f57ec7e2b8fa0709546708628196

In the end I relented and reverted on the condition they fixed their project, fedora >=23 no longer uses the patch (I never reverted for f22 and consider dbus-python broken).

full details below

http://pkgs.fedoraproject.org/cgit/rpms/dbus-python.git/log/

infirit, 2016-06-22:
Yeah, something funny with dbus on fedora. I have to wrap to dbus types before dbus accepts the path to register the agent, see below.

Our agent uses a ``dbus.service.Object`` as base and we use the ``get_object_path`` from it. But this seems to return a normal string which is then rejected by dbus as the signature does not match with what it expects. The signature it expects is ``os`` but we send ``ss``. I have the same dbus-python version (vanilla) here on Gentoo it is fine here. So I am not sure why it now fails on fedora but not on Gentoo or Debian. Is Fedora using kdbus? If so is it more strict?

```diff
diff --git a/blueman/bluez/AgentManager.py b/blueman/bluez/AgentManager.py
index e4abb82..6411f71 100644
--- a/blueman/bluez/AgentManager.py
+++ b/blueman/bluez/AgentManager.py
@@ -16,7 +16,7 @@ class AgentManager(BlueZInterface):
 
     @raise_dbus_error
     def register_agent(self, agent, capability='', default=False):
-        path = agent.get_object_path()
+        path = dbus.ObjectPath(agent.get_object_path())
         self.get_interface().RegisterAgent(path, capability)
         if default:
             self.get_interface().RequestDefaultAgent(path)
```

infirit, 2016-06-22:
@arehtykitna if you want to build blueman master (2.1) you can make a rpm yourself with https://github.com/blueman-project/blueman/issues/422#issuecomment-169293906

infirit, 2016-06-22:
full backtrace.

```python
_________
set_adapter_state (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:90)
True 
Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/lib/python3.5/site-packages/blueman/plugins/applet/KillSwitch.py", line 122, in reply
    cb(True)
  File "/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py", line 114, in __call__
    self.check()
  File "/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py", line 119, in check
    self.parent.set_adapter_state(self.state)
  File "/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py", line 94, in set_adapter_state
    adapter.set("Powered", state)
  File "/usr/lib/python3.5/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
  File "/usr/lib/python3.5/site-packages/blueman/bluez/errors.py", line 146, in warp
    return func(*args, **kwargs)
  File "/usr/lib/python3.5/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 27, in set
    return self.__properties_interface.Set(self.get_interface_name(), name, value)
  File "/usr/lib64/python3.5/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "Set" with signature "ssb" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

arehtykitna, 2016-06-22:
@infirit Thanks! I'll definitely keep that for the future reference
and give it a go in the safe confines of VM over on the desktop. As it
stands now, 2.0.4 is running well with the udev command implemented. I
don't know enough to maintain and support the package officially for
Fedora Project like @leigh123linux kindly did. Moreover, I've never used
Python so get the gist of the above discussion (Fedora's dbus is a
unique pain in the backside) but not all of it.

infirit, 2016-06-23:
Master is working well, here is my whole rpmbuild dir including the rpms and sources exluding the build files. https://www.dropbox.com/sh/98wpjxkv9g7t52q/AABf6mmKSXwjyV5qLG1yds4Ya?dl=0

arehtykitna, 2016-06-23:
thanks for that, I tried building for F24 earlier using the latest
commit but ran into a problem with a missing hardened-dbus something or
other and forgot to make a note of it. I was doing too many things
at once. I'll try again at some point though. Your rpm installed fine
and the adaptor is physically working (music from my phone is playing
back via audio source mode), however that obex error is still coming
back to bite though. If there's a bluez rhbz open I'll join the CC list.

by the way, with dropbox links you can alter them for direct download
links as follows if sharing with people who know where the link is from:

https://dl.dropbox.com/sh/98wpjxkv9g7t52q/AABf6mmKSXwjyV5qLG1yds4Ya?dl=1

cheers, Mike

arehtykitna, 2016-06-23:
obex transfer definitely works with that fix, happy days! thanks for all your help guys

danielbgto, 2016-06-27:
Hi everyone! 
i'm was "fighting" with Fedora 23 (on 3 PCs) and now with Fedora 24 (after the upgrade) and bluetooth transfers. Latest working version for me is blueman-2.1-0.3.git7a2e20e.fc23, which i recompiled on F24. Sending and receiving via phone, bluetooth speakers are working fine. 
arehtykitna's builds (https://dl.dropbox.com/sh/98wpjxkv9g7t52q/AABf6mmKSXwjyV5qLG1yds4Ya?dl=1) are working by "sending" to phone  BUT "receiving" from the phone to the PC is not working. I got error message on the phone - "remote transfer refused by the target device". On my F24 box i got this:
Jun 27 15:24:40 pavitra-desktop bluetoothd[13420]: Unable to get io data for Object Push: getpeername: Transport endpoint is not connected (107)
Jun 27 15:24:40 pavitra-desktop obexd[2953]: disconnected: Transport got disconnected
Jun 27 15:24:40 pavitra-desktop obexd[2953]: DISCONNECT(0x1), SUCCESS(0x20)
Jun 27 15:24:40 pavitra-desktop obexd[2953]: DISCONNECT(0x1), (null)(0xffffffff)
Jun 27 15:24:40 pavitra-desktop obexd[2953]: PUT(0x2), FORBIDDEN(0x43)
Jun 27 15:24:40 pavitra-desktop obexd[2953]: Agent replied with an error: org.freedesktop.DBus.Error.UnknownMethod, No such interface 'org.bluez.obex.Agent1' on object at path /o
Jun 27 15:24:40 pavitra-desktop obexd[2953]: PUT(0x2), (null)(0xffffffff)
Jun 27 15:24:39 pavitra-desktop obexd[2953]: CONNECT(0x0), (null)(0x0)
Jun 27 15:24:39 pavitra-desktop obexd[2953]: CONNECT(0x0), (null)(0xffffffff)

I have no problem to stay with the older version of blueman on my PCs but probably many has the same problem and they probably don't know how to recompile a package. Bluetooth is almost totaly broken on Fedora from few months(all version after blueman-2.1-0.3.git7a2e20e.fc23 ).
i'm ready to help and testing if need.
Greeting and sorry for my bad english

arehtykitna, 2016-06-27:
@danielbgto that wasn't my build. I tried recompiling but ran into problems. The only builds available here are the work of @infirit 

I found receiving files didn't work either but I don't connect this problem specifically to blueman. bluez appears to cause this failure for me as I tried ftp commands in terminal using obex tools and got the same results. As I rarely transfer from the phone using bluetooth that wasn't such an issue. I mostly send from the notebook or use the audio services.

I'm going to re-attempt this when I can give 100% concentration to it. In the meantime would you be willing to share your build please daniel?

danielbgto, 2016-06-27:
Hi!
I really don't know where is the problem but one thing is sure - with some versions of blueman everything is working fine and with some versions nothing's working (forgot to mention that i'm using Mate Desktop also).
At https://kojipkgs.fedoraproject.org/packages/blueman/ i found sources of  more versions of blueman package. Some are just not released. I build version 2.1/0.8.gita0408c1.fc24 and after few tests seems to be a good successor of  2.1/0.3.git7a2e20e.fc23

You can found both version on:

blueman-2.1-0.3.git7a2e20e.fc24.x86_64.rpm - https://drive.google.com/open?id=0B1PsDDA14fWoc3BwU1RINkhqUms 

blueman-2.1-0.8.gita0408c1.fc24.x86_64.rpm - https://drive.google.com/open?id=0B1PsDDA14fWoVGpxZ0VJNlotaDA

Don't forget to restart the bluetooth service (after the package upgrade) and exit/kill blueman-applet (before the package upgrade). Or just restart the PC. :)

arehtykitna, 2016-06-27:
On Mon, 27 Jun 2016 07:04:20 -0700
danielbgto <notifications@github.com> wrote:

> blueman-2.1-0.8.gita0408c1.fc24.x86_64.rpm

Thanks, I have installed this build and it works okay. Nice find
there ;)

-- 
Sent from claws-mail powered by Fedora 24 MATE-Compiz | Please don't
print this email unless absolutely necessary

danielbgto, 2016-06-27:
Actually i'm seeing now that you can find the official Fedora build of the same version at https://kojipkgs.fedoraproject.org/packages/blueman/2.1/0.8.gita0408c1.fc24/x86_64/blueman-2.1-0.8.gita0408c1.fc24.x86_64.rpm
I saw it now. Not tested by me (should work). I will keep the package compiled by me. (Tired of compiling and reinstaling :) )

arehtykitna, 2016-06-28:
That's what I installed after following the link you posted to koji, I
didn't download yours in the end. Yes it works okay. I am using yumex
as package manager so can avoid installing subsequent blueman updates
should someone take on maintenance again. At present it offers
2.0.4-2.I'm considering setting a package exclude
in the yum configuration file though to keep this unreleased github
build because I know I can rely on it.

davidgfnet, 2016-06-28:
Hey there! I haven't upgraded my fedora 23 to 24, but after a regular dnf update the blueman applet would refuse to enable bluetooth. I had to downgrade to blueman-2.1-0.3.git7a2e20e.fc23.x86_64 again to make it work. Have you guys spotted the bug so far?

danielbgto, 2016-06-29:
davidgfnet, many of us, which are using bluetooth spotted that bug. Version 2.0.4 is not "compatible" with current environment of F23 and F24. Looks like a permission problem between blueman, bluetoothd, dbus and obexd and PackageKit. I'm definitely not a software "guru" to redefine where and how to solve it but still there is workaround which you found by yourself - downgrading to older version is solving the problem (still). You may use a dnf (yum) versionlock or exclude package upgrade direct from dnf conf. Hoping the distribution software engineers will take a look and solve it. Similar bug reports are included for Ubuntu 16.04 and looks like this is not Fedora only related bug.

aries1980, 2016-06-30:
I rebuilt the F23 2.1.0 Rawhide rpm.

[blueman-2.1-0.8.gita0408c1.fc24.x86_64.rpm.zip](https://github.com/blueman-project/blueman/files/341611/blueman-2.1-0.8.gita0408c1.fc24.x86_64.rpm.zip)
[blueman-debuginfo-2.1-0.8.gita0408c1.fc24.x86_64.rpm.zip](https://github.com/blueman-project/blueman/files/341612/blueman-debuginfo-2.1-0.8.gita0408c1.fc24.x86_64.rpm.zip)

Remove the .zip ending from the files and install them. It might complain the 2.0.4 is newer than the one you want to install, then install it with `rpm -Uvh --force` .

davidgfnet, 2016-07-11:
Hasn't this been fixed in F24 yet? Freaking annoying...

arehtykitna, 2016-07-11:
A little, however blueman has been orphaned as mentioned in this
thread so it isn't surprising. The best build from the Fedora Project's
Koji server for my needs is blueman-2.1-0.8.gita0408c1 (I've not tried
Janos' rebuild as this Koji one works okay for Obex in both directions).
Obex is most reliable with that one.

http://koji.fedoraproject.org/koji/buildinfo?buildID=716747

akostadinov, 2016-07-19:
Here's my build (time to live 7 days):
http://wikisend.com/download/120600/blueman-2.1-0.7.gitf810c73.fc24.x86_64.rpm

As @infirit pointed, you can build according to #422 but see my last comment for 2 more details.

works for me. Unfortunately I have no idea what the adapter is. Probably it is same hardware as the wireless: 03:00.0 Network controller: Intel Corporation Wireless 8260 (rev 3a)

akostadinov, 2016-10-28:
FYI wrt file transfer question, here's how to get it working on fedora:
https://ask.fedoraproject.org/en/question/94337/bluetooth-isnt-working/

Pasting answer here although would be better to upvote the soul finding the solution:
```
sudo dnf install bluez-obexd
systemctl --user start obex
sudo systemctl --global enable obex
```

infirit, 2016-10-28:
So, bluetooth is started and working but let's cripple the machine by not enabling obex transfers, how very user friendly.. :disappointed:

arehtykitna, 2016-11-01:
yes, it's frustrating. I also posted a guide [here](http://www.forums.fedoraforum.org/showthread.php?t=310326)

cschramm, 2016-11-01:
Feel free to continue discussing Fedora-related issues and solutions here. I'll just set this to closed as to my understanding there's nothing to do from the blueman side (at the moment).

closed 2016-11-01 by cschramm

akostadinov, 2016-11-02:
Created a bugzilla issue. Everybody that cares about fedora support could whine in there:
https://bugzilla.redhat.com/show_bug.cgi?id=1391218

arehtykitna, 2016-11-02:
On Wed, 02 Nov 2016 12:48:54 -0700
Aleksandar Kostadinov <notifications@github.com> wrote:

> Created a bugzilla issue. Everybody that cares about fedora support
> could whine in there:
> https://bugzilla.redhat.com/show_bug.cgi?id=1391218
> 

this is my last comment on the matter here.

F25 MATE-Live ISO has bluez-obexd and symbolic links to the
service appear to be present (not tested, they were installed on my
desktop by the Live ISO). it may be worth checking the F25 XFCE image
for the same. my notebook is staying a release behind as always.

historically for file access and transfer to/from my phone I typically
install the fuse-obexftp package as well. it may be surplus to
requirements nowadays but since it is only 25k it can't hurt having it.

the adaptor power control policy for bluez has definitely been fixed
(that policy change also rolled out to F24). you can
get updated ISOs for F24 with the fixed policy from
http://alt.fedoraproject.org/pub/alt/live-respins/