blueman-project/blueman #538: Connection Failed: blueman.bluez.errors.DBusNotReadyError: Resource Not Ready

giacecco, 2016-07-02:
After waking up my Fedora 23 from suspend, I always find this error message when opening _blueman-applet_: ```Connection Failed: blueman.bluez.errors.DBusNotReadyError: Resource Not Ready```. I then restart the bluetooth service, re-open the applet again and all works fine. 

I don't know enough of the Linux bluetooth-blueman-bluez stack to tell where the issue is, can you give me a few hints to help debug the issue?

I'm running an up-to-date Fedora 23, kernel 4.5.7-200 on a Lenovo Thinkpad T450s. _blueman-applet_'s version is 2.0.4. 

Thanks.

infirit, 2016-07-02:
Restarting the applet with perhaps the "auto power" option set should be enough.

My F24 vm with blueman master gave me the output in this gist [1]. The interesting bit is ``/org/bluez/hci0 Powered False `` at the end which indicates something powered the adapter down, not sure it it was blueman.

@giacecco, can you also post the output from ``blueman-applet`` from before the suspend and then until after and you run into that problem.

[1] https://gist.github.com/infirit/1364bc5f6c70b0896ef3919d3492c503

infirit, 2016-07-03:
Calling ``hciconfig hci0 down`` will show up as below in the log. Notice the ``UpdatePowerState`` shows up before the property changed.

```
UpdatePowerState (/usr/lib64/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:154)
off False 
foff False 
on True 
current state True 
new state True 
_________
_on_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/PropertiesBase.py:38)
/org/bluez/hci0 Discovering False 
_________
_on_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/PropertiesBase.py:38)
/org/bluez/hci0 Class 0 
_________
_on_properties_changed (/usr/lib64/python3.5/site-packages/blueman/bluez/PropertiesBase.py:38)
/org/bluez/hci0 Powered False 
```

closed 2016-09-06 by cschramm

alex700, 2020-04-14:
I have the same issue on Ubuntu 18.04.4

marcelocampelo, 2021-05-23:
In my case, restarting was not doing anything. I've realized that the bluetooth dongle was on a USB Hub, and It would stop working only when there was another USB Port being used. 
Everything is working fine after i've put the dongle directly on the Motherboard.

hazeled, 2021-10-10:
Still have this issue, almost nothing fixes it except a restart

faelpinho, 2023-04-30:
I got the same problem. No solution yet?
Since I bought a new laptop, it was worked until few minutes ago. Now its rarely to connect a new device.
I tryed to reset services and all settings, but the bluetooth only works again if I boot another OS from SSD or USB.