blueman-project/blueman #541: Unclosed bold tags on Gnome applet

andreyluiz, 2016-07-03:
Hello.

I see an unclosed bold tag in the gnome applet.

[image]

My blueman version follows.

[image]

The version seems to be the latest. My Ubuntu is pt_BR. So I checked the translations for blueman but I had no success finding such a clue.

Thank you.

andreyluiz, 2016-07-03:
Found it. It is in pt.po. Not pt_BR.po. I'll submit a Pull request.

closed 2016-07-03 by andreyluiz

infirit, 2016-07-03:
Thanks, PR is welcome but it also needs to be updated at Transifex otherwise it will be overwritten next time we update the translations for 2.1.

Please do the PR for the 2-0-stable branch :smile: