blueman-project/blueman #546: blueman-manager fails to start

sebastiencs, 2016-07-05:
I just compile blueman from master to install it.
When I try to run blueman-manager, I get this error: 
```{r, engine='sh', count_lines}
$ blueman-manager 
Traceback (most recent call last):
  File "/usr/bin/blueman-manager", line 237, in <module>
    manager = Blueman()
  File "/usr/bin/blueman-manager", line 46, in __init__
    super(Blueman, self).__init__(self, title=_("Bluetooth Devices"))
  File "/usr/lib64/python2.7/site-packages/gi/overrides/__init__.py", line 326, in new_init
    return super_init_func(self, **new_kwargs)
TypeError: could not convert value for property `type' from Blueman to GtkWindowType
```

blueman: master (last commit: 35992a7)
BlueZ: 5.39
Distribution: Fedora 23 x86_64
Desktop environment: Cinnamon

infirit, 2016-07-06:
Does reverting https://github.com/blueman-project/blueman/commit/b97a74839d46cd83d09b78a623a8d64f8bfeff41 fix the problem?

sebastiencs, 2016-07-09:
Yes reverting this commit fix the problem

infirit, 2016-07-09:
Please try the below patch, it removes self from the line with super().

```diff
diff --git a/apps/blueman-manager.in b/apps/blueman-manager.in
index 054defc..5c7a721 100755
--- a/apps/blueman-manager.in
+++ b/apps/blueman-manager.in
@@ -43,7 +43,7 @@ signal.signal(signal.SIGINT, signal.SIG_DFL)
 
 class Blueman(Gtk.Window):
     def __init__(self):
-        super(Blueman, self).__init__(self, title=_("Bluetooth Devices"))
+        super(Blueman, self).__init__(title=_("Bluetooth Devices"))
 
         self._applet_sig = None
```

infirit, 2016-07-11:
Pushed the above as a9605587f45bce85cbaeb4e863ea18fe7d9ab822

closed 2016-07-11 by infirit