blueman-project/blueman #547: blueman.bluez.errors.DBusFailedError: Protocol not available...

Phil171, 2016-07-10:
Hello everybody,
I experience troubles when trying to connect a handset Philips SHB7000 on my Dell Vostro 1510 powered with Xubuntu 16.04 newly installed and up to date.
I'm using Blueman-manager V2.0.4
Handset is recognized, I see its MAC Adress but when trying to pairing it as Headset, Handsfree, peripheral is added but no connexion.
I've check all links related here about A2DP, troubleshootings, blueman-project and all I have found about, no solution.
I've installed PulseAudio in case of, nothing more positive.
Blueman-manager returns the error : blueman.bluez.errors.DBusFailedError: Protocol not available...
The headset works fine on my Samsung Galaxy SG3, on a Dell 3700...
Any idea ??

Please, Try to be simple when helping me, I'm new in Xubuntu, Linux, and come from W7.
And last but not least, I'm French.
Best Regards,
Philippe from France

cschramm, 2016-07-11:
You're missing pulseaudio-module-bluetooth which picks up a connected audio device.

closed 2016-07-11 by cschramm

Phil171, 2016-07-11:
Thanks cschramm, 
After reading your answer,I have  installed this module in Terminal, same trouble, same error message.
I have tried with another handset, a Philips SHB7250, same problem.
I'm searching if that is ME who is wrong  :bad use of blueman software or making same error to connect my device.

infirit, 2016-07-11:
The steps are first: run ``sudo apt-get install pulseaudio-bluetooth`` from a terminal and second: ``killall pulseaudio`` or restart your session, then try again. Works for me when I tried this.

Phil171, 2016-07-12:
OK.
If I run : `sudo apt-get install pulseaudio-bluetooth` Terminal returns :
`Impossible to find paquet pulseaudio-bluetooth`.
Then i run : `sudo apt-get install pulseaudio-module-bluetooth`, Terminal returns :
`pulseaudio-module-bluetooth is already the newest version (1:8.0-0ubuntu3)` 
Because I installed it yesterday.
I'll try by 'killall pulseaudio' and come back.
I'm on Xubuntu 16.04

Phil171, 2016-07-12:
It works, **thanks**, infirit !
But it seems I have to select manually output in Volume Control.
There is no "automatic switch", I'll continue to investigate to simplify this operation and the way of use is a bit different than W7 where all is automatic.

infirit, 2016-07-12:
That may be fixed by editing a config file, taken from https://wiki.debian.org/BluetoothUser/a2dp
>In order to auto-connect a2dp for some devices, add this to /etc/pulse/default.pa
>>load-module module-switch-on-connect

Phil171, 2016-07-13:
Thank you.
I have to know where to look, what to apply, understand, and more in a foreign language, nice !!
;-)
On that side, Windows 7 is more within the reach of all, it looks like Linux and derivative does not try to make it simple for the novice.
I will read all this, digest and see if it works.
I have so many problems with Xubuntu I did not have with W7 it gets annoying!
Thank you very much for your help !

infirit, 2016-07-13:
I don;t want to turn this into a full debate and this will be the only thing, but here goes :smile: 

>On that side, Windows 7 is more within the reach of all, it looks like Linux and derivative does not try to make it simple for the novice.

It depends on what you are trying to do. Bluetooth Audio under debian and derivatives (*ubuntu) is problematic because you need to install some packages before it is working properly. Other distributions do not have this but have other limitations. Typically it is good to go to the wiki for information and forums of your distribution and post your questions there.

>I have so many problems with Xubuntu I did not have with W7 it gets annoying!

Yes, gnu+linux is not windows, do not apply your windows expectations onto a completely different operating system :wink:. There will be things that work much better and other things that are relatively hard, but if you are willing to learn you will see there is so much more to gnu+linux than you can imagine. For example I _have_ to work with windows, and keep trying to move and resize windows with alt+left/right mouse click, this does not work and is very annoying for me to use windows.

>And I don't know how to modify /etc/pulse/default.pa, I don't have right for... and don't want to do a mistake with "chown" command.

Make a backup (safe a copy to your home dir), then go to [this](http://www.howtogeek.com/howto/42980/the-beginners-guide-to-nano-the-linux-command-line-text-editor/) tutorial and learn how to edit files with ```nano```. And then edit the file with ``sudo nano   /etc/pulse/default.pa`` and just try it. No reason to be scared the worst that can happen is that your audio stops working, but then you have a copy of the original to go back to.

If you run into problems in general, the distribution forums are a great resource :+1:.

Phil171, 2016-07-13:
Yes, you're completely right about difference between W7 and Linux and all I'm trying to understand, it is a bit more different and complicated for me. Packages names are not easy to find or to understand.

Now, it seems working but not as yesterday.
I've modified (and cross my fingers before!) default.pa file.
Now, it works but I have to start manually Bluetooth and activate switch to ear with headset.
It should be normal to manually start Bluetooth module but I have to learn more to connect automatically when  BT device is connected.
Second problem is I don't have A2DP activated, sound is poor.
I return to search for right forums to correct this problem.

Thanks a lot for your help !
Have a good day,
Philippe

texadactyl, 2017-04-12:
Installing  pulseaudio-module-bluetooth followed by a reboot worked for me too. Thank you.  

Now, my obly puzzle is why didn't my *buntu distro somehow apprised me of this so I wouldn't have to engage in guessing games!  (-:

ed-fruty, 2017-12-13:
@infirit, Thnx! ;)

nekromoff, 2018-12-13:
Still present on the latest Ubuntu

sjpatel21, 2019-02-28:
> Oddly, I had this working on lubuntu out of the box. Then it failed. And now I'm fooling around again ;)

Exactly the case with me too. I am using UbuntuMate 18.04. It worked fine initially for few months and it doesn't connect at all to my headphone.

CarnageKeith3301, 2019-07-08:
I'm running it on BackBox 6 (based off of Ubuntu 18.04 LTS), and what fixed it for me was apt-get update && apt-get upgrade. then installing pulseaudio and all that shit. and finally rebooting

evgenijrenke, 2020-01-22:
> > Oddly, I had this working on lubuntu out of the box. Then it failed. And now I'm fooling around again ;)
> 
> Exactly the case with me too. I am using UbuntuMate 18.04. It worked fine initially for few months and it doesn't connect at all to my headphone.

same here

fgmehlin, 2020-01-23:
from https://wiki.debian.org/BluetoothUser

Loading the module was the trick for me

`pactl load-module module-bluetooth-discover`

AlexCuan, 2021-12-05:
> The steps are first: run `sudo apt-get install pulseaudio-bluetooth` from a terminal and second: `killall pulseaudio` or restart your session, then try again. Works for me when I tried this.

Greetings from 2021. Your answer helped me a lot, thank you from the bottom of my heart