blueman-project/blueman #549: ManagerDeviceMenu: Networkaccess does not update when disconnecing

infirit, 2016-07-11:
Documenting so we do not forget.

In current master, the menu entry does not update and keeps showing "Connected". It appears when we regenerate the menu the service is returning itself as connected but in fact it is not physically.

From a quick check it appears that bluez.Network return Connected True but further investigating is needed.

infirit, 2016-07-11:
Well, this is weird, below is a idle session where I connect and disconnect but I only ever see the property change when connecting. I am not sure we are at fault..

```python
>>> from gi.repository import GLib, Gio
>>> proxy = Gio.DBusProxy.new_for_bus_sync(Gio.BusType.SYSTEM,
            Gio.DBusProxyFlags.NONE, None, 'org.bluez',
            '/org/bluez/hci0/dev_6C_83_36_B9_C6_D6', 'org.bluez.Network1')
>>> def on_g_properties_changed(proxy, changed, invalid):
	print(changed, invalid)

	
>>> proxy.connect('g-properties-changed', on_g_properties_changed)
9
>>> loop = GLib.MainLoop()
>>> loop.run()
{'Connected': <true>, 'Interface': <'bnep0'>, 'UUID': <'00001116-0000-1000-8000-00805f9b34fb'>} []
{'Connected': <true>, 'Interface': <'bnep0'>, 'UUID': <'00001116-0000-1000-8000-00805f9b34fb'>} []
```

cschramm, 2016-07-11:
I *do* see property changed in every case, i.e. 2.0.4, in master, and with your test code (different object path of course):

```
{'Connected': <true>, 'Interface': <'bnep0'>, 'UUID': <'00001116-0000-1000-8000-00805f9b34fb'>} []
{'Connected': <false>} ['UUID', 'Interface']
```

Did you replace your Gentoo with that strange, broken Fedora thingy? :wink:

infirit, 2016-07-12:
Haha, wel, I do run unstable version of Gentoo and I do tend to forget I have to restart my session/machine once in a while :smile:  so things may have gotten confused. Thanks for checking you do see the update :+1:

infirit, 2016-07-12:
I like to keep this one open for now to make sure we verify all is well when it is alpha/beta time.

infirit, 2016-12-31:
This is not a problem for me anymore, closing.

closed 2016-12-31 by infirit