blueman-project/blueman #558: Blueman does not support setting up a local A2DP receiver

emempi, 2016-07-23:
The 'Local services' menu does not contain any way to set a local A2DP receiver and a quick skim through the code does not reveal any support in the current master, although googling reveals there used to be some support for this (setup: Debian, blueman package version is 2.0.4-1).

emempi, 2016-07-24:
According to https://thecodeninja.net/2016/06/bluetooth-audio-receiver-a2dp-sink-with-raspberry-pi/ , the simplest thing to enable local audio sink might be enabling the 'Audio' bit (0x200000) in the "Class of Device".

infirit, 2016-07-25:
I am not sure we should be modifying this from blueman. We interact with bluez through dbus and the ``Class`` property is readonly. I'll mark it as Feature request for now and wait for @cschramm to comment but I am not so keen on this myself.

cschramm, 2016-07-29:
This works out of box for me with bluez 5.36, pulseaudio 9.0, a Nexus 7, and a device class of 0x1e010c (i.e. no audio bit).

> the simplest thing to enable local audio sink might be enabling the 'Audio' bit (0x200000) in the "Class of Device".

That does not enable anything. It might be that your source device expects to find the Audio bit and filters / ignores / rejects the device otherwise, but I cannot think of any other effect.

> googling reveals there used to be some support for this

BlueZ 4 used to have /etc/bluetooth/audio.conf where you needed to enable `Source`s. Blueman supported that but it is not relevant for BlueZ 5.

closed 2016-09-06 by cschramm

emempi, 2016-09-12:
Yes, looks like that's the case. I tested everything once more and even when I paired successfully, after disabling the audio class bit, I was not able to play audio anymore. Unfortunately, my device is a generic Android phone with a stock image, so I believe this might affect a pretty wide range of devices. Apologies for delay, I'll try to provide more information shortly.

emempi, 2016-09-19:
So, I tested few user pair-and-playback scenarios; one device was always my debian-running htpc with bluez speakers and permanently enabled discovery; the other one was was either android nexus phone, a chromebook or a linux/gnome laptop. I unpaired the devices prior to any testing.

1) hci0 class 0x1c0104 (or whatever the default is, with no audio major service class bit set)
nexus phone: paired, but was not able to connect and start audio playback
chromebook: I was able to connect and play audio
linux laptop/gnome bluetooth: pair ok, did not allow me to connect to the device

2) hci0 class 0x3c0104 
nexus phone: after pairing it connected automatically, I was able to play music
chromebook: paired, playback ok
linux laptop/gnome bluetooth: pair ok, did not allow me to connect to the device

I believe the linux laptop might have its own misconfiguration issues, but the other results should be representative. Unfortunately, I don't have any iDevices to compare with.

I read https://www.bluetooth.com/specifications/assigned-numbers/baseband , which at one point says that one should not assume any available services based on major or minor device class. This is a little strange (or intentional?), because it does not mention Major Service Classes and thus it's up to the reader to interpret this.

emempi, 2016-09-19:
If you need any more info to reopen this issue, please let me know. If you feel Android is to be blamed, I'll gladly file the bugreport against it, but I don't feel I have enough arguments (one vague possible misinterpretation in bt docs). If there is any authoritative BT spec stating that Major Service Class is to be ignored for any service discovery purposes, please point me there.

infirit, 2016-09-19:
>which at one point says that one should not assume any available services based on major or minor device class

Rightly so, a device may belong to a certain class but may only expose a subset of possible services.

>This is a little strange (or intentional?), because it does not mention Major Service Classes and thus it's up to the reader to interpret this. 

It is intentional. UUID's are used to expose services on a device, see https://github.com/blueman-project/blueman/blob/master/blueman/Sdp.py

I don't like messing with this, if you want to do lowlevel, device specific changes like this it is up to you to manager. My suggestion would be a udev rule that runs hcitool, smililar to [this](https://github.com/blueman-project/blueman/wiki/Troubleshooting#turn-on-bluetooth-device-on-boot)

emempi, 2016-09-19:
Yes, I've been already planning such a hack as a temporary measure, but it brings its own pile of problems. What I would really like to do is to have this fixed correctly, be it the fact that there already have been bugs for similar issues, the fact that blueman behaves differently where it should not, ... . Under what conditions would be a branch fixing this issue mergeable (in case I would find some spare time to implement it)?

infirit, 2016-09-19:
>be it the fact that there already have been bugs for similar issues

One for the same thing, if there are other in your opinion which?

>the fact that blueman behaves differently where it should not

How and compared to what does it behave differently? Blueman exposes what the bluez dbus api provides, nothing more nothing less.

As I explained before, the uuids determine what services are discovered not the device class. Neither of them are writeable in the bluez dbus api.

emempi, 2016-09-20:
Issues similar to this one were e.g. https://github.com/blueman-project/blueman/issues/534 ; the other ones I was thinking about were generally about a2dp bluetooth support and don't apply here; apologies.

Behaving differently -- from user perspective some devices are able to play audio and some are not. If neither the device class nor the uuids are writable by blueman, nothing can be done here -- I agree it would be a pretty low-level hciconfig hack and this should be really fixed elsewhere. Thanks a lot for all the information. :)

cschramm, 2016-09-20:
So you're saying you can only stream audio from a Nexus device with an Android system to a Debian system if the Debian system's controller has the audio bit set?

As I said, streaming from a Nexus 7 worked fine when I last tried it with bluez 5.36, pulseaudio 9.0, and a default device class of 0x1e010c on a Debian system as well, so this cannot be a generic issue. My Android version is 6.0.1 if that helps.

infirit, 2016-09-20:
@cschramm the problem is, as far as understand it, device specific, in this case the raspberry pi. Things work fine for me as well on a normal pc.

@emempi.
>the other ones I was thinking about were generally about a2dp bluetooth support and don't apply here

Debian and many derivatives do not install the required packages to use a2dp. Most of the issues raised here are because of this.

>from user perspective some devices are able to play audio and some are not

With bluez5 the audio handling was dropped and had to be re-implemented in Pulse. With this change normally no configuration should be required as long as the correct services are exposed. When I was browsing the internets yesterday I came across [1] which may be of use to automate all this stuff on your rpi. Direct link to the scripts [2] :smile:.

[1] http://www.instructables.com/id/Raspberry-Pi-BluetoothAirplay-Audio-Receiver-combo/
[2] https://bitbucket.org/ehsmaes/raspberry-pi-audio-receiver-install/src/3be0c3884a9a08521e2007cce4b573f0aa631957?at=master

emempi, 2016-09-20:
I updated Debian (testing) on my device yesterday and everything started working like a charm. Unfortunately, I could not find not changelog note for version change from bluez-5.36+b1 to bluez-5.36+b2 (which I'm running right now). So, I guess thanks a lot for all the help and patience :)

@cschramm right now I'm running the same setup as you do, with the only difference that I don't have the 'networking' bit set in the device class. Unfortunately, I was not keen enough to get a snapshot of exported services yesterday and today, so I have no idea what could have changed.

@infirit I'm not running a rpi :) . The htpc is running on x86, Debian testing; so the stuff does not apply to me directly. On the other hand, the install scripts from [2] are nice documentation for all the rpi changes; thanks a lot for that :) Also, next time this stuff breaks, at least I'll know that the bluez pulseaudio module is to blame. :)