blueman-project/blueman #562: TypeError when attemting to use A2DP

yars068, 2016-07-27:
HI!
```
RequestPowerState (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/PowerManager.py:127)
Requesting True 
_________
on_power_state_change_requested (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:120)
True 
_________
Run (/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py:208)
Function on_power_state_change_requested on KillSwitch Failed 
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/blueman/main/PluginManager.py", line 212, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py", line 129, in on_power_state_change_requested
    Mechanism().SetRfkillState('(b)', state, result_handler=reply, error_handler=error)
  File "/usr/lib64/python2.7/site-packages/gi/overrides/Gio.py", line 152, in __call__
    raise TypeError('first argument must be the method signature string: %r' % signature)
TypeError: first argument must be the method signature string: u'(b)'
_________
check (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/PowerManager.py:115)
callbacks done 
```
This is under python 2.7.11, Slackware64-14.2.

infirit, 2016-07-28:
Confirmed under python 2.7 here, doesn't make any sense though so I am properly confused :confused:

infirit, 2016-07-28:
It appears to be a weird unicode/gobject problem. Below works but is imho a bit ugly.
```diff
diff --git a/blueman/plugins/applet/KillSwitch.py b/blueman/plugins/applet/KillSwitch.py
index 2340f0b..fa8cecf 100644
--- a/blueman/plugins/applet/KillSwitch.py
+++ b/blueman/plugins/applet/KillSwitch.py
@@ -126,7 +126,7 @@ def reply(*_):
         def error(*_):
             cb(False)
 
-        Mechanism().SetRfkillState('(b)', state, result_handler=reply, error_handler=error)
+        Mechanism().SetRfkillState(str('(b)'), state, result_handler=reply, error_handler=error)
 
     def on_query_status_icon_visibility(self):
         # Force status icon to show if bluetooth is soft-blocked
```

yars068, 2016-07-28:
Yes, it works, but maybe I should try another pygobject version (currently is used pygobject3-3.18.2)?
>  Below works but is imho a bit ugly.

I agree, but I not familiar with Python and I can't propose a better solution than you. Sorry :(
Also, there is another similar issues, with a similar reasons, so supporting two different python versions is a bit complicated... For example, see [http://pastebin.com/y77AtEqa](url), this appears when I tried to disconnect my headset (left click on tray icon, right click on the headset's item, select Disconnect).

infirit, 2016-07-28:
>maybe I should try another pygobject version

Nah, I have it also and I know why it happens.

>I agree, but I not familiar with Python and I can't propose a better solution than you. Sorry :(

No need to say sorry :smile:, we decided to keep support both python versions. The main problem is that pygobject under python2 expects bytes and can not handle codepoints. We have similar solutions in other places to deal with this as well.

I'll push a patch that adds ``str()`` around these in the coming days.

yars068, 2016-07-28:
So please test and review: 8070395.

infirit, 2016-07-28:
There are a few more places to check and fix. See below.

Please make the commit subject: ```Call str() on dbus signatures for new GDBus AppletService and Mechanism proxies```

Thx

```bash
$ grep -R -e "main\.AppletService" -e "main\.Mechanism" blueman/ apps/ module/
blueman/services/meta/SerialService.py:from blueman.main.Mechanism import Mechanism
blueman/plugins/services/Transfer.py:from blueman.main.AppletService import AppletService
blueman/plugins/services/Network.py:from blueman.main.Mechanism import Mechanism
blueman/plugins/services/Network.py:from blueman.main.AppletService import AppletService
blueman/plugins/applet/PPPSupport.py:from blueman.main.Mechanism import Mechanism
blueman/plugins/applet/Networking.py:from blueman.main.Mechanism import Mechanism
blueman/plugins/applet/KillSwitch.py:from blueman.main.Mechanism import Mechanism
blueman/plugins/applet/DhcpClient.py:from blueman.main.Mechanism import Mechanism
blueman/plugins/manager/Services.py:from blueman.main.AppletService import AppletService
blueman/Functions.py:from blueman.main.AppletService import AppletService
blueman/gui/manager/ManagerDeviceMenu.py:from blueman.main.AppletService import AppletService
apps/blueman-assistant.in:from blueman.main.AppletService import AppletService
```

yars068, 2016-07-29:
Done, also rebased to single commit.

infirit, 2016-07-29:
Thanks, I think something did not go ok. I put your work in https://github.com/infirit/blueman/commit/f1b649cfdef871775cfec092b4c0da93be978a6a and credited you in the commit message (hope you don't mind). This should be all of them :smile:

yars068, 2016-07-29:
Ok, thanks you :)

infirit, 2016-07-29:
Merged, thanks for reporting. Let me know if you run into more issues like this.

closed 2016-07-29 by infirit