blueman-project/blueman #563: Clicking on blueman applet doesn't work on KDE Plasma 5

alive4ever, 2016-07-30:
I am having issue with blueman applet on KDE Plasma 5 desktop (5.7.2).

The applet is shown (started via xdg autostart) but clicking the applet doesn't give any response. So, the only way to quit blueman-applet is to issue `pkill -TERM blueman-applet` from command line.

Although it's not affecting bluetooth functionality since plasma desktop has its own bluetooth indicator, it would be better for blueman to function with different desktop environment. The only way to tell if blueman applet is functional is by clicking the applet to show applet's context menu and it doesn't work here on KDE Plasma 5 desktop.

[image]

infirit, 2016-07-30:
Yes, I noticed this too. Below change to the autostart file shoud stop it from loading.

@cschramm, any objection adding this by default?

```diff
diff --git a/data/blueman.desktop.in b/data/blueman.desktop.in
index 945c972..2eb28d2 100644
--- a/data/blueman.desktop.in
+++ b/data/blueman.desktop.in
@@ -5,4 +5,5 @@ Icon=blueman
 Exec=blueman-applet
 Terminal=false
 Type=Application
+NotShowIn=KDE
 Categories=
```

cschramm, 2016-07-31:
I'm not 100 % sure what you lose if you do not have the applet. I think it's at least the PowerManager feature (which means you cannot turn bluetooth off via blueman) and the features of the NetUsage and RecentConns plugins.

Did you try turning the AppIndicator plugin on / off?

infirit, 2016-07-31:
With appindicator plugin it does kinda works but not like it does with unity or xfce. We do now have a menu on right click but left click does nothing.

But I am not looking to fix this any time soon (or at all) for the simple reason that KDE comes with it's own bluetooth manager/applet.  So if someone want to pick this up who uses blueman under kde have at it :smile:

cschramm, 2016-08-01:
> We do now have a menu on right click but left click does nothing.

I think that's intentional as libappindicator does not have the concept of left and right clicks. All you do is set a menu and it's up to the implementation when and how to show that menu. In Xfce I get it on both left and right click. I guess KDE Plasma 5 does not have the concept of a right click (menu) on the status icons at all or are there other icons that do have some right click action?

I'm not sure why our StatusIcon does not work. All we do is connect to the popup-menu (right click) and activate (left click) signals of the GtkStatusIcon, so the conclusion would be that GtkStatusIcon is not compatible with KDE Plasma 5 which sounds strange...

Anyway, a solution to this is that the package maintainer or user enables the AppIndicator plugin if blueman is to be used with KDE Plasma 5 just as they need to do it for e.g. Unity.

cschramm, 2016-08-01:
> the conclusion would be that GtkStatusIcon is not compatible with KDE Plasma 5 which sounds strange...

Well, it's actually not *that* strange if you take into account that GtkStatusIcon got deprecated in GTK+ 3.14.

alive4ever, 2016-08-02:
> Did you try turning the AppIndicator plugin on / off?

I didn't try turning AppIndicator on at all, because the plugin is not available (v2.04)
[image]

> Yes, I noticed this too. Below change to the autostart file shoud stop it from loading.

I deleted the blueman xdg autostart file to prevent applet loading on KDE Desktop. I just need to add blueman-applet on my window manager autostart file to use it instead of relying on xdg autostart.

cschramm, 2016-08-02:
Unless support was disabled when your version was built, that probably means that libappindicator3 or its GI bindings are not installed.

cschramm, 2016-10-05:
I think we can close this.

Recap:

StatusIcon does not work in KDE Plasma 5.

It's not clear why but it might be that it's just not compatible with GtkStatusIcon in general. As that class is deprecated, we do have an open task to port to GNotification (#306).

AppIndicator works fine and gets enabled by default, however it will only be available if libappindicator3 is available via the GI repository.

closed 2016-10-05 by cschramm

alive4ever, 2017-01-16:
I think it would be wise to disable blueman xdg startup on known desktop environment, such as Gnome or KDE, to prevent loading blueman applet.

The desktop file should be modified as @infirit shows
```
diff --git a/data/blueman.desktop.in b/data/blueman.desktop.in
index 945c972..2eb28d2 100644
--- a/data/blueman.desktop.in
+++ b/data/blueman.desktop.in
@@ -5,4 +5,5 @@ Icon=blueman
 Exec=blueman-applet
 Terminal=false
 Type=Application
+NotShowIn=KDE
 Categories=
```

cschramm, 2017-01-16:
There's no reason to not start blueman-applet on KDE. To my knowledge the AppIndicator plugin works fine. It remains unknown why you do not have it. Possible reasons, as mentioned, include that it was disabled on build or libappindicator3 or its GI bindings are not available.