blueman-project/blueman #565: NetUsage NM serial device was removed in 0.9

infirit, 2016-08-02:
https://developer.gnome.org/NetworkManager/1.0/ref-migrating.html#id-1.4.3.12

I personally do not have a DUN phone so it's hard for me to port this part of the plugin to GDBus. But IMO we should fix this before we release the 2.1 final.

infirit, 2016-08-03:
Looking at the NM dbus api there is no such thing as PppStats anywhere to be found...

cschramm, 2016-08-04:
There is still some stats stuff in the current NM code, but I cannot find a way to access it (also via glib). We could read it ourselves using ioctl, though:

```
        struct ifreq req;
        struct ppp_stats stats;

        memset (&req, 0, sizeof (req));
        memset (&stats, 0, sizeof (stats));
        req.ifr_data = (caddr_t) &stats;

        strncpy (req.ifr_name, priv->ip_iface, sizeof (req.ifr_name));
        ioctl (priv->monitor_fd, SIOCGPPPSTATS, &req);
```

infirit, 2016-08-04:
Good, I don't have a phone like that so no means of working on this.

infirit, 2018-08-31:
These statistics are exposed in dbus [org.freedesktop.NetworkManager.Device.Statistics](https://developer.gnome.org/NetworkManager/stable/gdbus-org.freedesktop.NetworkManager.Device.Statistics.html)

Now that I have a way of testing dun when I have some time I'll update NetUsage to use this interface.

infirit, 2018-09-01:
However org.freedesktop.NetworkManager.Device.Statistics requires a dun connection to be created in NM properly like I did in #910. I am not sure but I suspect it will not be available if we use PPPSupport unless NM has a connection for the dun device.

Also, rfcomm does not show up in /sys/class/net/ (as one would expect as its not a network device) so using the Monitor class will not work for DUN, only for PAN. What I'll probably do is some cleanup and drop support for DUN when using PPPSupport.

Actions items I have right now are:

 * For DUN with PPPSupport do not create a monitor just log an error that it is not implemented.
 * Drop storing a list of bluetooth addresses that have stats in gsettings, its pointless and likely something that was needed with GConf.
 * Use A Gio.FileMonitor reading the rx/tx_bytes for bnep interfaces
 * Implement a Gio.DBusProxy class that will use the nm statistics interface
 * Fix bugs along the way like never setting a log start time on first device use.

infirit, 2019-04-27:
@cschramm removed the remaining bits from NetUsage. But I do like to add support for [org.freedesktop.NetworkManager.Device.Statistics](https://developer.gnome.org/NetworkManager/stable/gdbus-org.freedesktop.NetworkManager.Device.Statistics.html) at some point. But not in 2.1 so removing milestone.