blueman-project/blueman #571: Blueman assuming wrong powered state

haarp, 2016-08-10:
blueman-git from Sunday (should be the latest)
bluez-5.39
Gentoo Linux

When blueman-applet starts, it tends to power on my Bluetooth adapter (BCM20702).

However if the applet is already running when the adapter appears (e.g. after a suspend, or after disabling/enabling it via hardware switch), the adapter defaults to being powered off. Blueman will pretend that it's on ("Bluetooth enabled").

This can be worked around by restarting blueman-applet or by issuing "power on" in bluetoothctl.

infirit, 2016-08-10:
Yes, there is something not getting set/read properly and we get confused.

infirit, 2016-09-10:
Documenting things as I go. When ``self.Applet.Plugins.Run("on_power_state_query", self)`` runs it appears to always return 2's regardless whether the adapter is powered or not, this is not right from what I can gather.

infirit, 2016-09-12:
Here is the basic problem, when the adapter (or when the ``Powered`` property changes) is removed  we do not update ``self.adapter_state`` before we run ``on_power_state_query``. Also the only place we handle this properly, ``on_manager_state_changed`` is not run as we do not restart bluez and the dbus name remains.

Below patch will fix ``adapter_state`` set properly but we then still need to auto-power the adapter as this is only done in ``on_manager_state_changed``. @cschramm any suggestions on autopowering, move it to ``UpdatePowerState``?

```diff
diff --git a/blueman/plugins/applet/PowerManager.py b/blueman/plugins/applet/PowerManager.py
index 9466c7e..c0da79d 100644
--- a/blueman/plugins/applet/PowerManager.py
+++ b/blueman/plugins/applet/PowerManager.py
@@ -211,6 +211,7 @@ class PowerManager(AppletPlugin):
                 dprint("adapter powered on while in off state, turning bluetooth on")
                 self.RequestPowerState(True)
 
+            self.adapter_state = value
             self.UpdatePowerState()
 
     def on_bluetooth_toggled(self):
```

haarp, 2016-09-17:
Thanks so far, I can confirm that the power state is now shown correctly.

infirit, 2016-09-17:
I pushed the change to stable (1ead551b17a90cf047e455677aee6340973cc0d3) and master (13998885aa31404fdbf97e45c77c77a49abcdf54). Left is to properly auto power the adapter so this issues has to remain open.

cschramm, 2016-09-20:
I think the condition for calling `RequestPowerState(true)` at that point should not be just `value` but `value or self.get_option("auto-power-on")` to auto-power a newly connected adapter.

infirit, 2016-09-20:
I'll rework this a to use the ``adapter-added`` signal on the ``Manager``. This is the right time to set the initial power state. Then ``on_properties_change`` can simply update ``self.adapter_state`` and the rest nothing more.

infirit, 2016-09-21:
Ok, this works nicely for me.

```diff
diff --git a/blueman/plugins/applet/PowerManager.py b/blueman/plugins/applet/PowerManager.py
index 83e5907..2e12db2 100644
--- a/blueman/plugins/applet/PowerManager.py
+++ b/blueman/plugins/applet/PowerManager.py
@@ -69,10 +69,7 @@ class PowerManager(AppletPlugin):
         if state:
             def timeout():
                 self.adapter_state = self.get_adapter_state()
-                if self.get_option("auto-power-on"):
-                    self.RequestPowerState(True, force=True)
-                else:
-                    self.RequestPowerState(self.adapter_state)
+                self.RequestPowerState(self.adapter_state)
 
             GLib.timeout_add(1000, timeout)
 
@@ -225,10 +222,10 @@ class PowerManager(AppletPlugin):
         adapter = Bluez.Adapter(path)
 
         def on_ready():
-            if not self.adapter_state:
-                adapter.set("Powered", False)
+            if self.get_option("auto-power-on"):
+                self.RequestPowerState(True, True)
             else:
-                adapter.set("Powered", True)
+                adapter.set("Powered", self.adapter_state)
 
         wait_for_adapter(adapter, on_ready)
```

cschramm, 2016-09-21:
Code looks fine, but the question is if we want to change that or not. The auto-power feature always related to the point where blueman gets started. Whether or not to enable bluetooth when a new adapter gets added is a different question.

I guess most users do not have that situation, so they do not care. Most users that do will probably want to use the newly connected adapter, so auto-powering makes sense. The third group, however, might expect disabled bluetooth to stay disabled no matter if they add an adapter or not. Overall, I support the change if we have a consent on that.

infirit, 2016-09-21:
Agree on the 3 points, just on the third, see below.

>The third group, however, might expect disabled bluetooth to stay disabled no matter if they add an adapter or not

Perhaps, to not annoy users that deliberately turned bluetooth off, we keep track if the user has manually turned bluetooth on or off and include this into the consideration? Something like https://github.com/infirit/blueman/commit/c2052c22e8ca478c24255248039007f30ec982ac

haarp, 2016-09-22:
I can confirm that the latest patch works, I can now disable the adapter through blueman-applet. After doing so, the icon disappears however, so I can not enable it again the same way. Is this intentional?

My hardware switch also makes the blueman-applet icon disappear and reappear when triggered.

cschramm, 2016-09-28:
@infirit: Sounds good.

@haarp: Fixed in 2a0282f. This would also have fixed the initial problem. :smile:

infirit, 2016-09-28:
So commit 2a0282f4e1fe27ddab0e5676e9023e3008144267 fixed everything, no need to move things around. On insertion the adapter is powered according to ``auto-power-on`` setting now.

Reverted the commits I pushed to update ``self.adapter_state``. This works for me properly now but @haarp please confirm that with @cschramm commit 2a0282f4e1fe27ddab0e5676e9023e3008144267 it works properly for you as well.

haarp, 2016-10-05:
I used the git version from a few days ago.

Hardware switch: Blueman disappears, reappears once hardware switch is back on
"Turn off" menu item: Blueman stays and may be reactivated
Suspend: Blueman returns in a working state

Looks like everything is in order :) Thanks a lot for your work!

closed 2016-10-05 by haarp