blueman-project/blueman #574: Audio sink with A/V remote control

nshopik, 2016-08-17:
blueman: 2.0.4
BlueZ: 5.37
Distribution: Xbuntu
Desktop: XFCE

I wonder if I miss something, I have external bluetooth speakers and they work perfectly fine if connected as audio sink with A2DP profile. But I've notice that volume control isn't sync if I use buttons on speakers. 

If I connect them as handsfree with HFP audio profile its volume controls now in sync, though quality isn't what I've getting via A2DP but If I switch to A2DP profile while it still connected as handsfree there won't be any audio. And `gmusicbrowser` won't even play song when you press play and timer will be sit still like i'm on pause.

So is it possible to use A2DP with A/V Remote Control profile, from my understanding this is only way to have sync volume control as handsfree isn't suppose to deliver high quality audio.

infirit, 2016-08-17:
Some background; Although blueman shows 2 connect entries in the menu for audio, technically there is no difference between the two under bluez5. This is on the todo list to have a generic connect method.

And the short disclaimer answer is that there is little we can do about this. Blueman does not provide any remote control functionality and just handles the connecting. The entry to switch audio profile only controls pulseaudio, you can do the same with the pavolumecontrol with the same results :smile:.

>I wonder if I miss something, I have external bluetooth speakers and they work perfectly fine if connected as audio sink with A2DP profile. But I've notice that volume control isn't sync if I use buttons on speakers

What do you mean with sync, there is a delay?

nshopik, 2016-08-17:
> Some background; Although blueman shows 2 connect entries in the menu for audio, technically there is no difference between the two under bluez5. This is on the todo list to have a generic connect method.

I'm not sure what 2 connect entries you are talking about. For me its 3, `headset` `handsfree` and `audio sink`

> What do you mean with sync, there is a delay?

I mean there is 2 volume controls, pulseaudio mixer and speaker itself. What I was expecting is have master volume control just like if I connect with handsfree with HFP audio profile. If I move slider on pulseaudio, master volume on speaker goes down too and vica versa. 

But with Audio Sink no remote control works, so I guess this is issue/feature with pulseaudio then, more specifically `pulseaudio-module-bluetooth` I guess

closed 2016-08-17 by nshopik

infirit, 2016-08-17:
>I'm not sure what 2 connect entries you are talking about. For me its 3, headset handsfree and audio sink

Sorry, I only have two for my phone, yes all of them are technically the same and use the generic bluez5 connect method. Connecting services separately is not possible any-more under bluez5.

>I mean there is 2 volume controls, pulseaudio mixer and speaker itself. What I was expecting is have master volume control just like if I connect with handsfree with HFP audio profile. If I move slider on pulseaudio, master volume on speaker goes down too and vica versa. 

I don't have a device like speaker but it probably then uses the old deprecated mediacontrol (dbus org.bluez.MediaControl1) api which does provide volume control, the new mediaplayer (dbus org.bluez.MediaPlayer1) api doesn't.

nshopik, 2016-08-17:
Yeah I've noticed that too reading doc https://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/media-api.txt seems weird deprecate stuff and doesn't provide replacement.

nshopik, 2016-08-17:
> Sorry, I only have two for my phone, yes all of them are technically the same and use the generic bluez5 connect method. Connecting services separately is not possible any-more under bluez5.

Yeah but for me it actually different, connecting as `headset` and select HFP profile will provide pulseaudio with AVRCP. But selecting `audio sink` and HFP profile doesn't not.

infirit, 2016-08-17:
>Yeah but for me it actually different, connecting as headset and select HFP profile will provide pulseaudio with AVRCP. But selecting audio sink and HFP profile doesn't not.

The key point and difference being the audio profile, the connection method of the device is identical in blueman. You _should_ notice that all of them show connected once connected and vice versa. See https://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/device-api.txt#n11 for more on the ``generic`` connect method.

nshopik, 2016-08-17:
Yeah you right. 
Thanks for hints :+1: I'm new to bluetooth stuff on Linux