blueman-project/blueman #580: KeyError: 'Class'

orf, 2016-09-01:
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
self._handler(*args, **kwargs)
File "/usr/lib/python3/dist-packages/blueman/bluez/PropertiesBlueZInterface.py", line 55, in wrapper
handler(name, value, **kwargs)
File "/usr/lib/python3/dist-packages/blueman/plugins/applet/GameControllerWakelock.py", line 36, in on_device_property_changed
klass = Device(path).get_properties()["Class"] & 0x1fff
KeyError: 'Class'

infirit, 2016-09-01:
You can work around this by disabling the ```GameControllerWackelock`` plugin in the plugins dialog.

However I doubt that that is why it is not pairing. Please see https://wiki.debian.org/BluetoothUser/a2dp to setup a2dp and if the problem persists follow https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

orf, 2016-09-01:
Hey, thanks for your reply @infirit. I'm not too fussed about the pairing, that boils down to an 'authentication failure' with the headset. Annoying, but others have experienced the same. I just wanted to report this error in the plugin.

I actually did disable the plugin in the menu item, but it still persisted. I had to move the `ameControllerWakelock.py` out of the directory and restart some stuff for it to work.

infirit, 2016-09-01:
Ok, we actually have implemented a pin database (uses the gnome db actually) that sends pre-defined pin codes in the master branch which eventually will be version 2.1. It may be of use for your problem.

The below patch should fix the problem, it would be nice if you could confirm it works for you :smile: .

ps: The Class property is optional so this is why the error happens.

```diff
diff --git a/blueman/plugins/applet/GameControllerWakelock.py b/blueman/plugins/applet/GameControllerWakelock.py
index 9d63cfa..9f86686 100644
--- a/blueman/plugins/applet/GameControllerWakelock.py
+++ b/blueman/plugins/applet/GameControllerWakelock.py
@@ -33,7 +33,11 @@ class GameControllerWakelock(AppletPlugin):
 
     def on_device_property_changed(self, key, value, path):
         if key == "Connected":
-            klass = Device(path).get_properties()["Class"] & 0x1fff
+            props = Device(path).get_properties()
+            if not "Class" in props:
+                return
+
+            klass = props["Class"] & 0x1fff
 
             if klass == 0x504 or klass == 0x508:
                 if value:
```

infirit, 2016-09-05:
Fixed with https://github.com/blueman-project/blueman/commit/38fb8e8bfd68d3f6850ee6cda56ce4f0d481f8c3. Let me know if you have further issues with this.

closed 2016-09-05 by infirit