blueman-project/blueman #586: Alpha release

cschramm, 2016-09-09:
To speed up a little bit I'd target a first alpha release for version 2.1 next Wednesday, September 14th. By then the Changelog should be up to date and everything that's nearly finished should be merged.

From my side I want to get #578 in and I'll skim through all open issues in the next couple of days to identify obsolete ones and ones that should make it into 2.1. I doubt that there's anything that fits a first alpha though.

We could also think of a 2.0.5 bug fix release. There's nothing really important in stable-2-0 but it's at least a list of five different fixes that piled up.

infirit, 2016-09-09:
>I want to get #578 in

I'll have a go at it this weekend.

>We could also think of a 2.0.5 bug fix release

Sound good, like you said nothing important but small fixes.

I don't have anything further for the first alpha. I do like to get all the plugins ported to GDBus before the final but I lack the hardware for one. I can probably make some initial patches which others can then use as a base. If you have a a dun phone let me know and I'll post a patch to  #568 :smile:

cschramm, 2016-09-09:
> If you have a a dun phone

I don't (as far as I know), but I'll try to simulate one with an HC-05.

closed 2016-09-16 by cschramm