blueman-project/blueman #588: Problem with bluetooth pair

Vairetz, 2016-09-10:
Hi, I have a problem while I'm pairing my smartphone with raspberry pi.
With other phone work but with my huawei p9 lite I can't pair.
What is the problem and can I solve it? 

Raspberry Pi 2 Model B with Raspbian

infirit, 2016-09-10:
If other phones work then there is probably nothing wrong in blueman. Does it pair when using ``bluetoothctl`` from a terminal? If not see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging and post the output from blueman-applet.

Vairetz, 2016-09-10:
When I pair my device on the screen not appear the window of PIN and failed the connection.
whit comand bluetoothctl print: 
pi@raspberrypi:~ $ bluetoothctl
[NEW] Controller 00:02:72:02:55:0A RaspberryPi2 [default]
[NEW] Device D4:A1:48:57:DD:44 Davide
[NEW] Device 08:21:EF:31:FA:AC Galaxy J3
[bluetooth]# 

my device is the second, now what should I do?

infirit, 2016-09-12:
issue ``pair 08:21:EF:31:FA:AC`` from within bluethoothctl.

Vairetz, 2016-09-13:
Error message is: 
"Failed to pair: org.bluez.Error.AuthenticationTimeout".
It's possible that the broblem is the old version of my bluetooth usb dongle?
version bluettoth usb is 2.1 and version bluetooth huawei p9 is 4.1

Vairetz, 2016-09-13:
Error message is: "Failed to pair: org.bluez.Error.AuthenticationTimeout".
I try to star pair from smartphone and it's work but my raspberry don't recognizes phone as audio source.
With my old phone works.. I don't understand what is the problem

Vairetz, 2016-09-13:
Raspberry it's doesn't connected with my phone, is only pair.

Vairetz, 2016-09-14:
A problem could be the fact that raspberry don't support automatic connection so my huawei can't connect.. Someone could help me?

Vairetz, 2016-09-16:
I solved the problem of association but raspberry can't comunicate with my phone, please help me :(

Vairetz, 2016-09-16:
I solved the association's problem but raspberry can't comunicate with my phone, please help me :(

Vairetz, 2016-09-16:
Errore message is: 
Connection failed, possible causes is: remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired or the network connection was broken.

infirit, 2016-09-17:
That means there is a problem communicating between the adapter and the device. Unfortunately there is very little we can do about that.

cschramm, 2016-10-06:
Looks like some dbus issue. Please provide the full output, see https://github.com/blueman-project/blueman/wiki/Troubleshooting.

closed 2016-11-18 by cschramm