blueman-project/blueman #589: Error pairing bose headphones due to gbus Authentication Failed error

justinabrahms, 2016-09-15:
I'm having some issues pairing my Bose QC35's on my ubuntu desktop. I'm getting an authorization failed error.

Versions
- bluez: 4.101-0ubuntu13.1
- blueman: 1.23-git201403102151-1ubuntu1 (via the blueman ppa, but also tried via the one in default apt with the same error)
- desktop env: ubuntu's unity
- ubuntu: 14.04 on kernel 4.2.0-42-generic

What I did:
- run `blueman-applet`
- devices > search > add device > trust > pair 

Result: Authentication Failed
Desired result: bluetooth audio.

Logs

```

_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Bose QuietComfort 35 
_________
on_device_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:168)
list: device_prop_ch Connected 1 /org/bluez/32021/hci0/dev_04_52_C7_0E_11_D3 () {} 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:286)
row update event Connected 1 
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Bose QuietComfort 35 
_________
monitor_power_levels (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:190)
starting monitor 
_________
monitor_power_levels (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:190)
Failed to get power levels 
_________
Release (/usr/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py:151)
Agent.Release 
_________
Cancel (/usr/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py:158)
Agent.Cancel 
_________
err (/usr/bin/blueman-manager:251)
(DBusException(dbus.String(u'Authentication Failed'),),) 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/applet/BluezAgent.py:148)
Agent on path /org/blueman/agent/temp/0452C70E11D3 deleted 
_________
on_device_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:168)
list: device_prop_ch Connected 0 /org/bluez/32021/hci0/dev_04_52_C7_0E_11_D3 () {} 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:286)
row update event Connected 0 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Bose QuietComfort 35 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/32021/hci0/dev_04_52_C7_0E_11_D3 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/32021/hci0/dev_04_52_C7_0E_11_D3 
/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerProgressbar.py:174: Warning: Source ID 1905 was not found when attempting to remove it
  gobject.source_remove(self.gsource)
```

infirit, 2016-09-17:
That is an old version of blueman, many things changed since then. However, Please run through https://wiki.debian.org/BluetoothUser/a2dp and make sure everything is installed. So many issues are because of missing packages :smile: .

Second, can you post the output of __blueman-applet__.

justinabrahms, 2016-09-19:
I don't see `bluez-firmware` anywhere for Ubuntu 14.04, so is that the issue? I have installed `bluez`, `blues-alsa`, and `bluez-cups` if those help. They're all version `4.101-0ubuntu13.1`

Full output is available here: https://gist.github.com/justinabrahms/0927393eeab901f7780a3e6d83fb011e

justinabrahms, 2016-09-19:
I also installed bluez from source (via the user-space bluez package here http://www.bluez.org/download/ )

Newest output here: https://gist.github.com/justinabrahms/9d3300810fdbb8324f32f79aa1332b7b

cschramm, 2016-10-05:
Maybe the BlueZ log (syslog) has some information. Otherwise running `bluetoothd` with the `-d` flag might uncover some hint what's going wrong.

LAfricain, 2016-10-14:
I have the same problem with JBL speakers.

closed 2016-11-18 by cschramm

justinabrahms, 2016-11-18:
This is what syslog had to say:

```
Nov 18 19:04:04 u705a0f8245ab57448c53 bluetoothd[18487]: Discovery session 0x558077192640 with :1.1608 activated
Nov 18 19:04:12 u705a0f8245ab57448c53 bluetoothd[18487]: Unknown command complete for opcode 37
Nov 18 19:04:20 u705a0f8245ab57448c53 bluetoothd[18487]: Stopping discovery
Nov 18 19:04:20 u705a0f8245ab57448c53 bluetoothd[18487]: Unknown Management opcode 28 (index 0)
Nov 18 19:04:20 u705a0f8245ab57448c53 bluetoothd[18487]: No GATT server found in 58:91:CF:7D:F7:6B
Nov 18 19:04:20 u705a0f8245ab57448c53 bluetoothd[18487]: No GATT server found in 58:91:CF:7D:F7:6B
Nov 18 19:04:20 u705a0f8245ab57448c53 bluetoothd[18487]: Attribute server attach failure!
Nov 18 19:04:28 u705a0f8245ab57448c53 bluetoothd[18487]: Unknown command complete for opcode 37
Nov 18 19:04:36 u705a0f8245ab57448c53 bluetoothd[18487]: No GATT server found in 58:91:CF:7D:F7:6B
Nov 18 19:04:36 u705a0f8245ab57448c53 bluetoothd[18487]: No GATT server found in 58:91:CF:7D:F7:6B
Nov 18 19:04:36 u705a0f8245ab57448c53 bluetoothd[18487]: Attribute server attach failure!
Nov 18 19:04:36 u705a0f8245ab57448c53 bluetoothd[18487]: hci0: Cancel Pair Device (0x001a) failed: Invalid Parameters (0x0d)
Nov 18 19:04:50 u705a0f8245ab57448c53 bluetoothd[18487]: No GATT server found in 58:91:CF:7D:F7:6B
Nov 18 19:04:50 u705a0f8245ab57448c53 bluetoothd[18487]: No GATT server found in 58:91:CF:7D:F7:6B
Nov 18 19:04:50 u705a0f8245ab57448c53 bluetoothd[18487]: Attribute server attach failure!
Nov 18 19:04:50 u705a0f8245ab57448c53 bluetoothd[18487]: hci0: Cancel Pair Device (0x001a) failed: Invalid Parameters (0x0d)
```

cschramm, 2016-11-24:
I think http://askubuntu.com/questions/574268/microsoft-arc-touch-mouse-doesnt-working-on-ubuntu-14-04 is your problem.

Nick-Deubert-Bose, 2017-06-20:
I know this thread is a few months old, but I am having the exact same problem and I am on 16.04 with bluez 5.37.

I am pretty sure I have all the necessary packages installed, in fact the first time I paired the headphones with this machine it worked prefect, I set the profile to A2DP and configured the volume controls and I had audio playing. Now however I cannot get it to pair again (after a few days and a reboot).

First what led me to this thread is I am getting this from bluetooth-assistant:
```
monitor_power_levels (/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py:185)
Failed to get power levels
Get connection info failed 
Authentication Failed
```
 
Then I am getting these errors when I restart bluetoothd (sudo /etc/init.d/bluetooth restart):
```
Jun 20 10:27:46 dusseldorf systemd[1]: Stopping Bluetooth service...
Jun 20 10:27:46 dusseldorf bluetoothd[24519]: Terminating
Jun 20 10:27:46 dusseldorf bluetoothd[24519]: Endpoint unregistered: sender=:1.61 path=/MediaEndpoint/A2DPSource
Jun 20 10:27:46 dusseldorf bluetoothd[24519]: Endpoint unregistered: sender=:1.61 path=/MediaEndpoint/A2DPSink
Jun 20 10:27:46 dusseldorf bluetoothd[24519]: Stopping SDP server
Jun 20 10:27:46 dusseldorf bluetoothd[24519]: Exit
Jun 20 10:27:46 dusseldorf systemd[1]: Stopped Bluetooth service.
Jun 20 10:27:46 dusseldorf dbus[1040]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service'
Jun 20 10:27:46 dusseldorf systemd[1]: Starting Bluetooth service...
Jun 20 10:27:46 dusseldorf systemd[1]: Started Bluetooth service.
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Bluetooth daemon 5.37
Jun 20 10:27:46 dusseldorf dbus[1040]: [system] Successfully activated service 'org.bluez'
Jun 20 10:27:46 dusseldorf systemd[1]: Started Bluetooth service.
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Starting SDP server
Jun 20 10:27:46 dusseldorf dbus[1040]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service'
Jun 20 10:27:46 dusseldorf systemd[1]: Starting Hostname Service...
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Bluetooth management interface 1.10 initialized
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Failed to obtain handles for "Service Changed" characteristic
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Not enough free handles to register service
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Error adding Link Loss service
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Not enough free handles to register service
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: message repeated 2 times: [ Not enough free handles to register service]
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Current Time Service could not be registered
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: gatt-time-server: Input/output error (5)
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Not enough free handles to register service
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Not enough free handles to register service
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Sap driver initialization failed.
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: sap-server: Operation not permitted (1)
Jun 20 10:27:46 dusseldorf pulseaudio[2258]: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Endpoint registered: sender=:1.61 path=/MediaEndpoint/A2DPSource
Jun 20 10:27:46 dusseldorf bluetoothd[25593]: Endpoint registered: sender=:1.61 path=/MediaEndpoint/A2DPSink
Jun 20 10:27:46 dusseldorf dbus[1040]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jun 20 10:27:46 dusseldorf systemd[1]: Started Hostname Service.
Jun 20 10:27:47 dusseldorf dbus[1040]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Jun 20 10:27:47 dusseldorf org.blueman.Mechanism[1040]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Jun 20 10:27:47 dusseldorf org.blueman.Mechanism[1040]: Unable to init server: Could not connect: Connection refused
Jun 20 10:27:47 dusseldorf org.blueman.Mechanism[1040]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Jun 20 10:27:47 dusseldorf org.blueman.Mechanism[1040]: Unable to init server: Could not connect: Connection refused
Jun 20 10:27:47 dusseldorf blueman-mechanism: Starting blueman-mechanism
Jun 20 10:27:47 dusseldorf dbus[1040]: [system] Successfully activated service 'org.blueman.Mechanism'
Jun 20 10:27:47 dusseldorf org.blueman.Mechanism[1040]: (blueman-mechanism:25618): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Jun 20 10:27:47 dusseldorf blueman-mechanism: loading Network
Jun 20 10:27:47 dusseldorf blueman-mechanism: loading Rfcomm
Jun 20 10:27:47 dusseldorf blueman-mechanism: loading Ppp
Jun 20 10:27:47 dusseldorf blueman-mechanism: loading RfKill
Jun 20 10:28:17 dusseldorf blueman-mechanism: Exiting
```
Please tell me what else I can do to troubleshoot this, I really want to get this working again. Thanks.

cschramm, 2017-07-13:
@Nick-Deubert-Bose: Searching for your bluetoothd errors hints at some RF capability issue. Check `rfkill list`.

Nick-Deubert-Bose, 2017-07-13:
I see this:
```
$ rfkill list
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
199: nfc0: NFC
        Soft blocked: yes
        Hard blocked: no
200: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

cschramm, 2017-07-13:
No idea then, sorry, but you clearly have some low level issue with your bluetoothd, firmware, or whatever.

basickarl, 2020-05-01:
Getting the same issue on Ubutnu 16.04 with my UE speaker.