blueman-project/blueman #592: Bluetooth PANU - 'failed to obtain IP address'

alkopop79, 2016-09-22:
I have managed to setup Bluetooth tethering with my phone using the Blueman manager (phone-server, Pi-client). However I cannot do the same from terminal. I have tried `test-network <address>` but it gives me the error 'failed to obtain IP address'. Again, when using the GUI (blueman-manager), the connection works fine. How can I establish the same connection form command line?

infirit, 2016-09-22:
So this is a bit out of scope for an issue on blueman but... We use dbus to activate the networkmanager connection. First find the connection and then use it to activate the connection, see https://github.com/blueman-project/blueman/blob/master/blueman/plugins/applet/NMPANSupport.py

alkopop79, 2016-09-22:
Thank you for the help!

alkopop79, 2016-09-23:
SOLVED: `test-network 40:B8:37:68:F1:62 **nap**`

That extra 'nap' bit made all the difference!

closed 2016-09-23 by alkopop79