blueman-project/blueman #593: blueman-manager fails to connect to phone while bluetoothctl connects

parkerlreed, 2016-09-25:
Arch x86_64
bluez 5.41-2
bluez-utils 5.41-2
blueman-git 2.1.alpha1.r1.g1399888-1

I already had the phone added in bluetothctl

````
[parker@x53e ~]$ bluetoothctl 
[NEW] Controller 70:77:81:57:FE:76 x53e [default]
[NEW] Device A0:91:69:97:79:6A Nexus 5X
[NEW] Device A0:14:3D:95:90:65 HandsFreeLink
[bluetooth]# connect A0:91:69:97:79:6A 
Attempting to connect to A0:91:69:97:79:6A
[CHG] Device A0:91:69:97:79:6A Connected: yes
Connection successful
[Nexus 5X]# exit
````

When I launch blueman-manager and right click the phone and select connect audio source i get this

````
_________
set_op (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:95)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f316c16f5a0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xea4f90)> 
_________
set_op (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:95)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f316c16fdc8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xea5220)> 
/usr/lib/python3.5/site-packages/gi/overrides/GObject.py:541: Warning: gsignal.c:2635: instance '0xea5220' has no handler with id '1811'
  return func(*args, **kwargs)
_________
unset_op (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:108)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f316c16f5a0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xea4f90)> 
_________
Generate (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:187)
Nexus 5X 
_________
unset_op (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:108)
op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f316c16fdc8 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0xea5220)> 
_________
fail (/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:133)
fail g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying (4) 
/usr/lib/python3.5/site-packages/blueman/gui/manager/ManagerProgressbar.py:163: Warning: Source ID 526 was not found when attempting to remove it
  GLib.source_remove(self.gsource)
````

The error in the GUI is the same

````
Connection Failed: Message recipient disconnected from message bus without replying
````

infirit, 2016-09-25:
Can you post the output from ``blueman-applet`` as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging please.

parkerlreed, 2016-09-26:
blueman-applet doesn't run at all since I don't have appindicator3 (I'm not on Ubuntu so this has to be compiled separately just to get it installed) I'll take a stab at getting it installed right now and see if I can get it running.

parkerlreed, 2016-09-26:
Ooooooh, I didn't realize the applet HAD to be running for it to work. After getting the applet running, manager connects fine! Thanks for the pointer.

closed 2016-09-26 by parkerlreed

infirit, 2016-09-26:
>since I don't have appindicator3

It is not required for the applet to run.

>I didn't realize the applet HAD to be running for it to work.

Yes, the applet is the main part of blueman. It provides the services needed over dbus to connect among other things :).

parkerlreed, 2016-09-26:
> It is not required for the applet to run.

How do you run it without? Is there a configure flag for it?

````
[parker@e73 ~]$ blueman-applet 
blueman-applet version 2.1.alpha1 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:61)
['ExitItem', 'AppIndicator', 'StatusIcon', 'DiscvManager', 'PowerManager', 'TransferService', 'AuthAgent', 'DhcpClient', 'SerialManager', 'Menu', 'StandardItems', 'Networking', 'NMDUNSupport', 'PPPSupport', 'GameControllerWakelock', 'NMPANSupport', 'ShowConnected', 'NetUsage', 'RecentConns', 'DBusService', 'KillSwitch'] 
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 27, in <module>
    BluemanApplet()
  File "/usr/lib/python3.5/site-packages/blueman/main/Applet.py", line 42, in __init__
    self.Plugins.Load()
  File "/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py", line 91, in Load
    __import__(self.module_path.__name__ + ".%s" % plugin, None, None, [])
  File "/usr/lib/python3.5/site-packages/blueman/plugins/applet/AppIndicator.py", line 10, in <module>
    gi.require_version('AppIndicator3', '0.1')
  File "/usr/lib/python3.5/site-packages/gi/__init__.py", line 102, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AppIndicator3 not available
````

(Just installed on another machine to show what I was getting)

parkerlreed, 2016-09-26:
Aha --disable-appindicator

infirit, 2016-09-26:
Apparently setting the required version needs it installed, that is crappy of gi. The below will fix that,

```diff
diff --git a/blueman/plugins/applet/AppIndicator.py b/blueman/plugins/applet/AppIndicator.py
index 22bd9fd..c45b7c8 100644
--- a/blueman/plugins/applet/AppIndicator.py
+++ b/blueman/plugins/applet/AppIndicator.py
@@ -7,8 +7,12 @@ from __future__ import unicode_literals
 from blueman.plugins.AppletPlugin import AppletPlugin
 
 import gi
-gi.require_version('AppIndicator3', '0.1')
-from gi.repository import AppIndicator3 as girAppIndicator
+
+try:
+    gi.require_version('AppIndicator3', '0.1')
+    from gi.repository import AppIndicator3 as girAppIndicator
+except ValueError:
+    raise ImportError
 
 
 class AppIndicator(AppletPlugin):
```