blueman-project/blueman #594: Program secretly needs gnome-user-share

ACinnamonUserInUK, 2016-09-28:
Well done to `Blueman` for actually working on Mint 18 (`Blueberry` didn't) but there do seem to be problems and infelicities. Here is one such thing (I'll file separate reports for the others).

At least on Mint, `Blueman` works properly  - can receive files only with `gnome-user-share` installed (and configured). But `Blueman` does not tell the user this. Indeed `gnome-user-share` is not even a recommended package under `apt' (although `indicator-application` and `update-motd` are - should I install them?).

Linux Mint 18 x64 Cinnamon
BlueMan 2.0.4-1

cschramm, 2016-09-28:
This is a packaging bug, but as I'm the maintainer of the Debian package it's fine to discuss it here. I assume Linux Mint 18 has some Bluez 5.37 version. I have no idea why `gnome-user-share` would be a requirement. Please check [Troubleshooting](https://github.com/blueman-project/blueman/wiki/Troubleshooting) for reporting what happens if it's not installed.

ACinnamonUserInUK, 2016-09-28:
`I have no idea why gnome-user-share would be a requirement. Please check Troubleshooting for reporting what happens if it's not installed.`

Well, what happens is that, on the sending device - an Android Moto G 1st generation - I receive a popup notification saying that the file I had tried to send by bluetooth was not sent - and that seems to happen with whatever file types I try to send, and from whatever Android app I try to send them.

I would provide information of a more technical sort - as I take you to be suggested that I do - but I discover that actually I haven't quite got the thing working yet. For, the bluetooth manager is not starting at boot. Once I've fixed this, I might return here and add details of the sort you request. In the meantime, just getting the thing to work is the priority!

cschramm, 2016-09-29:
I guess when `gnome-user-share` is installed it handles the incoming file transfers exclusively and it does not work with blueman at all. Without any data we cannot say why, so I'll close this for now.

closed 2016-09-29 by cschramm