blueman-project/blueman #595: Dialogue box for choosing a device misleads

ACinnamonUserInUK, 2016-09-28:
It seems to me that there is something rather confusing about the window/dialogue box for choosing a device as a recipient. Specifically, I have in mind the bar at the bottom. That bar gave me the impression that had to wait, or perhaps that, no, I should not wait, because there will be a timeout. In reality, the bar indicates - I think - a search for as-yet-unlisted bluetooth devices. The fact that Android has a similar bluetooth interface doesn't make it less confusing. How about an explanatory label?

Also, any chance of a 'send to' in the context-menu of file managers such as Nemo?

cschramm, 2016-09-28:
Regarding the "send to": We once shipped a nautlius-sendto plugin until they dropped plugin support. We discussed basically the same thing for caja in #220. There's a feature request at https://github.com/mate-desktop/caja-extensions/issues/12. We will look into that for blueman 2.1. The repository for nemo would be https://github.com/linuxmint/nemo-extensions. I guess this is very similar except that we do not have the GTK+ 2 barrier.

infirit, 2017-01-21:
So, this is what I like to do.

 * Turn the progress bar into a spinner and have a label with a text, something like "Discovering devices".
 * Do not timeout on discovery  any more and stop discovery when device is selected or app is closed.
 * Add a small delay for adapters that can not communicate properly when discovering

Objections?

ACinnamonUserInUK, 2017-01-21:
Sounds good to me, though for the moment I've given up on Bluetooth / replaced it with wifi technologies.

infirit, 2017-08-09:
First WIP PR  #744

infirit, 2017-08-19:
#744 was merged, closing.

closed 2017-08-19 by infirit