blueman-project/blueman #596: Extensionless files are not sent

ACinnamonUserInUK, 2016-09-28:
**What happens**

There is a problem sending a certain type of file from Linux to Android.

- _The Linux device:_ blueman 2.0.4-1; BlueZ 5.37-0ubuntu5; distribution - Linux Mint 18; desktop environment - Cinnamon.
- _The Android device_: Moto G 1st generation (model XT1032).
- _The type of file:_ an extensionless text file (that isn't marked as executable). This is a very common Linux filetype.
- _The problem_: the file does not get sent. There seems to be no error message, either on the Linux box or on the phone.

**What should happen**

Ideally, the file would be sent. Less good, but still better than present behaviour, would be some sort of error message and, moreover, one that indicated that the problem was with the file type. Admittedly, the problem may owe, not to the Blueman running on Linux, but to Android.

infirit, 2016-10-02:
I do get an error and it is in the systemlog (sudo journalctl -f) by obexd, see below. This means the phone rejected the transfer and I suspect you have a similar message in your logs. It is also consistent between gnome-bluetooth and blueman.

ps: out of curiosity, cinnamon uses gnome-bluetooth by default, why are you using blueman?
```
Oct 02 18:01:02 sander-VirtualBox obexd[2964]: Transfer(0x560c2a155730) Error: Forbidden
Oct 02 18:01:24 sander-VirtualBox obexd[2964]: Transfer(0x560c2a1557a0) Error: Forbidden
Oct 02 18:01:36 sander-VirtualBox obexd[2964]: Transfer(0x560c2a155730) Error: Bad Request
```

closed 2016-10-02 by infirit

ACinnamonUserInUK, 2016-10-02:
So the problem is with the phone? Right. I wonder whether there is a way of making the phone accepted extensionless files. (I do have a rooted 'phone, so some moderately deep adjustments are possible.)

I use blueman for the following reason. When I used Cinnamon's built-in bluetooth software (which I had thought was called 'blueberry' . .), my Linux machine would not receive any files via bluetooth.

infirit, 2016-10-02:
>So the problem is with the phone

Yes, but I wouldn't call it a problem. Android is very strict in what it accepts and filters a lot, it is intentional. 

>I wonder whether there is a way of making the phone accepted extensionless files. 

Not without installing third party tools, see [1].

>which I had thought was called 'blueberry

It uses gnome-bluetooth in the background and it only a frontend.

>my Linux machine would not receive any files via bluetooth

I suggest reporting this to the mint folks, not that I want you to stop using blueman :smile:, but it sound like a bug to me.

[1] http://htc-one.wonderhowto.com/how-to/bypass-androids-file-type-restrictions-bluetooth-file-sharing-0157237/

ACinnamonUserInUK, 2016-10-02:
> stop using blueman 😄, but it sound like a bug to me.

Thank you. That's all very helpful - and I had already reported the
problem to Mint.

cschramm, 2016-10-02:
Android's internal handler for OBEX FTP is an app with the package name `com.android.bluetooth`. It tries to find a mime type from for the file either from the bluetooth header (will not be present in our case) or from the file extension. It will only accept a file it can find a mime type for (which already eliminates extensionless files *and* whose mime type is not `text/x-vcalendar` *and* that is either sent from a whitelisted device or has a whitelisted mime type (see https://android.googlesource.com/platform/packages/apps/Bluetooth/+/master/src/com/android/bluetooth/opp/Constants.java#218).

To make it accept other files you will need a third-party app as @infirit mentioned or a modified `com.android.bluetooth app`. The location for that modification would be `com.android.bluetooth.opp.BluetoothOppObexServerSession.onPut`.

ACinnamonUserInUK, 2016-10-02:
Thanks. I think my hands are fairly tied, though. For, (1) as yet there
doesn't seem to be a version of the Xposed framework that runs on my
phone (and the apps that could change the behaviour in question need
that framework), (2) I think I am unable to change
|com.android.bluetooth app|.