blueman-project/blueman #604: GUI: some menu items have no accelerator

guntbert, 2016-10-26:
- Blueman 2.0.4
- on Ubuntu 16.10

Several menu items (especially beneath the "Devices" and "View" menus) don't have the accelerator for accessing them directly without using the mouse or arrow keys.

I'd be happy to work on this. Can you tell me **where** in the project I can find the relevant code?

infirit, 2016-10-26:
Yeah, see  #605. No offense but me explaining it would have taken a lot more time :smile:.

There are (dis)connectables like serial ports, pan, audio etc that do not need to have accelerators. While the number we use is limited, making sure things are always unique will be very difficult.

infirit, 2016-10-29:
Pushed to master. This will be in the next major release.

PR #605 (2016-10-29): Manager: add accelerators to manager menus when missing
closes #604
commit af4c8876 (2016-10-29): Manager: add accelerators to manager menus when missing

closes #604
closed 2016-10-29 by infirit