blueman-project/blueman #606: interface problem

rwst, 2016-10-29:
Trying to connect with a Samsung Soundbar via the main Setup button will open the Bluetooth Authentication box and then take some time until presenting me with "Device added successfuly(sic) but failed to connect". However, if I after that use the right mouse button on the entry and Setup I quickly get "Device added and connected successfuly". This is the only method I can connect. It seems that the main Setup button, as well as the code I start via the Device->Setup menu entry is missing something which only the right-click combo gets right.

remborg, 2016-10-29:
I have exactly the same problem with an HK bluetooth speaker on linux mint.
I ran the blueman-applet from the command line and here is what I get when creating the device for the first time (resulting a "Device added successfully but failed to connect")

`__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:67)
deleting device None 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:89)
invalidating device None 
_________
Run (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py", line 207, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
    add_menu_item(manager_menu, service)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
    if service.connected:
  File "/usr/lib/python2.7/dist-packages/blueman/Service.py", line 54, in connected
    return self._legacy_interface.GetProperties()['Connected']
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Handsfree" doesn't exist

_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:165)
HK Onyx Studio 2 
_________
on_device_property_changed (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:156)
list: device_prop_ch Connected 1 /org/bluez/781/hci0/dev_0C_A6_94_8F_62_BF () {} 
_________
row_update_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Connected 1 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:21)
caching initial properties 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:67)
deleting device /org/bluez/781/hci0/dev_0C_A6_94_8F_62_BF 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:89)
invalidating device /org/bluez/781/hci0/dev_0C_A6_94_8F_62_BF 
_________
Run (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py", line 207, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
    add_menu_item(manager_menu, service)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
    if service.connected:
  File "/usr/lib/python2.7/dist-packages/blueman/Service.py", line 54, in connected
    return self._legacy_interface.GetProperties()['Connected']
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Handsfree" doesn't exist

_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:165)
HK Onyx Studio 2 
_________
[... 93 lines cut ...]
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
    add_menu_item(manager_menu, service)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
    if service.connected:
  File "/usr/lib/python2.7/dist-packages/blueman/Service.py", line 54, in connected
    return self._legacy_interface.GetProperties()['Connected']
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "GetProperties" with signature "" on interface "org.bluez.Handsfree" doesn't exist

_________
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:165)
HK Onyx Studio 2 
_________
level_setup_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:322)
animating down 
_________
update (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected) 

`

infirit, 2016-10-29:
Most issues related to audio streaming are resolved by installing the neccesary packages. So before looking into this please look at the two wiki pages mentioned in https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect

For debian derivatives like mint read the debian wiki first.

infirit, 2016-10-29:
Also before I forget, please do report which disto and importantly the version, what is mentioned on the wiki [here](https://github.com/blueman-project/blueman/wiki/Troubleshooting#reporting-a-bug--asking-for-support)

rwst, 2016-10-31:
This is OpenSuSE Leap 42.2 with pulseaudio-bluetooth installed. The original problem is not about not connecting, but only via a mysterious combo of interface actions.

rwst, 2016-10-31:
I'm using blueman git master.

cschramm, 2016-11-11:
All right, I feel like speculating and combining your little information today.

You gave away that you're running OpenSuSe Leap 42.2. According to the distributor's package search you thus cannot have any official blueman package and probably have bluez 5.35 (I'm a little bit confused here as that version is only listed for 42.1 and it says 42.2 would not have an official package).

The log shows an output from some 2.0-ish version of blueman which seems to have detected BlueZ 4.

Even though you gave us very little information, it already contradicts (note that I took @remborg's information that you two have "exactly the same problem" literally :wink:).

rwst, 2016-11-11:
It's bluez-5.41-1.2 and as I said it's blueman git master (specifically commit ce5f7774507e1e4f4e7e72c47dcbdbdcb615dd88).

cschramm, 2016-11-11:
Your issue does not fit @remborg's data then. Please give us the blueman-applet output for the failing procedure, see @infirit's link.

closed 2017-02-02 by cschramm