blueman-project/blueman #609: Appindicator get_geometry off

infirit, 2016-10-31:
When we enable the AppIndicator plugin we run into the below error and show the notification outside the screen. Without the status icon returns proper geometry, see the second block. Not sure what is going on but we need to fix this before final release.

```
(blueman-applet:25655): Gdk-CRITICAL **: gdk_window_get_origin: assertion 'GDK_IS_WINDOW (window)' failed
_________
geometry (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/StatusIcon.py:45)
1274976 32542 1 1 
```

```
geometry (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/StatusIcon.py:45)
1565 3 22 22 
```

cschramm, 2016-11-01:
How do you get that error? Cannot reproduce (in Xfce panel).

infirit, 2016-11-01:
I get it the moment we request the geometry under xfce panel.

btw, [API: Remove GtkStatusIcon](https://github.com/GNOME/gtk/commit/d2a8667f835ddce4ac88b9bee143556f9efa93c4)

infirit, 2017-05-19:
#637 drops this so we can close it when it lands in master

closed 2017-11-28 by cschramm