blueman-project/blueman #610: GtkStatusIcon removed from Gtk+

infirit, 2016-11-01:
GtkStatusIcon was [removed](https://github.com/GNOME/gtk/commit/d2a8667f835ddce4ac88b9bee143556f9efa93c4) in Gtk+ master in September 2016 so very likely it will be gone in Gtk+ 3.24. We need to plan what to do as we rely on it.

* Use Appindicator plugin as the default?
* Which Gtk+ version will we target for 2.1?
* More things we need to look at?

Using appinficator we loose [tooltip](https://bugs.launchpad.net/indicator-application/+bug/527458) and opening/closing blueman-manager as left click is not under our control.

cschramm, 2016-11-01:
We have #306 lying around. I didn't really have a look at the GNotification stuff and what's necessary for that.

> Which Gtk+ version will we target for 2.1?

Well, we'll definitely need to support newer versions. Using GNotification would raise the minimum GLib version to 2.39. Do you see other reasons to raise the minimum version?

infirit, 2016-11-01:
I edited the original text, the new version will likely be 4 and 3 will be the left alone. See https://blogs.gnome.org/desrt/2016/06/13/gtk-4-0-is-not-gtk-4/ and https://wiki.gnome.org/Hackfests/DeveloperExperience2014/GtkRoadmap. But we still need to fix this eventually though.

Using GNotification still leaves us without an icon under __all__ desktops. It is not meant to replace the trayicon. It is similar to the current notification spec but with extra feature like persistence across sessions.

>raise the minimum GLib version to 2.39

The issues is with the slack users, everyone else is most likely already on 2.39 when 2.1 is released.

cschramm, 2016-11-01:
> I edited the original text, the new version will likely be 4 and 3 will be the left alone.

Right, it'll be 3.90 which will be called `gtk+-4.0` in pkg-config. So GtkStatusIcon won't get dropped in GTK+ 3 and as 2.24 is still receiving patches I don't see a reason to migrate away from it any time soon. :smiley: 

> Using GNotification still leaves us without an icon under all desktops.

Alright, rifled through gnome-shell and now starting to understand the situation: GNotification does not provide any indicator icon functionality. GNOME still has indicators as provided by GtkStatusIcon, but they are implemented in GNOME Shell internally (with Gjs and [the internal St module](https://developer.gnome.org/st/stable/index.html)).

If I'm not missing something, the AppIndicator plugin should already be used by default. It only requires the presence of the libappindicator bindings of course. This is already necessary as the GtkStatusIcon probably does not work in Unity at all and is [somehow broken in KDE Plasma 5](https://github.com/blueman-project/blueman/issues/563).

An alternative to AppIndicator could be KNotifications. Not sure what's the price for that...

infirit, 2016-11-01:
Why don't we separate out the services blueman-applet provides and only have a small status icon module? We would then have,

 - blueman-tray (small tray app, could potentially be gtk2, gtk3 or even qt)
 - blueman-daemon (or some better name that handles plugins, dbus service etc)
 - the already existing script mechanism etc

Then we can also simply move over to GNotification and GApplication and not worry about the tray.

cschramm, 2016-11-02:
Makes sense, yes. If we keep the name blueman-applet (for the blueman-daemon) and keep a StatusIcon plugin that just invokes blueman-tray, even nothing would change for the user. However, a lot of communication has to be implemented between the applet and the tray application as a lot of plugins use `self.Applet.Plugins.StatusIcon`. That'd be a new `org.blueman.tray` service I guess.

> Then we can also simply move over to GNotification and GApplication and not worry about the tray.

Why should we? I always mistakenly thought of GNotification as some kind of replacement for the GtkStatusIcon and only saw the requirement to use GApplication to be able to use GNotitifcation.

cschramm, 2016-11-02:
> If we keep the name blueman-applet (for the blueman-daemon) and keep a StatusIcon plugin that just invokes blueman-tray, even nothing would change for the user.

Just to mention it: Another idea would be to rename blueman-applet -> blueman-daemon and call the new blueman-tray blueman-applet. If it invokes the daemon on startup as other apps like blueman-manager do es well, it would still be the same at least for users who have the StatusIcon plugin enabled (others would get an icon then as well of course as they just invoked the process for that).

infirit, 2016-11-02:
>a lot of communication has to be implemented between the applet and the tray application as a lot of plugins use self.Applet.Plugins.StatusIcon.

They use it only to get the geometry (position hint) for notifications en set the icon name plus tooltip. We could have a dbus signal on the blueman-daemon (lets use this name for now we can bikeshed on it later :smile:) that the StatusIcon/tray can listen for and update the icon, show/hide and tooltip accordingly. As for the position hint, is it really that important that the notification is always shown near the tray applet? If not we drop that and let the notification daemon handle positioning.

alive4ever, 2017-01-16:
``NotShownIn=KDE;GNOME;`` entry should be added to the desktop file to prevent those desktop environment to load blueman applet.

cschramm, 2017-01-16:
@alive4ever: Why?

ACinnamonUserInUK, 2017-01-22:
It might be worth mentioning the following here. On the Mint-Y 'desktop' theme that is now the encouraged theme on Linux Mint (or at least its Cinnamon version) the blueman applet appears as a very hard to see black on grey - thus (see to the right of the pointer):

[image]

Still, blueman is disabled in that screenshot - were it enabled, probably it would be slightly easier to see (because in that case it would be a lighter colour).

infirit, 2017-01-22:
@ACinnamonUserInUK Mint-Y appears to include an icon for blueman which is what you see. The default icon we provide is the traditional blue with the bluetooth logo in it. If this is unacceptable please raise it with the theme authors.

rworkman, 2017-02-07:
Re this: "The issues is with the slack users, everyone else is most likely already on 2.39 when 2.1 is released."

Won't be an issue - latest stable release (14.2) has glib2-2.46.x, and blueman-2.1.x can't be used on earlier releases due to them being bluez-4.x only, so carry on. Thanks for the thought though - much appreciated!

cschramm, 2017-11-28:
With #637 merged, we can easily get rid of `Gtk.StatusIcon` as `blueman.main.indicators.GtkStatusIcon` is the only class that uses it. Currently, `blueman.main.indicators.AppIndicator` is the only alternative but we could add more if there are useful options.

closed 2017-11-28 by cschramm