blueman-project/blueman #611: A2DP Device No Longer Working

PieterJanVanAeken, 2016-11-02:
Suddenly, my Bluetooth A2DP headphones stopped working. The blueman-applet does not appear on startup, but is clearly running.

This is the information coming from blueman-report. Note that I filled in distribution myself and sometimes I can see that XDG_CURRENT_DESKTOP is a "NoneType" which cannot be cast to a string. Not sure if this can cause issues. 

    blueman: 2.0.4
    BlueZ: 5.37
    Distribution: Ubuntu 16.04
    Desktop: GNOME

This is what I am seeing in syslog on start up (the applet shows up in ps -ef but not in the tray where it normally shows up when working):

    Nov  2 15:27:39 enx-vanaepi bluetoothd[840]: Bluetooth daemon 5.37
    Nov  2 15:27:40 enx-vanaepi bluetoothd[840]: Starting SDP server
    Nov  2 15:27:40 enx-vanaepi bluetoothd[840]: Bluetooth management interface 1.10 initialized
    Nov  2 15:27:40 enx-vanaepi bluetoothd[840]: Failed to obtain handles for "Service Changed" characteristic
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: Not enough free handles to register service
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: Error adding Link Loss service
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: Not enough free handles to register service
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: message repeated 2 times: [ Not enough free handles to register service]
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: Current Time Service could not be registered
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: gatt-time-server: Input/output error (5)
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: Not enough free handles to register service
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: Not enough free handles to register service
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: Sap driver initialization failed.
    Nov  2 15:27:41 enx-vanaepi bluetoothd[840]: sap-server: Operation not permitted (1)
    Nov  2 15:27:42 enx-vanaepi NetworkManager[835]: <info>  [1478096862.9454] Loaded device plugin: NMBluezManager (/usr/lib/x86_64-linux-gnu/NetworkManager/libnm-device-plugin-bluetooth.so)
    Nov  2 15:27:43 enx-vanaepi NetworkManager[835]: <info>  [1478096863.4101] bluez: use BlueZ version 5
    Nov  2 15:28:00 enx-vanaepi bluetoothd[840]: Endpoint registered: sender=:1.61 path=/MediaEndpoint/A2DPSource
    Nov  2 15:28:00 enx-vanaepi bluetoothd[840]: Endpoint registered: sender=:1.61 path=/MediaEndpoint/A2DPSink
    Nov  2 15:28:15 enx-vanaepi dbus[791]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
    Nov  2 15:28:15 enx-vanaepi org.blueman.Mechanism[791]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
    Nov  2 15:28:15 enx-vanaepi org.blueman.Mechanism[791]: Unable to init server: Could not connect: Connection refused
    Nov  2 15:28:15 enx-vanaepi org.blueman.Mechanism[791]: Failed to connect to Mir: Failed to connect to server socket: No such file or directory
    Nov  2 15:28:15 enx-vanaepi org.blueman.Mechanism[791]: Unable to init server: Could not connect: Connection refused
    Nov  2 15:28:15 enx-vanaepi blueman-mechanism: Starting blueman-mechanism
    Nov  2 15:28:15 enx-vanaepi dbus[791]: [system] Successfully activated service 'org.blueman.Mechanism'
    Nov  2 15:28:16 enx-vanaepi org.blueman.Mechanism[791]: (blueman-mechanism:2043): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
    Nov  2 15:28:16 enx-vanaepi blueman-mechanism: loading Rfcomm
    Nov  2 15:28:16 enx-vanaepi blueman-mechanism: loading Network
    Nov  2 15:28:16 enx-vanaepi blueman-mechanism: loading Ppp
    Nov  2 15:28:16 enx-vanaepi blueman-mechanism: loading RfKill
    Nov  2 15:28:17 enx-vanaepi gnome-session[1167]: (blueman-applet:1931): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed
    Nov  2 15:28:41 enx-vanaepi bluetoothd[840]: Endpoint registered: sender=:1.109 path=/MediaEndpoint/A2DPSource
    Nov  2 15:28:41 enx-vanaepi bluetoothd[840]: Endpoint registered: sender=:1.109 path=/MediaEndpoint/A2DPSink
    Nov  2 15:28:41 enx-vanaepi bluetoothd[840]: RFCOMM server failed for Headset Voice gateway: rfcomm_bind: Address already in use (98)
    Nov  2 15:28:43 enx-vanaepi gnome-session[2236]: blueman-applet version 2.0.4 starting
    Nov  2 15:28:45 enx-vanaepi blueman-mechanism: Exiting

Then when I run Bluetooth manager from the activities menu (it starts but the applet still does not show up):

    Nov  2 15:29:47 enx-vanaepi /usr/lib/snapd/snapd[849]: api.go:402: DEBUG: use of obsolete "q" parameter: "/v2/snaps?q=blue"
    Nov  2 15:29:47 enx-vanaepi snapd[849]: 2016/11/02 15:29:47.503795 api.go:483: jumping to "find" to better support legacy request "/v2/snaps?q=blue"
    Nov  2 15:29:47 enx-vanaepi /usr/lib/snapd/snapd[849]: api.go:483: jumping to "find" to better support legacy request "/v2/snaps?q=blue"
    Nov  2 15:29:47 enx-vanaepi /usr/lib/snapd/snapd[849]: daemon.go:181: DEBUG: uid=1000;@  GET /v2/snaps?q=blue 284.348924ms 200
    Nov  2 15:29:49 enx-vanaepi /usr/lib/gdm3/gdm-x-session[2222]: Activating service name='org.blueman.Applet'
    Nov  2 15:29:49 enx-vanaepi org.blueman.Applet[2233]: blueman-applet version 2.0.4 starting
    Nov  2 15:29:49 enx-vanaepi org.blueman.Applet[2233]: There is an instance already running

And when I turn on my bluetooth headphones (they are already linked and trusted)

    Nov  2 15:22:54 enx-vanaepi blueman-manager.desktop[3430]: ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

When running from the command line or run blueman-report, the blueman applet shows up in the tray, connects to my headphones but still doesn't allow me to select any audio profile so I cannot get them to work that way either.

I have no idea how to further debug this. Any help would be greatly appreciated.

infirit, 2016-11-02:
Post the __full__ output from ``blueman-applet`` per https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging.

infirit, 2016-12-27:
No response for over a month, closing. Feel free to comment and provide the required info and we will reopen.

closed 2016-12-27 by infirit