blueman-project/blueman #612: cant send file and receive

DarknessHunter, 2016-11-04:
when i updated my OS (parrot security OS debian based) i cant receive or send files, before i update i was able to send and receive please any solve for my problem 
[image]
when i enable this plugin (transfer service) i get this error

infirit, 2016-11-04:
It means you do not have a proper obex server installed and running. Most common in recent distrobutions is ``obexd`` provided by bluez. Install one and it should work again.

See https://github.com/blueman-project/blueman/blob/master/CONTRIBUTING.md for the minimum required info needed to start helping people :wink:

DarknessHunter, 2016-11-05:
Thanks a lot!!

closed 2016-12-27 by infirit