blueman-project/blueman #615: Bluetooth Devices Window does not automatically close when exiting from the Tray(Exit Option)

jbmorris289, 2016-11-13:
I don't think there's much to say here...When I click the Bluetooth Tray Icon, and then the "Exit" Option from the Context Menu, the Tray Icon disappears, but if the Bluetooth Devices Window is open when you click the Option, it stays open.

I really don't know if blueman is supposed to do this, but I really think that this is a bug, I don't think this is supposed to happen.

--Additional/Required info--
Blueman version: 2.0.4
[image]

OS(Distribution)/Desktop Environment: Lubuntu 16.04/LXDE (Text in the picture below is supposed to say 'Lubuntu,' but..)
[image]

infirit, 2016-11-13:
I'll have a look but it is expected for the applet (tray icon) to be auto started when the manager is run. The applet provides a dbus service which is activatable when first used. The manager uses this dbus interface and it would make the applet start.

infirit, 2016-11-13:
Oh, I misread/misunderstood you. The blueman-applet and the blueman-manager are two distinct applications. This is how it was created, I am not sure it is worth it to tie all the application to the applet.

The manager can work for some parts without the applet, blueman-adapter will work completely without the applet.

closed 2016-11-18 by cschramm