blueman-project/blueman #618: Blueman and android phone

maureenmobbs, 2016-11-17:
I have Blueman on my linux mint 17.3 rosa.  It is paired with my android phone. I can easily send files to the phone, but cannot receive.  I have enabled file sharing of course. What is puzzling, is that just once, a photo was sent from my phone to my laptop, and never again.  I have uninstalled  and re-installed blueman, to no avail. Would love this to work properly.

infirit, 2016-11-17:
Can you post the blueman-applet logs as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

Make sure to post all not just error (if there are) :smile:.

maureenmobbs, 2016-11-19:
Please see screen capture I have enclosed. I have all this installed.   It worked once and then not again. Also, when I turn Blueman OFF the indicator light on my laptop starts flashing. When I turn it ON,  the flashing stops 

 

Another problem is changing settings for file sharing. and place to download files. Although I press the activate square, everything stays grey. I then press close, but when I return to my settings, they have all gone back to default.

 

I have uninstalled blueman ,for the time being.  I will try again later.

 

Thank you for your help  and information.  Very much appreciated.

 

Maureen Mobbs

 

  _____

closed 2016-11-24 by cschramm