blueman-project/blueman #619: Indicator for "trustedness" missing

pwil3058, 2016-11-24:
When I press one of the many "mark/unmark this device as trusted" buttons nothing happens in that there appears to be no indication of whether the device being trusted or not has changed.

cschramm, 2016-11-24:
With "many buttons" you mean the item in the device menu, the item in the device's right-click menu, and the button in the manager window, right?

What should happen is that a yellow star just like in those places should show up on the bottom left corner of the device's icon. If that's not the case either the status does not get changed or blueman fails to display that correctly. You can easily check the first point by running `bluetoothctl` while marking a device trusted in blueman. A respective event should show up in `bluetoothctl`.

In any case we'll need basic information and the log output from blueman-applet. Also check if blueman-manager outputs anything when you mark a device trusted.

pwil3058, 2016-11-24:
Re "many buttons", yes.  No matter which one I use nothing seems to happen.

This problem is occurring on a HP Spectre 360 (with Fedora 25 installed) which I'm not currently using so I can't check all of those items but there was definitely no star on the device's icon.  I'll get the other info from the laptop as soon as I can.

I should add that the reason I'm trying this is an attempt to get my bluetooth to connect automatically when I wake the laptop up.  This was working fine on the same machine with Korara 24 (which is Fedora 24 with a few extra bits from rpm fusion added).

While typing this reply I opened the app on my current computer and when I clicked the "always visible" button for the device I got a notification from abrt that "blueman quit unexpectedly" although it still seems to be running OK.  Although, I did notice that this "always visible" state seems to last only as long as the popup dialog is raised.  If I close it and open it again the state has gone back to "Hidden".

pwil3058, 2016-11-24:
OK.  I've got the laptop and can report that nothing shows up in bluetoothctl output when I do the toggle action for "trustedness".  The same is true for the visible radio buttons.

Also it took two attempts to get the mouse to connect and nothing at all showed up in the bluetoothctl output for the failed attempt.  No report of failure or anything like that.

Where do I find the log output for the blueman-applet that you're asking for?

pwil3058, 2016-11-25:
I would like to suggest that you mark an untrusted device with a star with a diagonal red stripe over it as the absence of the star as an indicator of untrustedness is not very user user friendly.

pwil3058, 2016-11-25:
I set trust manually via bluetoothctl and the symbol showed up on the blueman app (as you described) so the display part of the app is working.  Also, now that it is a trusted device, my mouse connects automatically when I wake the laptop and turn the mouse on.

So it would appear that the problem seems to be that settings made in the app aren't getting to the controller?

cschramm, 2016-11-25:
See https://github.com/blueman-project/blueman/wiki/Troubleshooting on how to get the output.

pwil3058, 2016-11-25:
blueman-manager:  2.0.4
BlueZ: 5.43
Distribution: Fedora 25
Desktop Environment: Cinnamon 2.0.7

Attempting to run the blueman-applet from a terminal failed with the following output so I'm unable to provide useful output from the app when attempting the operations that caused the problems I reported.

[peter@spectre ~]$ blueman-applet
blueman-applet version 2.0.4 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:60)
['Networking', 'NetUsage', 'Menu', 'PPPSupport', 'SerialManager', 'StandardItems', 'DhcpClient', 'NMPANSupport', 'ShowConnected', 'StatusIcon', 'TransferService', 'ExitItem', 'AuthAgent', 'KillSwitch', 'RecentConns', 'DBusService', 'GameControllerWakelock', 'PowerManager', 'DiscvManager', 'Headset', 'AppIndicator', 'NMDUNSupport'] 
_________
get_interface_version (/usr/lib/python3.5/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
Load (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:60)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3, introspection typelib not found 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3.5/site-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib/python3.5/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
on_dbus_name_owner_change (/usr/bin/blueman-applet:95)
org.bluez owner changed to :1.6 
_________
__init__ (/usr/lib/python3.5/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
initialize (/usr/lib/python3.5/site-packages/blueman/plugins/applet/RecentConns.py:144)
rebuilding menu 
_________
enumerate_connections (/usr/lib/python3.5/site-packages/blueman/plugins/applet/ShowConnected.py:53)
Found 1 existing connections 
_________
RequestPowerState (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:131)
Requesting True 
_________
check (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:119)
callbacks done 
_________
set_adapter_state (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:90)
True 
_________
set_adapter_state (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:90)
Exception occurred org.freedesktop.DBus.Error.UnknownMethod: Method "Set" with signature "ssb" on interface "org.freedesktop.DBus.Properties" doesn't exist
 
_________
UpdatePowerState (/usr/lib/python3.5/site-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 

(blueman-applet:26864): Gdk-CRITICAL **: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed

infirit, 2016-11-25:
Unfortunately blueman as shipped with fedora 24 is broken. See https://github.com/blueman-project/blueman/issues/537

closed 2016-11-26 by cschramm